ghaerr/elks #1625: Draft Release - ELKS v0.7.0

ghaerr, 2023-07-27:
It's been over a year since ELKS v0.6.0 was released on June 7, 2022. Following are draft release notes for v0.7.0, please comment if something or someone has been left out. If the release should be held for a particular feature other than the two below, please comment. Thank you to all that have contributed and are using ELKS!

It has been fairly quiet, I would like to add the following minor enhancements and release v0.7.0 shortly:
- Add ANSI cursor on/off sequence for IBM PC
- Add 386 CPU detection code to automatically enable XMS for system buffers when ELKS is run on a 386+.

***Draft Release - ELKS v0.7.0***

**ELKS v0.7.0 Downloads**

Prebuilt disk images for floppy and hard disks are downloadable below from Github Assets for this release. These images can be used to boot from PC/XT/AT or PC-98 hardware, or emulators, such as QEMU or dosbox-x. The images are built with network support by default, supporting NE2K, WD and 3Com NICs. See the Wiki HowTo for more information.

**ELKS v0.7.0 Enhancements**
Thank you to all who contributed to ELKS enhancements, bug identification, testing and fixes. The system continues to be very stable, with no known crashes and long uptimes.

**New Systems Supported**
- MK-88 Russian clone of using 8088 clone K1810BM88 w/256k RAM (Use CONFIG_HW_MK88) @Vutshi
- Monotech NuXT 2.0 board (NEC V20) using Sergey Kiselev’s BIOS (@tt1542)
- Extensive testing of older IBM 5160 PC/XT/AT and Amstrad 1640 (@toncho11)

**New Commands**
- Port `mail` application from MINIX (@ccoffing)
- Add `fm` visual file manager
- Add `disasm` for ELKS binary file symbolic disassembler
- Add `cons` to display console screen from remote console
- Add `ttyinfo` for testing mouse and ANSI arrow, function key and mouse parsing
- Add `hostdisasm` for host-based ELKS disassembler
- Add tiny `min_init` replacement for /bin/init on very small systems
- Add `matrix` screen saver
- Port text adventure game `advent` from Fuzix
- Port `sl` (train) program for mistyped `ls`
- Port `split` from BSD 4.3reno (@Mellvik)
- Support D-Flat Project and `memopad` editor application

**Updated Commands**
- Enhance `meminfo` to include process names
- Fix `cron` memory leak, other fixes (@ccoffing, @ghaerr)
- Fix `tinyirc` to operate using ping/pong protocol (@Redfoxymoon, @DutchComputerKid)
- Use PATH_MAX in all applications to prevent buffer overflow (@ccoffing)
- Nano-X won’t display cursor when no mouse present
- Fix `login` handling on locked accounts, no home dir or long login name (@ccoffing)
- Fix `passwd` to work on FAT filesystems
- Update `nxtetris` to work at constant speed and no mouse (adds GrGetNextEventTimeout)
- Update `disasm` disassembler from 86sim project, add symbolic disassembly to ELKS binaries
- Add `disasm -k` option to disassemble running kernel with symbols, -a for asm input
- Update `ls` for better operation with deeply nested directories and no permissions (@ccoffing)
- Update `ash` for invalid TERM variable checking (@ccoffing)
- Fix `xargs `with long input lines (@ccoffing)
- Fix possible memory corruption with `dd` errors (@ccoffing)
- Add MOUSE_PORT=none for all Nano-X applications
- Cleanup cmds source tree to remove many compilation warnings
- Fix `screen` to run from FAT filesystem
- Remove `pwdauth`, `prems` and mtools from distribution image
- Rename all test programs /bin/test_xxx
- Fix various `tr` bugs and crashes (@ccoffing)
- Fix `ash` build error on non-standard `rindex`/`index` definitions (@ccoffing)
- Fix `fsck` out of memory problem, display issues and -r (@Mellvik, @ghaerr)
- Update `mount` and `df` to display free space and all mounted filesystems (@Mellvik)
- Fix `fm` stack overwrite
- Use default drive for `fdisk` when none specified (@Mellvik)
- Fix `find` stack overflow
- Fix sometimes improper time sort and display using `ls -l`
- Enhances `more` to clear screen and display (END)
- Use `cp -R` in `sys` to fix large copy on large /bin directories (@Mellvik)
- Add PEEK and POKE to `basic` (@tyama501)
- Rename `mined` to `edit` (MINIX visual editor)
- Fix `hostbasic` for compilation on macOS
- Enhance `fsck` to use main memory, works on 65Mb MINIX filesystems

**Networking**
[... 125 lines cut ...]
- Fix limits.h recursion (@tkchia)
- Additional GMP download location (@tkchia)
- Keep directory permissions when building images
- Add `-maout-symtab` to elf2elks to create symbol table in ELKS executables
- Add `--symfile` option to elf2elks to produce .sym symbol table for kernel and applications
- Add `-finstrument-functions-simple` for ELKS function instrumentation (@tkchia,@ghaerr)
- Add `-Wno-prio-ctor-dtor` to remove compiler warnings on new constructors (@tkchia)
- Add ELKS-specific spec file for `ia16-elf-gcc` (@tkchia)
- Force Ubuntu 20.04 for CI runner, allows mtools-4.0.23 support for build
- Fix `sash` overwriting `ash` on 2880k and HD builds
- Fix medium model data segment relocations in elf2elks (@tkchia)

**Documentation**
- Major updates to man pages (@Mellvik)
- Spell check man pages and fix documentation typos (@ccoffing)
- Various C library and command sources reformatted to standard (@ccoffing)
- Updated driver docs (@johnathancn)
- Update parenthesis for better source readability (@marcin-laszewski)
- Create SVG vector version of ELKS logo (@eloydegen)

**ElksEmu**
- Remove any child processes on exit (@tkchia)
- Fix emulation of `execve` system call (@tkchia)

Chief maintainer: @ghaerr

jsedanoj, 2023-07-27:
> - Fix `cron` memory leak, other fixes ***@***.***, @ghaerr)
> - Fix `tinyirc` to operate using ping/pong protocol ***@***.***, @DutchComputerKid)
> - Fix `fsck` out of memory problem, display issues and -r ***@***.***, @ghaerr)
> - Add `__weaken` and `__LINK_SYMBOL` weak symbol support ***@***.***, @ghaerr)

Uau! ELKS is really alive.

A typo:

MK-88 Russian clone of __SOMETHING__MISSING__ using 8088 clone K1810BM88 w/256k RAM (Use ***@***.***

Thanks,

toncho11, 2023-07-27:
Looks very good!

In general I think there should be a freeze period of 2 weeks where everyone is testing on their machines. We can form a group of 5-6 people that we are testing on real hardware. After that the release should be published. This is just an overall strategy proposal. I also advocate for point releases such as 7.1 for example, in case of a major problem (a setting enabled by default instead of disabled by default that leads to crushes, etc).

Me, I will be able to test in September. So do not wait for me :). 

Thank you @ghaerr !

tyama501, 2023-07-27:
Looks great! Thank you.

cocus, 2023-07-27:
Absolutely astounding! Great work team!

toncho11, 2023-07-28:
I managed to test quickly. The fd360-minix boots OK on my Amstrad 1640.

ghaerr, 2023-07-28:
> The fd360-minix boots OK on my Amstrad 1640.

Thanks a lot @toncho11, as I'm hoping the make a cut soon, as the Book 8088 project could use it, since @Vutshi can't currently compile up a system.

The system should remain quite stable, as there haven't been any structural kernel changes for quite some time.

tyama501, 2023-07-28:
Hello @ghaerr ,

I tested fd1232.img of commit f4363cd
https://github.com/jbruchon/elks/actions/runs/5688362692

I checked 
clock, ps, meminfo, basic, nano-X with emulator. 
These are OK.

I also confirmed the serial console is OK with PC-9801BX.

Thank you.

ghaerr, 2023-07-28:
Hello @tyama501,

Thanks for your testing, good to know all the PC-98 specific apps are being built correctly :)

Thank you!

tt1542, 2023-07-28:
@ghaerr, I just tested the current build (FAT floppy, 1.44M) with my Monotech NuXTv2, at least briefly. Things seem to be fine, no complaints!

ghaerr, 2023-07-28:
> I just tested the current build (FAT floppy, 1.44M) with my Monotech NuXTv2, at least briefly. Things seem to be fine, no complaints!

@tt1542, wow, very good to know, thanks for the report! It seems then that our current build is working fine on Book 8088, while the v0.6.0 release does not. It seems that something was fixed after the v0.6.0 release that allows Book 8088 to work (at least on 360k FAT and 64M MINIX), and does not require a BIOS upgrade. 

@Vutshi isn't currently able to compile the latest at this time, so his reports in #1619 are based on the year-old v0.6.0 release. Definitely time for v0.7.0 to get out! We can continue to sort out the 32M HD FAT boot failure on Book 8088 after v0.7.0.

toncho11, 2023-07-28:
Good teamwork 👍 :smiley:

bocke, 2023-07-28:
Thankx for the hard work everyone. I'm waiting for my Book8088 to arrive so I can play with ELKS on a real computer for once.  

I'll have to try installing gcc-ia16 to play with too. I'm interested if some small modern-ish microemacs/vi clones would work on ELKS.

ghaerr, 2023-07-29:
@bocke, we've got a pretty decent `vi` (elvis) ported to ELKS now. It works well and is included in the 1.4M floppy distribution.

tt1542, 2023-07-29:
@ghaerr , please have another look at my comment above. I tested the current build with my other system, the Monotech NuXTv2, and just wanted to confirm that it does boot fine using the 1.44M floppy (NOT 360K).

ccoffing, 2023-07-29:
Looking good!

I just noticed that crontab (which got some fixes this cycle) was not being installed.  I have opened a PR.

tyama501, 2023-07-31:
Hello @ghaerr ,

Is there way to include nxtetris in pc-98 image?

vi shrank a lot so fd1232.img now have 186KB space left.

Thank you @ccoffing !

ghaerr, 2023-07-31:
Hello @tyama501,

> Is there way to include nxtetris in pc-98 image?

Not really easily, since all the Nano-X games are listed in `elkscmd/Applications` as `:other`, and would use up too much space. What we need is another category that perhaps can be specified in a config file that allows for selection of specific applications. Let me think of a way this might be done before v0.7.0 is released, so that `nxtetris` can be added to the PC-98 default image.

Thank you!

tyama501, 2023-07-31:
Thank you @ghaerr ,

nxlandmine, nxworld, and nxterm are now already included since I selected nano-X application in the config.

ghaerr, 2023-07-31:
Hello @tyama501,

> nxlandmine, nxworld, and nxterm are now already included since I selected nano-X application in the config.

Oh, I'm glad you mentioned that, as I was thinking that we needed a method to individually install an application. If we change elkscmd/Applications to include nxtetris on `:nanox` rather than `:other`, everything will fit on the 1440k floppy.

>  so fd1232.img now have 186KB space left.
(and 416KB space for fd1440. Wow!)

That's a lot of free space for both images! You must have a number of options turned off, I haven't looked at the details, as the IBM 1440k free space is much lower. We could go with the above option of changing nxtetris to `:nanox`, which works for the regular IBM PC 1440k distribution, although the free space drops to 38k.

On another matter, perhaps you should consider turning on CONFIG_APP_MAN_PAGES for the PC-98 1440k floppy, they would probably fit?

We could also go with introducing the ability to install a few applications directly by name. Please let me know what you think with some image size testing if possible, and we can make the changes.

Thank you!

tyama501, 2023-08-03:
Hello @ghaerr , 

Yes, I could add MAN Pages to 1440 KB image, but it only 90KB left if I add it.
So I would like to leave the space to add and debug ktcp in ELKS 0.8.

Thank you.

closed 2023-08-03 by ghaerr