ghaerr/elks #1692: VFS can't mount root

cathoderaydude, 2023-09-03:
Just downloaded the current release, 0.7.0. I'm trying to boot on a Toshiba T1100 Plus with DD drives, so I used fd720-fat.img. When I try to boot, I get "VFS: Unable to open root device." Tried with fd720-minix.img as well, same result. Screenshot below is with minix version. 
Have been using this same workflow to produce 720k disks, used in this machine, with no trouble, so I don't think it's a media or image writing issue.

[image]

toncho11, 2023-09-03:
Your system does not say:

`/dev/fd0: 80 cylinders, 2 heads, 9 sectors = 720.0 kb` after network probing. 
This is what it says when you use an 720 kb ELKS image.

ghaerr, 2023-09-03:
Hello @cathoderaydude,

Thanks for the screenshot - I can see what's happening: "device_setup: BIOS drive 0x01, root device 0x3a0" - the system is trying to mount the filesystem on drive B, based on the Toshiba BIOS saying that drive 1 is the boot drive (0x3a0 is /dev/fd1 on ELKS).

I'm not sure why this is happening on the T1100. Is there a config that allows setting the boot drive? I imagine it's working fine for DOS. You can force the filesystem on drive A by adding the line `root=fd0` to the /bootopts file (easiest done on FAT).

Are there two floppies on that system? If so, ELKS would mount the filesystem on drive B if present. 

Thank you!

cathoderaydude, 2023-09-03:
Hey @ghaerr,

Thanks for the quick response! So, this machine does have two FDDs, and I tried booting from both with the same results except that it IDs the device differently. The above screenshot was taken with the disk in drive 2, but if i move it to drive 1, it reports the disk as 0x0, root device 0x380, then fails again. 

I strongly suspect that the Toshiba's FDD or BIOS are doing some shenanigans. This machine is an extremely early laptop, one of the first PC compatibles, and contains a number of ASICs that I think Toshiba created. It's supposed to be 100% compatible, but perhaps that only applies to things that use the BIOS routines as expected. I am partly suspicious of the FDD hardware because this machine has a strange feature: it supports an external 5.25" disk, but to enable it, you have to flip a switch on the side to choose one of the internal disks to replace. At one point while experimenting with this, I booted from disk 1, it came up as B:, but I was then able to access it at A:. So perhaps there's some odd remapping going on.

I have tested adding root=fd0 to bootopts; also tried fd1. I did notice the requirement that bootopts be smaller than 512b and removed all commented lines to make sure it's being parsed with no success. Also tried putting identical disks in both drives in case it's trying to access the other one; no success there either.

Let me know if you have any suggestions, but I suspect this is down to the hardware being not-quite-100% compatible. Thanks again!

ghaerr, 2023-09-03:
> The above screenshot was taken with the disk in drive 2, but if i move it to drive 1, it reports the disk as 0x0, root device 0x380, then fails again.

I see - so it fails when in drive 0, reporting Can't mount 0x380 also. Well, that means the issue isn't the boot device as I had guessed when I first say your screenshot.

In order to proceed further, I would need to know exactly what the error message is when booting from Drive 0, that is "VFS: Unable to open root device 0380"? If so, that likely means the BIOS isn't reporting the drives found properly (as @toncho11 pointed out). That means ELKS has effectively been told there isn't a drive 0 (or drive 1, it seems). To get this to boot, you'd need to build the cross-compiler, so that we can setup a config file with hard-specified mappings for FD0 and FD1, or add some kernel display of the BIOS "get drive info" function return results.

I'm pretty sure we can get the system booting in short order. Please run `tools/build.sh` to build the (lengthy) cross development system and compile an image if you'd like to continue proceeding.

As an alternative, I could probably post an image to try that might give us more information, but we'd still possibly have to have you build the system ultimately in order to get things completely figured out.

cathoderaydude, 2023-09-03:
[image]

I had meant to post this screenshot before, here it is trying to boot on the other drive. I'll see if I can get the compiler set up later when I have some time.

ghaerr, 2023-09-03:
> I'll see if I can get the compiler set up later when I have some time.

Great - it'll take a half hour or so after cloning the source tree, then run `tools/build.sh`, then `make`.

The above screenshot has the same problem, which indicates that ELKS thinks there are no floppy drives. There are a couple possible reasons for this.

I'm in the middle of another enhancement, but I'll include some code in that to display more of what's going on in the ELKS disk driver and post the binary up here for you to try, which should help speed up the process.

toncho11, 2023-09-03:
Yes, @ghaerr if you could provide an image where the availability of fd0 is hardcoded then this might work or at least it will give us a hint.

ghaerr, 2023-09-03:
Ok @toncho11. I think the problem here is related to an issue we saw on your very early IBM 5150 PC - that the PC BIOS didn't implement the "get drive parms" function to return the number of disks or disk types (yet). I fixed that for your system and was going to add additional print statements to display more precisely what is happening. By default, ELKS should be creating at least a single 360k drive in the case that the function(s) aren't implemented, but we're not seeing that either.

ghaerr, 2023-09-03:
@cathoderaydude,

Here's MINIX and FAT 720k boot versions with some additional debug print code as well as an attempt at fixing what the problem may be. Hopefully these will boot on the T1100. Please try when you have time and send a screenshot, thanks!

[fd720.zip](https://github.com/ghaerr/elks/files/12507844/fd720.zip)

cathoderaydude, 2023-09-03:
[image]

Interesting results. I wrote the fat version and it gets this far, then instantly reboots. i had to take video with my phone so I could grab a frame to show you before the screen clears.

ghaerr, 2023-09-03:
Ok @cathoderaydude, we're almost there. The T1100 BIOS is returning 0 drives; I have ELKS "creating" one. As you can see, the cylinders/heads/sectors are quite off... I'll upload another version shortly.

ghaerr, 2023-09-03:
@cathoderaydude, try this, it should boot:

[fd720-2.zip](https://github.com/ghaerr/elks/files/12508089/fd720-2.zip)

cathoderaydude, 2023-09-04:
[image]

Great job!! Came right up. I have to say, this is the nicest experience I've ever had with a bug report, hah. I've never had someone whip up a custom build for my particular broken platform before! 😛

Now that this is running, if there's any info you could use that would help with issues like this in the future, or anything you'd like me to test on this machine just in case it has other hardware/bios quirks, I'd be happy to help.

ghaerr, 2023-09-04:
>  I have to say, this is the nicest experience I've ever had with a bug report

Well, we aim to please over here... and thank you.

(And we also like to know why something isn't working :))

> Great job!! Came right up.

Glad it's working! You'll notice in the "fine print" in your last screenshot that the "equip drives" is 2, which means the BIOS is reporting 2 drives using a very old-fashioned method. Now that we have that information, I can build a version that actually supports two drives (which your latest build does not). The reason ELKS didn't boot on the T1100 in the first place is that we've never seen a system whose BIOS *says* it has a "get drive parms" function but then returns 0 drives and no error. (@toncho11's even earlier IBM 5150 BIOS reports failure of the "get drive parms" function and we handle that case). We're dealing here with an early time period for "PC-compatible" BIOSes and they weren't all "compatible" yet!

I'll post a fix for all of this in a bit, and then it would be nice if you tested it by perhaps mounting a 2nd floppy to test the 2nd drive feature using something like `mount /dev/fd1 /mnt`.

Thank you for the testing!

ghaerr, 2023-09-04:
> I included this specific screenshot just because I thought you might want to see the retry message. 

This has been seen before and is detailed in #1119. It's a completely separate issue and it seems to be only related to some BIOSes (including QEMU). It only happens in certain circumstances, typical when changing I/O from one drive to a second one. So this is very likely not related to your drive heads, but another matter, yet to be solved. The BIOS is returning an error message and ELKS retries, and we've never seen data loss.

Thus, we can consider the current issue closed, thank you! Can you please post a screenshot of the boot screen? I added some debug messages and would like to see them. Thank you again for your quick testing, I'll commit #1694 now as a confirmed fixed for this issue.

ghaerr, 2023-09-04:
>  I then rebooted with the new minix version in drive 1, confirmed rootfs mounted on fd0, and then mounted another disk on fd1 successfully, 

Re-reading your comment from above: it sounds like you booted from drive 1, then mounted stuff on drive 0, then switched floppies and mounted another floppy on drive 1. ELKS doesn't actually support hot-swapping floppies from drives (ever), so that might be the reason the BIOS complained of a retry. In ELKS, the boot drive filesystem must not be changed, regardless of any mount command executed. If ELKS is allowing you to (re)mount another floppy on the boot (root) drive, then that's an error.

However, mounting then unmounting filesystems on a drive that is not the boot/root floppy is fully supported.

cathoderaydude, 2023-09-04:
Oh no, I booted from drive one, then tested mounting disks on drive two. After that I restarted, booted from drive two, then tested mounting disks on drive one. So I never removed the disk that the OS had started from on each boot. I wanted to ensure that ELKS could _boot_ from both drives, and also could _read_ both drives, so I had to switch which drive I'd started from.

[image]

Let me know if that's the info you're looking for, I believe that's from the latest patch.

ghaerr, 2023-09-06:
Hello @cathoderaydude,

> I included this specific screenshot just because I thought you might want to see the retry message.

I managed to duplicate and fix the retry message on QEMU (as also discussed in #1119), but would like to see whether this also fixes the "track read retry" you were seeing on the T1100. Can you give it a test when you have time? If it works, I'll post a PR for it. You will then have the very latest version with no BIOS errors!

[fd720-4.zip](https://github.com/ghaerr/elks/files/12532616/fd720-4.zip)

Thank you!

closed 2023-10-07 by ghaerr