ghaerr/elks #1755: Soviet Computer Electronika MC1502

AirmanKaluga, 2023-10-26:
I have a Soviet computer Elektronika MS1502. ELKS works on it. It has several features. It does not have a DMA controller. And his disk drive controller is built on the KR1818VG93 controller (a copy of Western Digital FD1793). I have a disassembled BIOS of this computer. And he is functional. However, I cannot translate the int13 interrupt into C language in order to make direct access to the disk drive. Several disk drive controllers of Soviet analogues of IBM PC  are built on such a chip. Is there anyone who is ready to help write direct access to the drive for such a controller?

ghaerr, 2023-10-26:
Hello @AirmanKaluga,

I am not entirely sure I understand your request - are you wanting to write an ELKS device driver, similar to the "direct floppy driver" which will access the FDC chip directly from the ELKS kernel? Or are you wanting to use INT 13h from a user application program to access the floppy directly?

If it is the first case, then you will want to look hard at the complicated direct floppy driver in elks/arch/i86/drivers/block/directfd.c, and determine what is different about the KR1818VG93 chip than the NEC 765. Since your system does not have a DMA controller, you'll then have to have a programmed loop in order to read the data. This will be tricky depending on whether the KR chip interrupts to indicate ready to read, etc.

You're welcome to post the disassembled BIOS for INT 13h so we can understand how the BIOS reads/writes the floppy. Are you just trying to rewrite the BIOS INT 13h driver into C code for a driver? This also gets tricky, since the direct floppy driver is interrupt driven, versus the regular ELKS BIOS floppy driver, which is not. They require entirely different programming techniques. The BIOS itself is not interrupt-driven. I can help you with this, but will not be able to do any testing, as I have no KR chip hardware.

If it is the latter case, where you want to write an ELKS application that uses INT 13h, that is not supported by ELKS. Only the kernel can make BIOS calls, as otherwise ELKS, being a timesliced OS, might preempt the CPU while the application is in the middle of a BIOS call, which could crash the system if the kernel also re-entered with another BIOS INT 13h call.

Thank you!

AirmanKaluga, 2023-10-26:
Sorry a have very bad english.

I am not entirely sure I understand your request - are you wanting to write an ELKS device driver, similar to the "direct floppy driver" which will access the FDC chip directly from the ELKS kernel? Or are you wanting to use INT 13h from a user application program to access the floppy directly?

YES, I wanting to write an ELKS device driver, similar to the "direct floppy driver" which will access the FDC chip directly from the ELKS kernel

https://github.com/AirmanKaluga/MC1502_BIOS/blob/master/int13h.asm

ghaerr, 2023-10-26:
> I wanting to write an ELKS device driver, similar to the "direct floppy driver" which will access the FDC chip directly from the ELKS kernel

Do you just want to access the floppy directly, meaning just want to duplicate the INT 13h BIOS code in C, or do you want to implement a driver which uses asynchronous I/O? The latter is very hard, and may not be possible unless the FDC throws an interrupt when it is time to read the sector. And even then, it might not be worth all the trouble without DMA, since the CPU will be tied up reading/writing the floppy data.

The easiest approach would be to look at the C code for the ELKS BIOS INT13h floppy driver, which is synchronous. Those files are bioshd.c and bios.c. The bios.c file contains the only calls to INT 13h using the `call_bios` function. I would recommend you start there, and replace `call_bios` which is an ELKS ASM function that calls INT 13 with C code that access es your FDC directly.

Take a look at that and see what you think. The call_bios source is at elks/arch/i86/lib/bios13.S, but you won't be using that. 

Do you know C and ASM well enough to do the conversion? If you start with the synchronous BIOS version, it should be straightforward. I can help you with how to write specialized C code to access an I/O port or directly access far memory.

ghaerr, 2023-10-26:
It would also be a very good idea to find a data sheet on the KR chip! The ASM disassembly of your BIOS will be lots harder since it doesn't have actual symbol names, thus hiding the meaning and lots of possible guessing.

AirmanKaluga, 2023-10-26:
I can describe the actions of the BIOS. For me this is not a problem. I modified some parts of the BIOS to work better. But I know C very poorly and cannot write a direct access driver for this chip.

ghaerr, 2023-10-26:
Hmm.... I'm not sure how I myself can help directly. We need to find someone near you that knows C well enough to code a driver from your specification of how the KR chip works. It will be too much effort for me to write a driver for a machine that I can't test directly on, with no data sheet. There are likely to be many (small or large) problems in the very beginning that will require close talking between the programmer and you in order to get such a thing working. The C source for the direct floppy driver will be helpful to see how a NEC 765 chip is accessed, and could be used to then build a synchronous (easier) version that interfaces through the `call_bios()` function in bios.c for the ELKS BIOS driver.

That said, there may be others who are willing to jump in and help you get this running. I am happy to help another C programmer get this working.

toncho11, 2023-10-28:
Looks nice. 

[image]

@ghaerr here are some more pictures of Elektronika MS1502 (Электроника МС1502): https://pikabu.ru/story/sovetskaya_ibmpc_yelektronika_ms1502_7172267

@AirmanKaluga So these 2 port on the right can be used to attach 2 floppy drives with an extension cable. They are 720 kb and can be 5.25 or even 3.5?

Vutshi, 2023-10-29:
@AirmanKaluga greetings to a fellow  computer archeologist of the Soviet era machines! ;)

I wonder what is the point of (re)writing the BIOS INT 13 as a direct driver in C?

@ghaerr I also wanted to ask you about the new floppy driver in ELKS. Maybe it is already answered in some wiki note which I missed, then redirect me, please. 

Am I right that the new interrupt-driven driver aims to improve OS responsiveness while the disk hardware operates slowly? I'd like to test it on my Schneider and MK-88. What differences should I watch for compared to the original bioshd driver?

Best

ghaerr, 2023-10-29:
> I  wonder what is the point of (re)writing the BIOS INT 13 as a direct driver in C?

That is a good question. I haven't heard the answer yet as to whether the reason is to get async I/O features, as that'd be about the only reason, unless just for purity's sake, @AirmanKaluga would like a C driver that accesses hardware directly. In either case, one would want to write a synchronous version first, as it'd be much simpler.

> Am I right that the new interrupt-driven driver aims to improve OS responsiveness while the disk hardware operates slowly? 

Yes, exactly. Both read and write operations operate asynchronously, meaning that any application asking for I/O may not have to wait for it. However, in almost all cases, an application will (obviously) have to wait for a read to complete before returning :) Because of this, just sitting at a shell prompt reading executables and disk files won't show much difference at all. One will start to see enhancements only when running multiple tasks in the background and/or producing lots of written data, for which the application will no longer have to wait for write I/O complete.

> What differences should I watch for compared to the original bioshd driver?

Well, the first issue is whether it even boots/runs or not, and then you'll want to mount /dev/df0 or /dev/df1 to open the direct floppy driver. If the Schneider uses something other than a NEC765 compatible chip, it likely won't run at all without modifications.

Go ahead and play around with getting a version built with CONFIG_ASYNCIO and CONFIG_BLK_DEV_FD set. I have a config file that allows for booting from BIOS by default, but activating the async driver by either setting `root=/dev/df0` in /bootopts, or `mount /dev/df1 /mnt`. I have attached it for your convenience. (360k floppy).

[config.df-360.zip](https://github.com/ghaerr/elks/files/13198721/config.df-360.zip)

I'm thinking about making this dual-boot configuration the default.

ghaerr, 2024-02-06:
I'm closing issue for now, can be re-opened should interest be found for writing driver for this machine.

closed 2024-02-06 by ghaerr