ghaerr/elks #1756: panic at boot

sparky4, 2023-11-19:
**Description**
- What is expected ?
- boot to shell
- What is observed ?
- panic

**Configuration**
- Version (commit or release):
- 0.7.1 latest pull 
- Emulator or HW platform:
- a 25mHz 286 AT system

**How to reproduce ?**
- Does the problem always happen ?
- yes
- List the steps to reproduce the problem.
- just boot it

**Raw data**
- Screenshot(s):
- Log file:
give me a sec for this info

**Additional information**
Add any other context information about the problem here.

sparky4, 2023-11-19:
#
# Automatically generated by make menuconfig: don't edit.
#

#
# Kernel & Hardware
#

#
# System
#

CONFIG_ARCH_IBMPC=y
# CONFIG_ARCH_8018X is not set
# CONFIG_ARCH_PC98 is not set
# CONFIG_HW_COMPAQFAST is not set
# CONFIG_HW_MK88 is not set
CONFIG_HW_VGA=y
CONFIG_HW_SERIAL_FIFO=y
CONFIG_APM=y

#
# Kernel settings
#

CONFIG_BOOTOPTS=y
# CONFIG_ASYNCIO is not set
CONFIG_CPU_USAGE=y
# CONFIG_TIME_RTC_LOCALTIME is not set
# CONFIG_TRACE is not set
# CONFIG_TIMER_INT0F is not set
# CONFIG_TIMER_INT1C is not set
# CONFIG_ROMCODE is not set
CONFIG_FARTEXT_KERNEL=y

#
# Networking Support
#

CONFIG_SOCKET=y
# CONFIG_NANO is not set
CONFIG_INET=y
# CONFIG_UNIX is not set

#
# Filesystem Support
#

CONFIG_MINIX_FS=y
# CONFIG_ROMFS_FS is not set
CONFIG_FS_FAT=y
CONFIG_FS_DEV=y
# CONFIG_ROOT_READONLY is not set
# CONFIG_FS_RO is not set
CONFIG_FULL_VFS=y
CONFIG_32BIT_INODES=y
CONFIG_FS_EXTERNAL_BUFFER=y
CONFIG_FS_NR_EXT_BUFFERS=64
# CONFIG_FS_XMS_BUFFER is not set
CONFIG_PIPE=y
CONFIG_EXEC_COMPRESS=y
CONFIG_EXEC_MMODEL=y
CONFIG_EXEC_MMODEL=y

#
# Drivers
#

#
# Block device drivers
#

CONFIG_BLK_DEV_BFD=y
CONFIG_BLK_DEV_FD=y
CONFIG_TRACK_CACHE=y
# CONFIG_BLK_DEV_BFD_HARD is not set
CONFIG_BLK_DEV_BHD=y
CONFIG_IDE_PROBE=y
CONFIG_BLK_DEV_RAM=y
CONFIG_RAMDISK_SEGMENT=0
CONFIG_RAMDISK_SECTORS=128
CONFIG_BLK_DEV_SSD_NONE=y
# CONFIG_BLK_DEV_SSD_TEST is not set
# CONFIG_BLK_DEV_SSD_SD8018X is not set
CONFIG_BLK_DEV_CHAR=y

#
# Character device drivers
#

CONFIG_CONSOLE_DIRECT=y
# CONFIG_CONSOLE_BIOS is not set
# CONFIG_CONSOLE_8018X is not set
# CONFIG_CONSOLE_HEADLESS is not set
CONFIG_KEYBOARD_SCANCODE=y
# CONFIG_CONSOLE_SERIAL is not set
CONFIG_EMUL_ANSI=y
# CONFIG_EMUL_VT52 is not set
# CONFIG_EMUL_NONE is not set
# CONFIG_KEYMAP_BE is not set
# CONFIG_KEYMAP_DE is not set
# CONFIG_KEYMAP_DV is not set
# CONFIG_KEYMAP_ES is not set
# CONFIG_KEYMAP_FR is not set
# CONFIG_KEYMAP_IT is not set
# CONFIG_KEYMAP_SE is not set
# CONFIG_KEYMAP_UK is not set
CONFIG_KEYMAP_US=y
CONFIG_CHAR_DEV_RS=y
# CONFIG_CHAR_DEV_LP is not set
# CONFIG_CHAR_DEV_CGATEXT is not set
CONFIG_CHAR_DEV_MEM=y
# CONFIG_CHAR_DEV_MEM_PORT_READ is not set
# CONFIG_CHAR_DEV_MEM_PORT_WRITE is not set
CONFIG_PSEUDO_TTY=y
# CONFIG_DEV_META is not set

#
# Network device drivers
#

CONFIG_ETH=y
CONFIG_ETH_NE2K=y
CONFIG_ETH_WD=y
CONFIG_ETH_EL3=y

#
# Userland
#

# CONFIG_APPS_BY_IMAGESZ is not set
CONFIG_APP_ASH=y
CONFIG_APP_SASH=y
CONFIG_APP_DISK_UTILS=y
CONFIG_APP_FILE_UTILS=y
CONFIG_APP_SH_UTILS=y
CONFIG_APP_SYS_UTILS=y
CONFIG_APP_ELVIS=y
CONFIG_APP_MINIX1=y
CONFIG_APP_MINIX2=y
CONFIG_APP_MINIX3=y
CONFIG_APP_MISC_UTILS=y
CONFIG_APP_TUI=y
CONFIG_APP_NANOX=y
CONFIG_APP_BASIC=y
CONFIG_APP_SCREEN=y
CONFIG_APP_CRON=y
# CONFIG_APP_OTHER is not set
# CONFIG_APP_TEST is not set
# CONFIG_APP_BUSYELKS is not set
# CONFIG_APP_CGATEXT is not set
CONFIG_APP_KTCP=y
# CONFIG_APP_MAN_PAGES is not set
# CONFIG_SYS_DEFSHELL_SASH is not set
# CONFIG_SYS_NO_BININIT is not set

#
# Target image
#

# CONFIG_IMG_MINIX is not set
CONFIG_IMG_FAT=y
# CONFIG_IMG_RAW is not set
# CONFIG_IMG_ROM is not set
# CONFIG_IMG_FD2880 is not set
CONFIG_IMG_FD1440=y
# CONFIG_IMG_FD1232 is not set
# CONFIG_IMG_FD1200 is not set
# CONFIG_IMG_FD720 is not set
# CONFIG_IMG_FD360 is not set
# CONFIG_IMG_HD is not set
CONFIG_IMG_DEV=y
CONFIG_IMG_BOOT=y
CONFIG_APPS_COMPRESS=y
CONFIG_IMG_EXTRA_IMAGES=y

ghaerr, 2023-11-19:
Hello @sparky4,

Can you by chance post a screenshot of the boot? That would help me determine more as to exactly what is happening.

Are you compiling this from scratch with a custom .config? Perhaps try `cp ibmpc-1440.config .config` and `make clean; make`, or download a binary image from the [v0.7.0 release page](https://github.com/ghaerr/elks/releases/tag/v0.7.0). We can then determine if the reason is something under current development or not.

Thank you!

sparky4, 2023-11-20:
okay will do

sparky4, 2023-11-20:
scratch and custom
[image]

sparky4, 2023-11-20:
it worked with defaults

sparky4, 2023-11-20:
it was the 8.0 dev version

ghaerr, 2023-11-20:
Hello @sparky4,

The boot problem is that, for some reason, your system is trying to use CONFIG_BLK_DEV_FD, the new DF ('Direct to hw floppy support') driver, and the system isn't configured for CONFIG_ASYNCIO ('Use Async I/O in kernel'). This is shown by the line "root device 0400" but only "1 request hdrs".

The boot should default to using the BIOS disk driver (root device 0320) unless /bootopts has been modified with the line `root=df0`. Is that present on your system?

To fix this, either unset CONFIG_BLK_DEV_FD to remove the DF driver, or set CONFIG_ASYNCIO to enable async kernel I/O.

I will likely make a change that disallows including the DF driver when async I/O is not enabled so this isn't possible in the future.

Was there a particular approach you took to your custom system configuration?

Thank you!

sparky4, 2023-11-20:
no not really. and thank you! <3

ghaerr, 2023-11-20:
@sparky4, can you please zip up and post your .config file? I will use that to determine why the system tried to boot /dev/df0 (DF driver) rather than /dev/fd0 (BIOS driver).

Is your BIOS configure to boot from HD rather than floppy? That could possibly be a reason this occured when linked with DF driver, although "BIOS drive 0" indicates the BIOS will boot from floppy. Also, can you check that your /bootopts has the line `root=df0` in it, not commented out?

closed 2023-12-03 by ghaerr