ghaerr/elks #1759: tools/build.sh failed  Undefined symbols for architecture arm64:   "_host_hooks"

anchorz, 2023-12-02:
since I would like to verify ELKS on Apple Silicon I open the issue here:
 
clang: warning: argument unused during compilation: '-no-pie' [-Wunused-command-line-argument]
Undefined symbols for architecture arm64:
  "_host_hooks", referenced from:
      c_common_no_more_pch() in c-pch.o

(https://dev.haiku-os.org/attachment/ticket/17191/apple_silicon.patch)

ghaerr, 2023-12-02:
Hello @anchorz,

Thank you for the bug report on building `ia16-elf-gcc`. The toolchain itself is actually maintained by @tkchia over at https://github.com/tkchia/gcc-ia16; this should probably be reported there. Lets see what @tkchia has to say about it.

I'm not well-versed on gcc internals - have you tried the referenced Apple Silicon patch on this build?

Thank you!

anchorz, 2023-12-02:
Yes, this patch works. Still ELKS has some problem with apple M1 arm64. Is there a chance you could try to build elks on apple hardware?

ghaerr, 2023-12-02:
Unfortunately, I only have x86 apple hardware and don't have way to test or I gladly would. Could you elaborate a bit on the problems you're having? I would like to know whether the toolchain builds ELKS, and/or whether ELKS then won't run on QEMU on Aarch64, or more precisely what errors you are seeing.

closed 2024-12-29 by ghaerr