ghaerr/elks #1763: elf.c:34:18: error: use of undeclared identifier 'LIBELF_ARCH'

anchorz, 2023-12-03:
after make menuconfig; make clean; make

the build process stops at 
cd lib/elftoolchain-0.7.1/libelf;
do gcc -Wall -I/Users/jayne/Development/ws-git/elks/include -I/Users/jayne/Development/ws-git/elks/elks/include -I/Users/jayne/Development/ws-git/elks/elks/tools/elf2elks/lib/elftoolchain-0.7.1/libelf -I/Users/jayne/Development/ws-git/elks/elks/tools/elf2elks/lib/elftoolchain-0.7.1/common -D__KERNEL__ -c -I. -I../common "$src"; done;

elf.c:34:18: error: use of undeclared identifier 'LIBELF_ARCH'
        .libelf_arch            = LIBELF_ARCH,
                                  ^
elf.c:35:22: error: use of undeclared identifier 'LIBELF_BYTEORDER'
        .libelf_byteorder       = LIBELF_BYTEORDER,
                                  ^
elf.c:36:19: error: use of undeclared identifier 'LIBELF_CLASS'
        .libelf_class           = LIBELF_CLASS,
                                  ^

ghaerr, 2023-12-06:
Hello @anchorz,

Thanks for the fix for building `elf2elks` on M1. I'll look into add this as a patch to the build, since the elf2elks source is pulled from a specific external commit rather than being held resident in this repository.

Thank you!

anchorz, 2023-12-15:
can you just simply add the patch in ELKS repository?

ghaerr, 2023-12-15:
> can you just simply add the patch in ELKS repository?

Yes, thank you. I plan to do that.

PR #1800 (2024-02-06): [build] Fix elf2elks libelf build on Apple M1
Fixes #1763.
closed 2024-02-06 by ghaerr