ghaerr/elks #1764: 8018X system: bell-8254.c:24:14: error: ‘SPEAKER_PORT’ undeclared

anchorz, 2023-12-03:
make for 8018x system breaks in:

In file included from bell-8254.c:12:0:
bell-8254.c: In function ‘soundp’:
bell-8254.c:24:14: error: ‘SPEAKER_PORT’ undeclared (first use in this function)
     outb(inb(SPEAKER_PORT) | 0x03, SPEAKER_PORT);

not sure if the make rule is correct or not:
./elks/arch/i86/drivers/char/Makefile 

ifdef CONFIG_CONSOLE_DIRECT
OBJS += console-direct.o bell-8254.o
endif
endif

anchorz, 2023-12-15:
I just select the generic 8018x system. 

may be i am doing something wrong. selecting system 8018x in make menuconfig, does not allow any additional settings besides INTx and RTC.

ghaerr, 2023-12-15:
> selecting system 8018x in make menuconfig, does not allow any additional settings besides INTx and RTC.

Thanks, I'll take a look at this. The 8018X port doesn't allow much configuration, but selecting it should not cause a compilation error.

PR #1799 (2024-02-06): [build][docs] Miscellaneous fixes
Fixes #1789.
Fixes #1764.
Fixes #1787.
Prelim support for PC-98 in beep (https://github.com/ghaerr/elks/pull/1796#issuecomment-1925850107).
closed 2024-02-06 by ghaerr