ghaerr/elks #1770: Cannot compile v0.7.0, undefined reference to "hd_drive_map"

rationalism, 2023-12-08:
Hi! I tried to compile v0.7.0 on my Ubuntu 22.04 laptop, following the instructions here:

https://github.com/ghaerr/elks/blob/master/BUILD.md

When I ran "make -j 16 all", it ran for a few minutes, then crashed with this error:

```
make[2]: Leaving directory '/home/alyssa/elks/elks/arch/i86/drivers/net'
(cd ../.. ; ia16-elf-ld.gold  -M  -T /home/alyssa/elks/elks/elks-small.ld \
	arch/i86/boot/crt0.o \
	init/main.o '-(' fs/fs.a kernel/kernel.a lib/lib.a net/net.a fs/minix/minixfs.a fs/msdos/msdos.a arch/i86/kernel/akernel.a arch/i86/lib/lib86.a arch/i86/mm/mm.a  arch/i86/drivers/char/chr_drv.a arch/i86/drivers/block/blk_drv.a arch/i86/drivers/net/net_drv.a '-)' "`ia16-elf-gcc -ffreestanding -fno-inline -melks -mcmodel=small -msegment-relocation-stuff   -I/home/alyssa/elks/include -I/home/alyssa/elks/elks/include -I/home/alyssa/elks/libc/include -D__KERNEL__ -Os -fno-delete-null-pointer-checks -Wnull-dereference -Wall -W -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wredundant-decls -Wuninitialized -Winline -Wswitch -Wshadow -Wchar-subscripts -Wwrite-strings -Wdiscarded-qualifiers -Wcast-align -Wcast-qual -Wno-sign-compare -Wno-cast-qual -Wno-unused-parameter -Wno-missing-prototypes -Wno-empty-body -print-libgcc-file-name`" \
	-o arch/i86/boot/system > arch/i86/boot/system-full.map ; \
	ia16-elf-nm arch/i86/boot/system | sed -e '/&/d; /!/d' | sort > \
		arch/i86/boot/system.map; \
	elf2elks --symfile arch/i86/boot/system.sym arch/i86/boot/system)
arch/i86/drivers/block/blk_drv.a(genhd.o):function add_partition: error: undefined reference to 'hd_drive_map'
arch/i86/drivers/block/blk_drv.a(genhd.o):function add_partition: error: undefined reference to 'hd_drive_map!'
ia16-elf-nm: 'arch/i86/boot/system': No such file
elf2elks: error: cannot open input file `arch/i86/boot/system': No such file or directory
make[1]: *** [Makefile:131: boot/system] Error 1
```

The string "hd_drive_map" appears in the assembly file `blk_drv.a`, but it doesn't seem to be in the repo anywhere, so I'm not sure where this is coming from. The only Google hit I found was from this old post on the mailing list:

https://www.spinics.net/lists//linux-8086/msg00530.html

but this code doesn't seem to exist anymore, so I'm not sure what's going on. Thanks

ghaerr, 2023-12-09:
Hello @rationalism,

> The string "hd_drive_map" appears in the assembly file blk_drv.a, but it doesn't seem to be in the repo anywhere,

`hd_drive_map` got renamed to `bios_drive_map`, possibly before v0.7.0, but that's what used now for the same name. There are a couple possible reasons for your problem: 1) `make -j 16` isn't tested nor supported, try running `make kclean; make`, which will clean just the kernel build and make again. If the undefined `hd_drive_map` still errors, then I'm pretty certain that that name actually does exist within the v0.7.0 source, and the reason for the build problem is that, for some reason, the BIOS FD/HD driver (CONFIG_BLK_DEV_BFD or CONFIG_BLK_DEV_BHD) is not being defined in your .config file. This will cause `genhd.c` to reference the variable which is only declared in bioshd.c, the BIOS FD/HD driver. I would say this might be a build bug.

Alternatively , please post your .config file so I can look at it. Are you aware of whether your configuration is built with the BIOS disk driver disabled?

Thank you!

rationalism, 2023-12-09:
@ghaerr Thanks! Unfortunately, doing `make kclean; make` did not work, I got the same error. Here is the config file
[config360.txt](https://github.com/ghaerr/elks/files/13622262/config360.txt)

rationalism, 2023-12-09:
@ghaerr Disabling the "BIOS disk driver" option in the config fixes it! (But will that cause hard drives to not work in general?)

rationalism, 2023-12-09:
@ghaerr Yeah, the image boots, but without the driver it can't find the hard drive. (The stock 0.7.0 image at https://github.com/ghaerr/elks/releases does work)

ghaerr, 2023-12-09:
> Disabling the "BIOS disk driver" option in the config fixes it! 

That is strange! Enabling "BIOS disk driver" should fix it, not disabling. Something is off. Your config file looks fine.

> (But will that cause hard drives to not work in general?)

Correct. You'll want it enabled to access floppy or hard drives. Even stranger is how the kernel boots off floppy without the BIOS driver. Please post a boot screenshot, much might be learned.

Perhaps try building everything anew using the "standard" configuration: `cp ibmpc-1440.config .config; make clean; make`.

If the above builds, but you need a 360k floppy rather than 1440k, then do a `cd image; make images`, which will build all images in the images/directory. Use `fd360-{fat,minix}.img` for your boot.

ghaerr, 2024-02-05:
While fixing #1795, I looked further into the posted undefined references, and have concluded this was only a problem in v0.7.0, and not reproducible under v0.8.0. So closing this issue.

closed 2024-02-05 by ghaerr