ghaerr/elks #1772: Incompatibility with XTIDE (?)

cathoderaydude, 2023-12-09:
I recently obtained a device called a "3inONEder", which is a combo XTIDE, sound card, and NE2000 NIC for Toshiba's extremely early PC laptops, and found that it disrupts the ELKS boot process in profound ways. I should note that I'm testing with the same Toshiba T1100+ that had boot issues with the current release version of ELKS, which we resolved in #1692. I have tested with the latest version in that thread (ending -4) and also the one from #1694 (ending -3) and had the same results.

First, here it is working, with the 3inONEder card removed, just the base T1100+ laptop:

[image]

Next, I plug the card in, with a CF card in the slot, and the below happens:

[image]

I should note that I can read this card if I boot from a DOS floppy, and in fact I fdisked, formatted, and sys'ed it from a DOS 3.3 disk, but the machine hangs if I try to boot from it. So, it's possible the card itself is corrupted in some way, and that may be a separate problem, but it seems like an unreadable FS on a secondary storage device shouldn't hang ELKS. Additionally, I don't even see an indication that it detected the HDD, there's no mention of e.g. /dev/hda, so I suspect this has nothing to do with the contents of the CF card itself.

Next, I remove the CF card and attempt to boot again. The following sequence occurs:

[image]

[image]

This is using the minix version. If I use FAT, it gets about this far, but just keeps reporting continuously incrementing track read retries until I power it off.

I found one other issue, #1094, reporting failures with using a 3inONEder, but it's on a T1200 and also using the CF card exclusively, so I don't know how relevant that is. And frankly, given that this is a one-off custom peripheral for a specific and ancient machine, I will totally understand if it's a WONTFIX situation, since I have no idea how you'd debug this.

Happy to help from my end of course, but for all we know the card is defective and simply spitting noise onto the system bus; it's entirely possible, given that this is a handbuilt hobbyist device. I figured I would report it anyway though and see if anything stands out from the logs.

Thanks!

cathoderaydude, 2023-12-09:
Sorry, I had originally titled it after the XTIDE since that seemed like the most likely culprit, before I checked and found other people seem to be using XTIDEs with no trouble.

ghaerr, 2023-12-09:
Hello @cathoderaydude,

IIRC, @vutshi had issues very similar to this when running on an ancient (Russian or German?) machine with an XTIDE BIOS. I will have to look in the closed issues to find it, it was sometime during the summer. The issue(s) were finally resolved after replacing the CF card multiple times until one that finally worked with XTIDE was found, for unknown reasons! The errors turned out to have nothing to do with ELKS, per se.

In general, I think the ELKS BIOS floppy and hard disk driver(s) are working quite well, and unlikely to have bug(s) of this magnitude.

Thank you!

ghaerr, 2023-12-09:
[Here's the issue](https://github.com/ghaerr/elks/issues/1619) https://github.com/ghaerr/elks/issues/1619.

closed 2024-02-06 by ghaerr