ghaerr/elks #1775: fd_type number mismatch for FD1232

tyama501, 2023-12-23:
Hi @ghaerr ,

I have been working to support 720KB FD read/write for PC-98 and noticed one thing.
It seems that bios.c created after rev 0.7.0 has index number mismatch for FD1232.
Is this intended?

Although the sectors and cylinders are overwritten in the probe_floppy,
sector_size is not.
I am not sure why it looks working.

bios.c
#define FD360           0
#define FD1200          1
#define FD720           2
#define FD1440          3
#define FD2880          4       /* QEMU returns 5 */
#define FD2880_DUP      5       /* Dosbox returns 6 */
#define FD1232          6       /* PC/98 only, not returned */ <----------  Number is "6"

struct drive_infot fd_types[] = {   /* AT/PS2 BIOS reported floppy formats*/
    {40,  9, 2, 512, 0},            /* 360k */
    {80, 15, 2, 512, 1},            /* 1.2M */
    {80,  9, 2, 512, 2},            /* 720k */
    {80, 18, 2, 512, 3},            /* 1.4M */
    {80, 36, 2, 512, 5},            /* 2.88M */
    {80, 36, 2, 512, 6},            /* 2.88M */
    {77,  8, 2, 1024,7},            /* 1.232M PC/98 only */     <------------------------- index is 7 for FD1232
};

Thank you.

tyama501, 2023-12-23:
Oops, sorry...

fd_typs has only 6 index.
So may be this is intended.

tyama501, 2023-12-23:
One more thing in bioshd.c, probe_floppy(),
Line 196 - 202
Why is this setting drivep->cylinders duplicating?

                drivep->cylinders =
                        (media == 0xFD)? 40:
#ifdef CONFIG_IMG_FD1232
                        (media == 0xFE)? 77:            /* FD1232 is 77 tracks */
#endif
                                         80;
                drivep->cylinders = (media == 0xFD)? 40: 80;

ghaerr, 2023-12-23:
Hello @tyama501,

> It seems that bios.c created after rev 0.7.0 has index number mismatch for FD1232.
> Is this intended?

No, that's a bug - nice catch! After v0.7.0, the BIOS driver was slightly enhanced to align the `fd_types` indices with the drive type in BL (less 1) actually returned by the BIOS INT 13h AH=8 (Get Disk Parms) function, so the BIOS return value could be directly used to set the drive type. Notice that it was found that QEMU returns 5 for 2.88M floppies, while DosBox returns 6:
```
// this code is correct
#define FD2880          4       /* QEMU returns 5 */
#define FD2880_DUP      5       /* Dosbox returns 6 */
#define FD1232          6       /* PC/98 only, not returned */
```
Use BIOS BL less 1 from INT 13h AH=8 Get Drive Parms:
```
    /* set drive type for floppies*/
    for (drive = 0; drive < ndrives; drive++) {
        /*
         * If type cannot be determined using BIOSHD_DRIVE_PARMS,
         * set drive type to 1.4MM on AT systems, and 360K for XT.
         */
        BD_AX = BIOSHD_DRIVE_PARMS;
        BD_DX = drive;
        BD_ES = BD_DI = BD_SI = 0;      /* guard against BIOS bugs*/
        if (!call_bios(&bdt)) {         /* returns drive type in BL*/
            *drivep = fd_types[(BD_BX & 0xFF) - 1]; // <---- BL less 1 is used to index the fd_types array
        ...
```

The actual problem is in this code, as you partially pointed out:
```
{80, 36, 2, 512, 5}, /* 2.88M */ // should be 4
{80, 36, 2, 512, 6}, /* 2.88M */ // should be 5
{77, 8, 2, 1024,7}, / *1.232M PC/98 only */ <------------------------- index is 7 for FD1232 (should be 6)
```
The only place where `fdtype` member is used is in the `bioshd.c::probe_floppy()` function line 283:
```
*drivep = fd_types[drivep->fdtype];
```
Since this only occurs when the probe fails on the first sector read, its likely to never have been noticed.

You're welcome to submit a PR with the above three lines indices changed, thank you!

ghaerr, 2023-12-23:
Hello @tyama501,

> One more thing in bioshd.c, probe_floppy(),
> Line 196 - 202
> Why is this setting drivep->cylinders duplicating?

Looks like you're correct, its a duplicate... not sure how that happened. Another nice catch!

```
                drivep->cylinders =
                        (media == 0xFD)? 40:
#ifdef CONFIG_IMG_FD1232
                        (media == 0xFE)? 77:            /* FD1232 is 77 tracks */
#endif
                                         80;
                drivep->cylinders = (media == 0xFD)? 40: 80; // <- this line should be removed!!
```
Go ahead and include this in a PR fix.

> I have been working to support 720KB FD read/write for PC-98 and noticed one thing.

Nice.  If possible, `FD720` should be used for PC-98 720k disks, but if they are non-standard, an additional index into `fd_types[]` after FD1232 would also be OK.

Thank you!

tyama501, 2023-12-24:
Isn't this code meaningless if drivep->fdtype and the index of fd_types are same?
*drivep = fd_types[drivep->fdtype];

ghaerr, 2023-12-24:
> Isn't this code meaningless if drivep->fdtype and the index of fd_types are same?
> *drivep = fd_types[drivep->fdtype];

Admittedly this does seem possibly not needed, but since the drivep->sectors (in drive_info[]) etc may have been modified from the ELKS/FAT BPB or other code above, this is supposed to reset the drive_info[] setting to the BIOS default. Are you seeing this would never happen, e.g. since `goto gotgeom` is likely taken for all those cases? This code could be a remnant from much earlier probe code, but protects in the case where sectors/heads/cylinders might be modified by the probe routine.

What do you think about those cases?

tyama501, 2023-12-25:
Hello @ghaerr ,

Oh, I see.
I have missed "*" in the code.
It is setting the data, not address.

And PC-98 code is skipping 1st entry with count = 0,
but the else code is reading with count = 0. If that fails, then resetting the drive_info.

PC-98 need to change device address, so a bit more complicated.
I have added another device address for 720KB and now debugging on the emulator.
At least I could read 720KB on the emulator. (But I think I injected some bug to probe 1440KB)
[image]

Well, I think we can close this topic for now.
I will open another one, if I notice something.

Thank you.

closed 2023-12-27 by tyama501