ghaerr/elks #1790: ftp not working

djohanse, 2024-01-13:
ftp doesn't seem to be working.

Running from terminal on Intel Mac OS.

Built elks with ne2k adapter.  Downloaded from GitHub and built from source.  start elks via ./qemu.sh and run net start.

I can connect to the elks ftp server but the only command that seems to work is cd:

derekjohansen@Dereks-MacBook-Pro ~ % ftp localhost 8021
Connected to localhost.
220 Welcome - ELKS minimal FTP server speaking.
Name (localhost:derekjohansen): root 
root
331 User OK.
Password: 
230 Password OK.
ftp> cd /bin
250 CWD successful.
ftp> cd /invalid
550 No action - directory not found.
ftp> ls
425 Can't open data connection.
503 Unknown command.
503 Bad sequence of commands.
ftp> get yes
425 Can't open data connection.
503 Unknown command.

ghaerr, 2024-01-14:
I'm pretty sure ftp does work on ELKS, but you're probably running into "host forwarding" problems with QEMU, which prohibit loopback network packets from working. The solution to this is a bit complicated due to QEMU, but something like the following is required in `qemu.sh`, uncommenting the last `HOSTFWD=` line:
```
# Host forwarding for networking
# No forwarding: only outgoing from ELKS to host
# HOSTFWD="-net user"
# Incoming telnet forwarding: example: connect to ELKS with telnet localhost 2323
# HOSTFWD="-net user,hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23"
# Incoming http forwarding: example: connect to ELKS httpd with 'http://localhost:8080'
# HOSTFWD="-net user,hostfwd=tcp:127.0.0.1:8080-10.0.2.15:80"
#HOSTFWD="-net user,hostfwd=tcp:127.0.0.1:8080-10.0.2.15:80,hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23"

# Simultaneous telnet, http and ftp forwarding
FWD="\
hostfwd=tcp:127.0.0.1:8080-10.0.2.15:80,\
hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23,\
hostfwd=tcp::8020-:20,\
hostfwd=tcp::8021-:21,\
hostfwd=tcp::8041-:49821,\
hostfwd=tcp::8042-:49822,\
hostfwd=tcp::8043-:49823,\
hostfwd=tcp::8044-:49824,\
hostfwd=tcp::8045-:49825,\
hostfwd=tcp::8046-:49826,\
hostfwd=tcp::8047-:49827,\
hostfwd=tcp::8048-:49828,\
hostfwd=tcp::8049-:49829"
```

ghaerr, 2024-01-14:
> Update: Think I figured this out. Had to start the ftpd daemon with the ftpd -q flag for QEMU. Seems to be working great now.

OK, didn't see this before I replied. Did you end up using a `hostfwd=` line in qemu.sh?

> But this -q flag should be documented somewhere or configurable. The average user probably isn't as hardheaded as me to pore through the ftpd code and figure this one out.

Completely agree. You're welcome to edit the wiki and add this. In the meantime, I've made a note of it, thank you!

djohanse, 2024-01-14:
From what I can tell the hostfwd isn't the right place.  I think it needs to be part of net start or a standalone command.  I don't think hostfwd has any affect on starting the daemons.

That being said I'm still in the ham-fisting my way around so I could very easily be wrong.  

Also if anyone better at Wiki pages can tell me how to fix that font in my comment above?  It's pretty obnoxious :-)

ghaerr, 2024-02-06:
> But this -q flag should be documented somewhere or configurable. 

I was about to update the wiki when I noticed that running `ftpd -q` is already documented as required for QEMU, thanks.

closed 2024-02-06 by ghaerr