ghaerr/elks #1797: Not an issue, but i am willing to donate the intel etherexpress 8/16 network card

sparky4, 2024-02-04:
i am willing to donate a card straight from ebay ^^
i just need an address.
Please contact me with address via email to sparky4@johnreedcenter.net
ill pay for everything

ghaerr, 2024-02-04:
Hello @Mellvik, are you interested in this? I realize TLVC isn't totally compatible with ELKS but it would be nice to work towards driver compatibility between the two (and I have some ideas on that), and you're the network/NIC expert. I haven't yet been able to get a NIC working in my desktop 386 or I'd take this on myself.

Mellvik, 2024-02-04:
Thanks for bringing this to my attention, @ghaerr. 
Not knowing anything about the Intel NIC, I'll have to come back to the actual offer after looking into it and evaluating the effort. Insatiable curiosity is a blessing and a curse, and I'll admit it sounds tempting.
As to NIC driver compatibility between ELKS and TLVC, I don't know what has changed in ELKS since the fork. If it's minimal, it should be 'low threshold'. 
Getting the 386 desktop networked - contact me out-of-band.

-M

sparky4, 2024-02-04:
here is the source code of the packet driver let me test it http://crynwr.com/drivers/exp16118.zip

ccoffing, 2024-02-20:
FWIW I also have several Intel EE 8/16 sitting around that I want to use in ELKS.  I am no networking guru, but if this effort gets started, I can help test.

Mellvik, 2024-02-20:
Thanks Chuck, that will be very useful.

The ‘project’ will commence when the NIC arrives from sparky4 - some time in March.
FYI the development will be on [TLVC](https://github.com/Mellvik/TLVC), not ELKS, but the result will presumably be a simple drop-in for ELKS.

sparky4, 2024-03-22:
ill send it out today

sparky4, 2024-04-12:
did ya get the part?

Mellvik, 2024-04-12:
Hi @sparky4 - I'm following the tracker,
2 weeks in florida, seems to be on its way now... will letyoy know when it arrives.

closed 2024-11-16 by ghaerr