ghaerr/elks #1868: Bulid errors

tyama501, 2024-04-12:
Hello @ghaerr ,

It seems that I got build errors after I pulled the latest updates.

Here is the build errors with current pc98-1232.config.
Something with libc/debug/syms.c
[image]

I also tried
CONFIG_APPS_BY_IMAGESZ=y
CONFIG_APPS_1232K=y
(I couldn't select CONFIG_APPS_1232K so I wrote it manually, although I haven't decided to use CONFIG_APPS_BY_IMAGESZ. 
More interested in CONFIG_APPS_COMPRESS.)
It gets same error.

Thank you.

ghaerr, 2024-04-12:
Hi @tyama501,

Thanks for the testing. I see from your screenshot that the error is trying to run `gcc` when compiling and building `nm86` in elkscmd/debug/Makefile. This is occurring because the system is trying to build a "host" version (instead of an ELKS version) of `nm86` which isn't required for the build, but is helpful to display our new symbol table lists from the desktop development system.

You're running Free86 right? The problem should go away by changing line 27 of elkscmd/debug/Makefile to the following:
```
#HOSTPRGS = nm86 hostdisasm
```
The bigger problem is that I assume that there is a `gcc` available on the host system, which is specified in the elkscmd/Make.defs:
```
#
# Compiler variables for programs to be compiled as host applications.
HOSTCC = gcc    # <--- change this to your host C compiler filename
HOSTCFLAGS = -O3
```

In the meantime, let me think about what to do about whether ELKS should try building host executables or not.

Thank you!

ghaerr, 2024-04-12:
> I couldn't select CONFIG_APPS_1232K so I wrote it manually

Oh, sounds like that needs to be added to image/config.in?

> although I haven't decided to use CONFIG_APPS_BY_IMAGESZ.

It would be interesting to test CONFIG_APPS_BY_IMAGESZ, rather than the application categories built using CONFIG_APP_xxx. The reason for this is that the automated distribution process for IBM PC was building
different images than those users who selected the default settings in `make menuconfig` as a result,
there were unintended differences in the disk image produced, which is probably not a good idea.

Currently, the automated image build for PC-98 uses `pc98-1232.config`, so this won't be an issue.
But it may be a good idea to have PC-98 match up with the IBM PC CONFIG_APPS_1200 so that the
applications specified with `:1200k` in elkscmd/Applications also work with PC-98. This will only get
more important as we move to specifying additional compressed executables to be added using `:1200c`.
Neither of these options are implemented yet.

> More interested in CONFIG_APPS_COMPRESS

Yes, things should work well, with the exception of the differences between 1232k IBM PC 1200k as 
described above. After hearing your report, we can add `:1200k` and `:1200c` and see how you 
like the image produced for PC-98.

ghaerr, 2024-04-13:
> The problem should go away by changing line 27 of elkscmd/debug/Makefile

I posted #1869 which should fix this issue without you having to do anything, thanks.

tyama501, 2024-04-13:
Thank you @ghaerr 
I will try it later.

One concern about the IMGZ is that applications I have been introducing in my Japanese website like nxlandmine and nxworld are not included.

There may also be application that need only for PC98 or IBM.

Currently we cannot add/remove extra application if we select the IMGZ in the menu config right?

ghaerr, 2024-04-13:
> One concern about the IMGZ

Are you referring to "Select Applications by Image Size"? I am unsure what IMGZ is.

> I have been introducing in my Japanese website like nxlandmine and nxworld are not included.

Are you saying that we need nxlandmine and nxworld on the 1232k PC-98 disk image? That could be done by adding a new `:1232k` (or `:1232c`) tag in elkscmd/Applications that pulls them both in for PC-98 builds, whenever CONFIG_APPS_BY_IMAGESZ is set.

> There may also be application that need only for PC98 or IBM.

The above tags would solve that for PC98 1232k. If needed separately for PC98 1440k, another special tag (`:1440-98`?) could be created.

> Currently we cannot add/remove extra application if we select the IMGZ in the menu config right?

No, there is no mechanism to add/remove specific applications using menuconfig. But we can make changes with tags, or perhaps use a new CONFIG_APP_PC98 (or just CONFIG_ARCH_PC98) to select various applications from elkscmd/Applications. This would be specially coded in elkscmd/Make.install.

tyama501, 2024-04-13:
Sounds great. I understand.
Thanks

ghaerr, 2024-04-13:
Let me know what you think is needed, I'd be glad to add the tags, or if you like you can take a pass at it yourself.

tyama501, 2024-04-13:
Hello @ghaerr ,

I could build image.

Maybe we can have separate 
elkscmd/Applications for each architecture ifdef is not desired.

I made a discussion for CONFIG_APPS_xxx and CONFIG_APPS_COMPRESS
https://github.com/ghaerr/elks/discussions/1870

You can close this issue.

PR #1869 (2024-04-13): [build] Don't automatically build host programs
Should fix #1868. To build host programs if desired, use `make host` within appropriate elkscmd/ subdirectories.
closed 2024-04-13 by ghaerr