ghaerr/elks #1884: serial 2 ps/2 mouse adapter mouse acts funny

sparky4, 2024-04-18:
**Description**
the mouse to work correctly
the mouse goes diagonally when moving up and down. It is quite laggy and it works perfectly in DOS

**Configuration**
latest
286 with serial 2 ps/2 mouse adapter mouse system

**How to reproduce ?**
- Does the problem always happen ?
- yes when using nx programs
- List the steps to reproduce the problem.
- get an adapter ps/2 to serial adapter and try it yourself. i have one and willing to test it

**Additional information**
i am willing to help fix this issue with nx system

ghaerr, 2024-04-18:
Hi @sparky4,

Is your mouse outputting Microsoft protocol, or perhaps pc/Logitech? This can be better determined by running the `mouse` program from the command line, which will display the x,y coordinates. You can also edit the mouse protocol in elkscmd/sys_utils/mouse.c (set MOUSE_MICROSOFT vs MOUSE_PC) to see if that changes things. If so, then this can be changed in Nano-X as well.

Thank you!

sparky4, 2024-04-18:
thanks ill try it out

sparky4, 2024-04-18:
[image]

ghaerr, 2024-04-18:
Did you try replacing the code in elkscmd/sys_utils/mouse.c to the following:
```
#define MOUSE_MICROSOFT     0       /* microsoft mouse*/
#define MOUSE_PC            1       /* pc/logitech mouse*/
```
Then move the mouse quite slowly, We're trying to determine exactly what is coming from your mouse so that the data is parsed correctly. On ELKS, the mouse data is parsed by the user program (Nano-X), not the kernel. If the above change doesn't do much, then we will have to possibly try to see the raw data itself and determine how to parse it.

sparky4, 2024-04-18:
okie ill boot it up

sparky4, 2024-04-19:
lets see the raw data?

ghaerr, 2024-04-19:
You’ll have to modify mouse.c to do that.

sparky4, 2024-04-19:
oh okay ill look into it when i get a chance

ghaerr, 2024-04-20:
@sparky4: I added a raw mode output to `mouse` in #1888 to help you. Run `mouse r` to see raw mouse data, then take a screenshot after lightly moving the mouse left/right and up/down a bit.

sparky4, 2024-04-20:
thanks ill check it out tomarrow

sparky4, 2024-04-22:
dang i need some time to test it more

ghaerr, 2024-11-16:
Closing due to inactivity.

closed 2024-11-16 by ghaerr