ghaerr/elks #1951: Elksemu support for OS/2 NE

rafael2k, 2024-08-03:
Hi,
Just reporting that when I run:
# elksemu/elksemu elkscmd/basic/basic.os2

In order to run the basic compiled with OW, I got:
Not a elks binary.

ghaerr, 2024-08-03:
> I got: Not a elks binary.

`elksemu` runs on Linux OS only and uses a specialized binfmt_misc interpreter in order to load and relocate executables and then run them directly in the active process space, with additional code for calling out to the Linux host for emulator specific system calls. Getting this running is akin to getting Linux itself to support loading OS/2 executables, as the ELKS kernel is not running when `elksemu` is used.

As a macOS developer, I can't run, test or develop any changes to `elksemu`, as it is non-portable and Linux-only, it has been infrequently maintained. I recommend using a number of other OS-portable emulators for testing, including QEMU, DosBoxX, VirtualBox, or specialized emulators like emu86 or blink16. These emulators are CPU, rather than ELKS user mode emulators.

As an emulator, `elksemu` is outdated and has not been kept up with a few new system calls, so it won't currently run all ELKS a.out executables. 

Taking a quick look through the source, it would be a fairly large effort to get `elksemu` to load and emulate the multisegment OS/2 NE format, which will likely not get done. Thus, I will probably close this issue as not planned.

rafael2k, 2024-08-03:
I tested basic Elks a.out in elksemu from current build and it worked. So for now I think I can just convert the binary using os2elks converter, right?
Btw, do you have a list not yet supported syscalls in elksemu? I really love to be able to run 16-bits executable natively on Linux. 
Embeddable Linux Kernel Subset : ))

ghaerr, 2024-08-03:
>  So for now I think I can just convert the binary using os2elks converter, right?

Yes and no. For executables that have only one code and one data segment, yes. But for many large/compact/medium model executables produced by OWC, more than one data (or code) segment is needed, and the a.out format can't support it. Thus, os2elks can't support the conversion either.

For this reason, the os2elks converter is somewhat deprecated, as the preferred path going forward for OWC support is to compile directly to OS/2 NE format and load and execute them directly using kernel CONFIG_EXEC_OS2 configured.

rafael2k, 2024-08-03:
Got it. Anyway, I'll keep using it, and if something breaks, I write here. Thanks @ghaerr!
: )

closed 2024-08-03 by rafael2k

rafael2k, 2024-08-03:
Btw, I can see @ccoffing recently made some fixes to elksemu. So definitely there are users and developers using and maintaining it. The code is not large, I'm reading it now. So cool!

https://github.com/ghaerr/elks/blob/master/elksemu/elks_sys.c
: ))))