ghaerr/elks #1952: Can not boot no HD image with supplied qemu.sh script

rafael2k, 2024-08-03:
While I can boot floppy disks (with -fda or the long options), with any harddrive image, kernel freezes at 
"bioshd: hda BIOS CHS 129,16,63".

Is this issue related to this?
https://github.com/ghaerr/elks/issues/348

rafael2k, 2024-08-03:
Also this issue is interesting:
https://github.com/ghaerr/elks/issues/199#

Btw, trying the image/hd32mbr-minix.img I get stuck at:
bioshd: hda BIOS CHS 62,16,63
(and freezes here)

rafael2k, 2024-08-03:
[image]

ghaerr, 2024-08-04:
Thanks @rafael2k, I've repeated the problem over here but haven't had time to look into it yet - something seems broken indeed. Will advise shortly.

rafael2k, 2024-08-04:
Thanks @ghaerr!

ghaerr, 2024-08-04:
Thanks for testing @rafael2k!! 

This was totally broken likely due to some header file changes made previously. All HD boots were affected when CONFIG_IDE_PROBE (default ON) was set, but they're fixed now. The reasons are unrelated to the couple previous issues you mentioned.

closed 2024-08-04 by ghaerr

rafael2k, 2024-08-04:
That was fast!
: )
Working now.

PR #1954 (2024-08-04): [kernel] Fix kernel crash calling get_ide_data
Fixes #1952.

Fixes crash during HD init when calling `get_ide_data`, due to improperly configured header file that missed far call INITPROC declaration of get_ide_data.