ghaerr/elks #1968: picocom for elks?

RNRetailer, 2024-08-20:
Hi,

Has anyone made picocom for elks?

I tried to port it but the compilation step is failing.

Here is my repo:
https://github.com/RNRetailer/picocom-elks/

Thanks!

ghaerr, 2024-09-03:
Sorry for my late response @RNRetailer, I missed this issue and it must have gotten lost in the shuffle.

Have you tried running `miniterm`, this is very similar and currently running on ELKS.

> I tried to port it but the compilation step is failing.

Please post your compilation output so that we can help, otherwise we'd just be guessing as to a problem.

Thank you!

RNRetailer, 2024-09-03:
ELKS miniterm didn't seem to have a good way to transfer files to and from the remote server.
I only saw options to log into a serial console.

I disabled linenoise support and tried to compile picocom.

Here is the resulting output:

```
make --always-make
Makefile:117: warning: overriding recipe for target '.c.o'
Makefile:54: warning: ignoring old recipe for target '.c.o'
../prefix/bin/ia16-elf-gcc -ffreestanding -fno-inline -mtune=i8086 -mno-segment-relocation-stuff -mcmodel=small -I/include -I/libc/include -I/elks/include -I../prefix/include -D__LIBC__ -Wall -Os  -Wall -g -DVERSION_STR=\"3.1\" -DTTY_Q_SZ=0 -DHIGH_BAUD -DUSE_FLOCK -o picocom.o -c picocom.c
cc1: warning: target system does not support debug output
picocom.c: In function ‘cleanup’:
picocom.c:741:13: warning: implicit declaration of function ‘flock’ [-Wimplicit-function-declaration]
             flock(tty_fd, LOCK_UN);
             ^~~~~
picocom.c:743:13: warning: implicit declaration of function ‘close’ [-Wimplicit-function-declaration]
             close(tty_fd);
             ^~~~~
picocom.c: In function ‘establish_child_signal_handlers’:
picocom.c:1128:44: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     dfl_action._signal_handlers.sa_handler = SIG_DFL;
                                            ^
picocom.c:1129:5: warning: implicit declaration of function ‘sigemptyset’ [-Wimplicit-function-declaration]
     sigemptyset (&dfl_action.sa_mask);
     ^~~~~~~~~~~
picocom.c:1132:5: warning: implicit declaration of function ‘sigaction’ [-Wimplicit-function-declaration]
     sigaction (SIGINT, &dfl_action, NULL);
     ^~~~~~~~~
picocom.c: In function ‘run_cmd’:
picocom.c:1150:5: warning: implicit declaration of function ‘sigaddset’ [-Wimplicit-function-declaration]
     sigaddset(&sigm, SIGTERM);
     ^~~~~~~~~
picocom.c:1152:5: warning: implicit declaration of function ‘sigprocmask’ [-Wimplicit-function-declaration]
     sigprocmask(SIG_BLOCK, &sigm, &sigm_old);
     ^~~~~~~~~~~
picocom.c:1154:11: warning: implicit declaration of function ‘fork’ [-Wimplicit-function-declaration]
     pid = fork();
           ^~~~
In file included from picocom.c:39:0:
../prefix/include/sys/wait.h:57:28: warning: implicit declaration of function ‘__WIFEXITED’ [-Wimplicit-function-declaration]
 # define WIFEXITED(status) __WIFEXITED (status)
                            ^
picocom.c:1174:14: note: in expansion of macro ‘WIFEXITED’
         if ( WIFEXITED(status) ) {
              ^~~~~~~~~
../prefix/include/sys/wait.h:54:30: warning: implicit declaration of function ‘__WEXITSTATUS’ [-Wimplicit-function-declaration]
 # define WEXITSTATUS(status) __WEXITSTATUS (status)
                              ^
picocom.c:1176:23: note: in expansion of macro ‘WEXITSTATUS’
                       WEXITSTATUS(status));
                       ^~~~~~~~~~~
../prefix/include/sys/wait.h:58:30: warning: implicit declaration of function ‘__WIFSIGNALED’ [-Wimplicit-function-declaration]
 # define WIFSIGNALED(status) __WIFSIGNALED (status)
                              ^
picocom.c:1178:21: note: in expansion of macro ‘WIFSIGNALED’
         } else if ( WIFSIGNALED(status) ) {
                     ^~~~~~~~~~~
../prefix/include/sys/wait.h:55:27: warning: implicit declaration of function ‘__WTERMSIG’ [-Wimplicit-function-declaration]
 # define WTERMSIG(status) __WTERMSIG (status)
                           ^
picocom.c:1180:23: note: in expansion of macro ‘WTERMSIG’
                       WTERMSIG(status));
                       ^~~~~~~~
picocom.c:1199:14: warning: implicit declaration of function ‘fcntl’ [-Wimplicit-function-declaration]
         fl = fcntl(fd, F_GETFL);
              ^~~~~
picocom.c:1205:9: warning: implicit declaration of function ‘dup2’ [-Wimplicit-function-declaration]
         dup2(fd, STI);
[... 128 lines cut ...]
             ^~~~~~~~~~~
../prefix/bin/ia16-elf-gcc -ffreestanding -fno-inline -mtune=i8086 -mno-segment-relocation-stuff -mcmodel=small -I/include -I/libc/include -I/elks/include -I../prefix/include -D__LIBC__ -Wall -Os  -Wall -g -DVERSION_STR=\"3.1\" -DTTY_Q_SZ=0 -DHIGH_BAUD -DUSE_FLOCK -o fdio.o -c fdio.c
cc1: warning: target system does not support debug output
fdio.c: In function ‘writen_ni’:
fdio.c:48:18: warning: implicit declaration of function ‘write’ [-Wimplicit-function-declaration]
             nw = write(fd, p, nl);
                  ^~~~~
fdio.c: In function ‘fd_readline’:
fdio.c:128:13: warning: implicit declaration of function ‘read’ [-Wimplicit-function-declaration]
         r = read(fdi, &c, 1);
             ^~~~
../prefix/bin/ia16-elf-gcc -ffreestanding -fno-inline -mtune=i8086 -mno-segment-relocation-stuff -mcmodel=small -I/include -I/libc/include -I/elks/include -I../prefix/include -D__LIBC__ -Wall -Os  -Wall -g -DVERSION_STR=\"3.1\" -DTTY_Q_SZ=0 -DHIGH_BAUD -DUSE_FLOCK -o split.o -c split.c
cc1: warning: target system does not support debug output
../prefix/bin/ia16-elf-gcc -ffreestanding -fno-inline -mtune=i8086 -mno-segment-relocation-stuff -mcmodel=small -I/include -I/libc/include -I/elks/include -I../prefix/include -D__LIBC__ -Wall -Os  -Wall -g -DVERSION_STR=\"3.1\" -DTTY_Q_SZ=0 -DHIGH_BAUD -DUSE_FLOCK -o termios2.o -c termios2.c
cc1: warning: target system does not support debug output
../prefix/bin/ia16-elf-gcc -ffreestanding -fno-inline -mtune=i8086 -mno-segment-relocation-stuff -mcmodel=small -I/include -I/libc/include -I/elks/include -I../prefix/include -D__LIBC__ -Wall -Os  -Wall - [... cut ...]
/home/RNRetailer/Music/elks/picocomgithub/picocom-elks-http/prefix/bin/../lib/gcc/ia16-elf/6.3.0/../../../../ia16-elf/bin/ld: cannot find -l:crtbegin.o: No such file or directory
/home/RNRetailer/Music/elks/picocomgithub/picocom-elks-http/prefix/bin/../lib/gcc/ia16-elf/6.3.0/../../../../ia16-elf/bin/ld: cannot find -l:crtend.o: No such file or directory
/home/RNRetailer/Music/elks/picocomgithub/picocom-elks-http/prefix/bin/../lib/gcc/ia16-elf/6.3.0/../../../../ia16-elf/bin/ld: cannot find -l:dos-t-c0.o: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:107: picocom] Error 1

```

ghaerr, 2024-09-04:
Most of the compilation warnings can be readily fixed by including various required header files, like <unistd.h>, <sys/wait.h>, etc. The functions can be grepped for in libc/include/{*,*/*.h}.

ELKS doesn't support `sigaction` so those routines must be rewritten using the only `signal`, which usually isn't a big deal if the signal is just trying to be caught or ignored.

The termios functions  `tcflush`, `cfsetospeed`, `cfmakeraw` etc are wrapper functions that are not (yet) implemented in ELKS, but a direct `ioctl` using the appropriate sub function will work - see elkscmd/misc_utils/miniterm.c for examples.

Looking at the baud rate value overflows it is evident that picocom.c thinks it should be compiled in 32-bit, there may be other non-portable to 16-bit portions of the program, so its guaranteed it will actually run until the source is more closely looked at. I would think it could be ported though.

closed 2024-11-16 by ghaerr