ghaerr/elks #1979: Problem with UMB

toncho11, 2024-09-02:
https://github.com/ghaerr/elks/wiki/Access-to-UMB-memory

* It is not clear the syntax value1:value2 corresponds from value1 to value1+value2 or it corresponds to from value1 to value2
* the default range does not seem correct - at C000 we have "VGA BIOS ROM"

I already tried 0xC000:0x800 and disabled the execution of graphics applications in an emulator, so there is evidence that something is wrong. It should start at C800 according to the screenshot in the wiki.

NathanGibbs3, 2024-09-02:
The format is
Start address:Offset expressed in hexadecimal format.
For UMB to work RAM must be present at the specified address, continue to the specified offset, and not conflict with any installed adapters / expansion cards / peripherals Integrated into the motherboard.
Early PC compatible systems are not equipped with Plug n Play, we have to figure it out ourselves.

tyama501, 2024-09-02:
I think this is the original pull request.
https://github.com/ghaerr/elks/pull/1590

> umb=0xC000:0x800,0xD000:0x1000 for example will allow ELKS to use memory situated from 0xC0000 to 0xC8000 and from 0xD0000 to 0xE0000.

tyama501, 2024-09-02:
If IBM-PC has VGA BIOS on C000 then it should start at C800 if there is RAM on C800.
(Not for PC-98)

toncho11, 2024-09-02:
Thanks. I was beginning to realize that is just some example.

So 0xC000:0x800 means 2096 bytes starting at segment 0xC000? So it reserves only 2kbs starting 0xC000 segment?
It also means that the first value is restricted to  0xA000, 0xB000, 0xC000, 0xD000, xF000?

ghaerr, 2024-09-02:
> and the second value is the length, in bytes.

Oops! The second value (length) is in paragraphs which is bytes >> 4. The internal code adds the second value to the first to get the ending segment value, because that happens to be the way the internal memory mana.ger works.

> So 0xC000:0x800 means 2096 bytes starting at segment 0xC000? So it reserves only 2kbs starting 0xC000 segment? 

Actually I think you meant 4096 bytes above. But, NO, I was incorrect. The 0x800 means 0x8000 bytes, which is 32k. And the second UMB block in the /bootopts example is `0xD000:0x1000` which means add 64K bytes UMB memory starting at segment D000.

> It also means that the first value is restricted to 0xA000, 0xB000, 0xC000, 0xD000, xF000?

Yes. While not actually restricted, the other memory is likely already used and grabbed by the system, which makes BIOS call INT 12h  that returns the number of K-bytes of RAM in the system. For normal systems, this usually returns 640, which means 640K, which is up *until* segment A000. The display adaptor RAM is typically between A000 and BFFF, but not always.

toncho11, 2024-09-02:
The second value 0x1000 is 4096 in decimal. This value is paragraphs and internally multiplied by 16 to 65536 in bytes. 
So you need decide how much bytes you need to reserve and divide this value by 16 in order to produce the paragraphs you need to set as value2.

toncho11, 2024-09-02:
OK. I updated the wiki.

closed 2024-09-04 by toncho11