ghaerr/elks #1994: Tandy 1000HX Nano-X apps not work

astrobleem, 2024-09-04:
**Description**
- What is expected ? launching any nx* binary to i change graphics mode and display the payload
- What is observed ? 
nothing really happens. it just sits there. I am able to hit control + C and return to shell.

**Configuration**
- Version (commit or release): 0.7.0
- Emulator or HW platform: Using x86box configured as a Tandy 1000HX with NEC V20 CPU. 

**How to reproduce ?**
- Does the problem always happen ? yes, whenever the machine is configured as a Tandy. It works when configured as an XT with EGA. I acknowledge these are two completely different display systems.
- List the steps to reproduce the problem.

**Raw data**
- Screenshot(s):
- 
[image]

- Log file:

**Additional information**
is this Nano-X possible with TGA?

toncho11, 2024-09-04:
I think not. I think there are VGA, CGA and PC98. For the CGA you need to also recompile the apps with the CGA driver enabled (instead of the default VGA).

astrobleem, 2024-09-05:
http://ftp.oldskool.org/pub/tvdog/tandy1000/documents/egat1k.txt
http://www.oldskool.org/guides/tvdog/1kfaq.html#IV.D
contains some information which might be useful. 
"The Tandy BIOS fails to set the Video Configuration Code byte at 0040:008A. The default value of this byte is 3Fh. The correct values are: 0Bh, for a single VGA color card; 0Ch, for a dual-monitor system with VGA color and MDA mono; 0Dh, for a single VGA mono board; and 0Eh, for a dual-monitor system with mono VGA and MDA."
[image]

ghaerr, 2024-09-05:
> The Tandy BIOS fails to set the Video Configuration Code byte at 0040:008A. 

Neither ELKS nor Nano-X use the BIOS byte you mention, so that is not the issue. I think the problem is that the Tandy 1000HX display hardware is not CGA, EGA or VGA compatible in graphics modes, is that correct? That is, other graphics programs (e.g. DOS games) don't run on the Tandy, or do they? 

> For the CGA you need to also recompile the apps with the CGA driver enabled (instead of the default VGA). 

Nano-X requires a specific graphics display driver to be compiled and linked into its application. We have EGA and CGA compatible drivers only. The EGA driver will also work on VGA but uses only EGA hardware features.

ghaerr, 2024-09-05:
If it is the case that the Tandy is not EGA compatible, you could write a new driver if you're willing. Or perhaps modify the EGA driver to work. The first step would be to find tech specs and hopefully programming manual on the display hardware.

toncho11, 2024-09-05:
Here is the discussion on the CGA driver.
And in another thread and specifically this reply: https://github.com/ghaerr/elks/discussions/1940#discussioncomment-10255343 is what you need! A CGA compiled nxtetris that will gives us clues on how compatible is Tandy with CGA.

tyama501, 2024-09-05:
Hi @astrobleem 

Yes, you can try the nxteric that is posted on 
https://github.com/ghaerr/elks/discussions/1940#discussioncomment-10255343
as @toncho11 mentioned.

It uses Int10 BIOS for mode setting so it may not work if that is not supported on the system.

Thank you,

closed 2024-11-16 by ghaerr