ghaerr/elks #2035: Draft Release - ELKS v0.8.0

ghaerr, 2024-09-23:
It's been over a year since ELKS v0.7.0 was released on August 3, 2023.
Following are draft release notes for v0.8.0, please comment if something or someone has been left out.
Thank you to all that have contributed and are using ELKS!

***Draft Release - ELKS v0.8.0***

**ELKS v0.8.0 Downloads**

Prebuilt disk images for floppy and hard disks are downloadable below from Github Assets for this release.
These images can be used to boot from PC/XT/AT or PC-98 hardware, or emulators, such as QEMU or dosbox-x.
The images are built with network support by default, supporting NE2K, WD and 3Com NICs.
See the Wiki HowTo for more information.

**ELKS v0.8.0 Enhancements**

Thank you to all who contributed to ELKS enhancements, bug identification, testing and fixes.
The system continues to be very stable, with no known crashes and long uptimes.

**Major Enhancements**

Major additions include support for:
- Application development using the  Open Watcom C compiler and OS/2 executable format, allowing for
large multisegment programs to run, using all available memory if necessary. Doom was recently ported.
- Add NE2K, WD and 3COM networking support to default build.
- Compressed executables by default allowing for many more applications per floppy image.
- Asynchronous I/O was added to the kernel along with @Mellvik's Direct Floppy disk driver, which
allows for other applications to run while waiting for floppy disk I/O.
- All kernel resources dynamically adjustable at boot, allowing for the kernel heap, tasks,
inode and file allocations to be adjusted as required for min or max memory usage.
- Increased symbolic debugging capabilities with enhancements to the symbolic disassembler,
system call and function tracing capabilities within running applications.
- Rewrite of L1/L2 buffer cache operation allowing for faster operation using fewer buffers.
- Precision timing routines and removal of most all long division in the kernel and C library for
fastest operation and better elapsed time measurements.

**New Commands**

* [cmds] Port ttyclock to ELKS
* [cmds] Port ttypong to ELKS
* [cmds] Port ttytetris to ELKS
* [cmds] Port md5 from Minix/NetBSD to ELKS as md5sum by @ccoffing
* [kernel,cmds] Add sysctl system call and option get/set command

**Updated Commands**

* [cmds] Add "mouse r" for raw mouse output debugging
* [cmds] Add -s, -r, and -p options to shutdown, remove reboot, poweroff
* [cmds] Add accurate train timing to sl
* [cmds] Add decomp - ELKS executable decompressor
* [cmds] Add fsck-dos FAT filesystem checker to build
* [cmds] Changes to basic.c for OpenWatcom C compilation
* [cmds] Cleanup fsck-dos messages and reduce memory requirement
* [cmds] Display usage on invalid mount -t parameter
* [cmds] Enable more to recognize line wrapping
* [cmds] Enhance ps to better recognize task struct size mismatches
* [cmds] Finish fsck-dos port to ELKS, add to distribution
* [cmds] Fix ELKS_VERSION= on application builds
* [cmds] Fix man page reader for filenames > 14 (MINIX max)
* [cmds] Fix more to display correct number of lines w/man
* [cmds] Fix off-by-one bug in getty @T command by @vkoskiv
* [cmds] Fix open file on spawn of programs from /etc/inittab
* [cmds] Fix rmdir -p
* [cmds] Get ttyclock working with keyboard commands
* [cmds] Improve fsck output formatting for -lvv and -lvvv options
* [cmds] Meminfo and ps updates, small fixes
* [cmds] Reduce size of ps to 6k
* [cmds] Reduce size of ramdisk app, fix error messages
* [cmds] Remove fsck-dos FAT32 support allowing it to work on floppies
* [cmds] Reorganize clock.c source code, add AST hardware RTC support
* [cmds] Replace ifdef __ia16__ with ifdef _M_I86
[... 317 lines cut ...]
* docker: install patch, required for build by @d33tah
* inital commit of elkscmd/beep by @floriangit
* popen must be matched with pclose, not fclose by @ccoffing
* remove nonsense, author used incorrect OpenWatcom configuration by @jmalak
* smaller which by @ccoffing
* support for MacOs brew qemu installation by @madejusz

**Documentation**

* [docs] Add fmemalloc man pages for section 2 and 3
* [docs] Add man page for fsck-dos
* [docs] Update documentation on calling conventions, update license texts for utils
* [Documentation] Update function calling conventions and prologues

## New Contributors
* @floriangit made their first contribution
* @madejusz made their first contribution
* @and3rson made their first contribution
* @d33tah made their first contribution
* @jmalak made their first contribution
* @vkoskiv made their first contribution

**Full Changelog**: https://github.com/ghaerr/elks/compare/v0.7.0...v0.8.0

Chief Maintainer: @ghaerr

tyama501, 2024-09-23:
Hello @ghaerr 
Wow, great list.
Could you move the following to the PC-98 Port area when the release? I think that is the major update for the PC-98.
[bioshd] 720KB 2DD support for PC-98 2HD/2DD common drive by [@tyama501](https://github.com/tyama501)

Thank you!

toncho11, 2024-09-24:
There are some documentation updates/additions:
* https://github.com/ghaerr/elks/wiki/Access-to-UMB-memory
* https://github.com/ghaerr/elks/wiki/Increase-available-memory

We should put a link to ElksDoom in the release notes. We can also add that ElksDoom proves that big memory consuming applications can be ported to ELKS and VGA game programming is doable (doable before, but maybe not at this scale). We could (just an idea) do as well an ELKS image with some of the memory optimizations to boot directly into ElksDoom for this release. We could call it "hd32-minix-doom.img" for example.

rafael2k, 2024-09-24:
Yay! Congrats to all contributors!

ghaerr, 2024-09-24:
@toncho11: I already mentioned Doom in the release notes (too much to read already, right?), but will add a link as well.

The Wiki documentation isn't really release oriented, but agreed its important that everyone can find the information they'd like. I'll add the links you mentioned to the release notes Documentation area. We could add a "What's New" section to the Wiki as well, but this can be done at any time.

Vutshi, 2024-09-24:
Congrats to @ghaerr and the contributors! This seems like the biggest ELKS update since ~0.5 when I started following it.

ghaerr, 2024-09-24:
Version 0.8.0 is released! Prebuilt images are available here: https://github.com/ghaerr/elks/releases/tag/v0.8.0.

closed 2024-09-24 by ghaerr

Vutshi, 2024-09-24:
`sl` appears to be the most important ELKS app, with three entries on the list :)

ghaerr, 2024-09-24:
Well @Vutshi you may have my comment to you a week ago in all the madness, but yes, a number of enhancements were made to the console and ascii conversion routines so that your favorite program can run a little bit faster! :)

Vutshi, 2024-09-24:
@ghaerr, I'll be celebrating the release with some ceremonial train benchmarking.

Vutshi, 2024-09-24:
[image]

floriangit, 2024-09-24:
lol, what a release party!

tyama501, 2024-09-24:
Hi @ghaerr 

Could you attach fd1232-pc98 and fd1440-pc98 to the release too?

thanks!

ghaerr, 2024-09-24:
Yes, sorry. I am traveling but will upload ASAP.

Thank you!

ghaerr, 2024-09-27:
Hello @tyama501,

I've uploaded the PC-98 images for v0.8.0, sorry for the delay!

Thank you!

tyama501, 2024-09-27:
Hello @ghaerr 

Never mind. Thank you!