ghaerr/elks #2039: Building ELKS in macOS Sequoia 15.0

Vutshi, 2024-09-25:
Hi @ghaerr,

I updated macOS to the latest version, hoping to fix my (probably) buggy Intel GPU driver. Now, Command Line Tools seem to be rearranged, and I’m getting this error while building ELKS:
```
ia16-elf-gcc -mcmodel=small -melks-libc -mtune=i8086 -Wall -Os -mno-segment-relocation-stuff -fno-inline -fno-builtin-printf  -o basic basic.o host.o host-ibmpc.o asm-ibmpc.o 
flex -I -8 scan.l
mv lex.yy.c scan.c
expect 1 shift/reduce conflict
yacc -d bc.y
xcode-select: Failed to locate 'yacc', requesting installation of command line developer tools.
make[2]: *** [y.tab.h] Error 72
make[1]: *** [all] Error 1
make: *** [all] Error 2
```
It appears there is some kind of `yacc`:
```
which yacc
/usr/bin/yacc
```
But if I run it a CLT update request pops up:
```
yacc
xcode-select: Failed to locate 'yacc', requesting installation of command line developer tools.
```
CLT is already updated and `clang` works well.

**Configuration**
- ELKS 0.8
- macOS 15.0, Command Line Tools (CLT) 16.0

**Additional information**
I found somewhat related [information](https://mail-index.netbsd.org/tech-pkg/2024/03/08/msg028980.html) about missing `yacc` after update to CLT 15.3. However, I don't understand what do they propose to do about it.

Vutshi, 2024-09-25:
Thx, the workaround worked.I can hear the train's steam hissing.

closed 2024-09-28 by ghaerr

LdBeth, 2025-01-24:
For folks on the internet experiencing same problem on missing working yacc from CommandLineTools, and want a work arounds, in the `/Library/Developer/CommandLineTools` remove the defunct `yacc` and replace with a shell script file of same name with context (will need sudo):
```
#!/bin/sh
exec $(bindir)/bison -y "$@"
```

This is what the GNU bison would install for the yacc command. For whatever reason Apple replaced it by a stub that asks user to install Xcode since Sequoia.

PR #2046 (2024-09-28): [cmds] Remove yacc requirement for building bc
Fixes #2039.