ghaerr/elks #2042: Elks fails to compile on Debian 12 due to libc failing to compile. typo maybe?

sparky4, 2024-09-28:
**Description**
- What is expected ?
- to compile the images
- What is observed ?
- it stops very VERY quickly

**Configuration**
- Version (commit or release):
- latest
- Emulator or HW platform:
- IBM PC

**How to reproduce ?**
- Does the problem always happen ?
- yes yes
- List the steps to reproduce the problem.
- try to run ./build.sh

**Raw data**
- Screenshot(s):
- 
[image]

- Log file:
-  * End of ELKS kernel configuration.
 *
 * You may now run `make clean` and `make` to build.

Cleaning all...
make -C libc clean
make[1]: Entering directory '/home/sparky4/p/elks/libc'
for DIR in asm ctype debug error gcc getent malloc math misc net regex stdio string system termcap termios time ; do make -C $DIR clean || exit 1; done
make[2]: Entering directory '/home/sparky4/p/elks/libc/asm'
Makefile:23: *** target pattern contains no '%'.  Stop.
make[2]: Leaving directory '/home/sparky4/p/elks/libc/asm'
make[1]: *** [Makefile:240: clean] Error 1
make[1]: Leaving directory '/home/sparky4/p/elks/libc'
make: *** [Makefile:36: clean] Error 2
Build script has terminated with error 4

**Additional information**
Add any other context information about the problem here.

sparky4, 2024-09-28:
I am compiling for ibm pc

ghaerr, 2024-09-28:
It seems there may be an incompatibility with the `make` you are running? Can you run `make --version`? My version of make is GNU v3.81.  It seems the error is occurring quite quickly during `make clean`. Try that alone to confirm.

sparky4, 2024-09-28:
C:\home\sparky4\p\elks> make --version
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

ghaerr, 2024-09-28:
This is quite strange; have you tried building before using exactly this method? I am trying to ascertain whether this is a new issue, or perhaps something to do with Debian 12. 

Are you running Debian or Windows? I see your path is `C:\home\sparky4\p\elks`: that's Windows isn't it? The ELKS build is completely untested on Windows. I [found the following post](https://stackoverflow.com/questions/15113138/target-pattern-contains-no-makefile) explaining that having a backslash in your PATH can cause this problem.

Can you try a simple `make clean`, then `make`, to see if that works? If not, then we're likely dealing with incompatibility with your system using Windows-style pathnames separated with backslashes; I won't have a fix for that.

sparky4, 2024-09-28:
oh the prompt thing is a custom one dont worry the actual pwd is
```
C:\home\sparky4\p\elks> pwd
/home/sparky4/p/elks
```

sparky4, 2024-09-29:
i think it's a new issue. i was able to compile it before. i dont know what happened the last few months. i just cannot make libc anymore

toncho11, 2024-09-29:
If this is a fresh install of Debian then you need to do something like that in the beginning: `apt install build-essential` or remove your old copy of elks source and git `clone again`.

toncho11, 2024-09-29:
I am using WSL (Windows Subsystem for Linux)/Ubuntu 20. I did `git pull` and I managed to compile ELKS on the second time. First time there was some error, but it works for me.

floriangit, 2024-09-29:
I'm on Debian 12 as well. Same `make` version as well. Works. It does not look like ELKS build uses your (exotic) $PS1 to set something odd, but I'm not sure about it much, so maybe try to change it to debian default (and have a clean git clone of ELKS as toncho11 mentioned).

sparky4, 2024-09-29:
oh
ok ill do that

ghaerr, 2024-09-29:
Are you running `./build.sh` or `make clean`? The latter is preferred. It also looks like perhaps you may not have configured ELKS. The preferred way to do this is:
```
$ make menuconfig
(check that .config exists)
$ make clean
$ make
```
Don't run `./build.sh`, it may be contributing to the problem?

ghaerr, 2024-09-29:
I would also run the above without any fancy prompt - it looks like you may be running a prompt with a linefeed in it? Try with standard PS1= to eliminate variables, since both @toncho11 and @floriangit are saying this is working fine for them.

I see you got an earlier "ELKS is not configured yet" as well as an error in "Keymaps" directory... it appears as though the build.sh script is continuing and not stopping with errors, which will contribute to the problem. You might want to do as @toncho11 suggested and remove everything and start with a git clone or git pull from scratch, and then only use "make" as described above, no shell scripts.

sparky4, 2024-09-29:
ok ill do that

sparky4, 2024-09-30:
[image]

sparky4, 2024-09-30:
[image]

sparky4, 2024-09-30:
[image]

ghaerr, 2024-09-30:
You're still running a shell script. That's not 'make clean' by itself.

ghaerr, 2024-09-30:
Nonetheless, something is definitely wrong with the build environment. I don't know what it is, and I cannot duplicate it since I don't have a Windows or Linux box.

tyama501, 2024-10-01:
Hello from Debian 12.7. OK, I got the system up.
I will try to get elks environment later this week.

sparky4, 2024-10-01:
let me build the kernel then

sparky4, 2024-10-01:
it is working! :D

ghaerr, 2024-10-01:
If it is now working, I suspect the problem was/is that you were running ./build.sh, a shell script that doesn't work with WSL, while `make`, `make clean` or `make kimage` works. Please confirm.

sparky4, 2024-10-01:
make and make clean never worked but make kimage works just fine

tyama501, 2024-10-01:
make kimage is only for kernel, right?

sparky4, 2024-10-01:
yeah it seems

tyama501, 2024-10-02:
Hi @sparky4 

Are you using build.sh in the repository root or
are you using tools/build.sh?

I am now following steps here that uses tools/build.sh
https://github.com/ghaerr/elks/blob/master/BUILD.md

tyama501, 2024-10-02:
I am still building the compiler but
it seems that mtool in my Debian is v4.0.33. That may not work for PC-98. 
https://github.com/ghaerr/elks/pull/1463

tyama501, 2024-10-02:
I could build gcc and started building elks.

I got xxd: not found.
I think I have seen this error before (on FreeBSD)
[image]

tyama501, 2024-10-02:
OK I could apt-get it.

tyama501, 2024-10-02:
I got error in elkscmd/sys_utils.

[image]

default .config creating by build.sh seems old.
I'll try make after copying ibmpc-1440.config to .config.

ghaerr, 2024-10-02:
The makeboot.c compilation error is because `xxd` failed creating the autogenerate header file mbr_autogen.c earlier in elks-master/bootblocks/Makefile:
```
mbr.bin: mbr.S
    $(CC) $(INCLUDES) -E -o mbr.tmp mbr.S
    $(AS) $(ASFLAGS) -o mbr.o mbr.tmp
    $(LD) $(LDFLAGS) -M -o mbr.bin mbr.o > mbr.map
    xxd -i mbr.bin > mbr_autogen.c
```
You'll probably have to `make clean` after installing xxd then restart.

If you've gotten this far, that's farther than @sparky4 got, since `make clean` is where he's getting his error, and I really don't know why.

tyama501, 2024-10-02:
OK. Done. The following worked.

cp -a ibmpc-1440.config .config 
make menuconfig
make clean
make

[image]

I could make fd1440.img.

So, basically it can be built with Debian 12.7 on native PC,
if correct .config is used.

sparky4, 2024-10-03:
ok
let me try

sparky4, 2024-10-03:
git diff is empty and here is the log 
[my.log](https://github.com/user-attachments/files/17237836/my.log)

sparky4, 2024-10-03:
yes i previously built elks on both debian systems and they worked before the git pull/git clone

sparky4, 2024-10-03:
```/home/sparky4/p/elks/libc/asm/Makefile:23: *** target pattern contains no '%'.  Stop.```
this is the error

ghaerr, 2024-10-03:
@sparky4, can you try editing libc/asm/Makefile and replacing the following line:
```
clean:
        $(RM) *.[aod]
```
with 
```
clean:
        rm -f *.o *.a *.d
```
Note that the `rm -f` *must* be preceded with a TAB character, so don't edit to the left of `$(RM)`, just replace $(RM) with rm -f.

If that doesn't work, can you run `printenv` or `env` and post the results or screenshot?

ghaerr, 2024-10-03:
Finally if the above three don't work, take the date/time out of your PS1= prompt. `make` gets confused when it runs into colons and this could cause the issue. You should try running with a standard PS1= prompt, not your custom version.

sparky4, 2024-10-03:
i think it's a synthax error

```
printenv
OWDOCBUILD=1
SHELL=Bourne
SESSION_MANAGER=local/compy4:@/tmp/.ICE-unix/2942,unix/compy4:/tmp/.ICE-unix/2942,inet6/compy4:43337,inet/compy4:38331
WINDOWID=41943077
QT_ACCESSIBILITY=1
COLORTERM=truecolor
PALMDEV=/home/sparky4/p/palmdev_V3
XDG_CONFIG_DIRS=/etc/xdg
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
HISTCONTROL=ignoreboth
XDG_MENU_PREFIX=xfce-
GTK_IM_MODULE=ibus
PKG_CONFIG_PATH=/home/sparky4/install/lib/pkgconfig/:/home/sparky4/install/share/pkgconfig/
TOPDIR=/home/sparky4/p/elks
SSH_AUTH_SOCK=/tmp/ssh-XXXXXXUTXFtI/agent.2942
OWGUINOBUILD=1
OS=LINUX
DEV_HOME=/home/sparky4/p/nouveau/
WIPFC=/dos/fdos/watcom/wipfc
XMODIFIERS=@im=ibus
DESKTOP_SESSION=xfce
SSH_AGENT_PID=3166
ACLOCAL=aclocal -I /home/sparky4/install/share/aclocal
GTK_MODULES=gail:atk-bridge
XDG_SEAT=seat0
ENV=LINUX
PWD=/home/sparky4/p/elks
LOGNAME=sparky4
XDG_SESSION_DESKTOP=xfce
NVD=/home/sparky4/install
XDG_SESSION_TYPE=x11
REXX=/usr/bin/rexx
XAUTHORITY=/home/sparky4/.Xauthority
XDG_GREETER_DATA_DIR=/var/lib/lightdm/data/sparky4
GDM_LANG=en_US.utf8
HOME=/home/sparky4
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.swp=00;90:*.tmp=00;90:*.dpkg-dist=00;90:*.dpkg-old=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:
XDG_CURRENT_DESKTOP=XFCE
VTE_VERSION=7006
OWOBJDIR=binbuild
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
CROSSDIR=/home/sparky4/p/elks/cross
CLUTTER_IM_MODULE=ibus
WATCOM=/dos/fdos/watcom
XDG_SESSION_CLASS=user
TERM=xterm-16color
USER=sparky4
ROOT=/home/sparky4/p/osfree
DISPLAY=:0.0
INCLUDE=/usr/include/
MESA_INSTALLDIR=/home/sparky4/p/mesa/
SHLVL=1
TOOLS=/home/sparky4/p/osfree/bin/tools
QT_IM_MODULE=ibus
XDG_VTNR=7
XDG_SESSION_ID=1
LD_LIBRARY_PATH=/home/sparky4/install/lib
XDG_RUNTIME_DIR=/run/user/1000
PS1=\[\033[1;32m\]\[\033[45m\]\[\033[1;35m\]\[\033[45m\][\[\033[1;32m\]\[\033[45m\]\t\[\033[1;35m\]\[\033[45m\]]\[\033[1;34m\]\[\033[45m\][\[\033[1;32m\]\[\033[45m\]\w\[\033[1;34m\]\[\033[45m\]]\[\033[40m\]\[\033[1;32m\]\n\[\033[40m\]\[\033[1;34m\][\[\033[40m\]\[\033[1;32m\]\u\[\033[40m\]\[\033[1;35m\]@\[\033[40m\]\[\033[1;32m\]\h\[\033[40m\]\[\033[1;34m\]]\[\033[40m\]\[\033[1;32m\]$\[\033[1;32m\]\[\033[40m\]\[\033[1;32m\] 
OWDOSBOX=dosbox
OWROOT=/home/sparky4/p/watcom2
WD_PATH=/dos/fdos/watcom/binl
XDG_DATA_DIRS=/usr/share/xfce4:/home/sparky4/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share:/var/lib/snapd/desktop:/usr/share
PATH=/home/sparky4/p/elks/elks/tools/bin:/home/sparky4/p/elks/cross/bin:/usr/lib/firefox:/home/sparky4/.local/bin:/dos/fdos/watcom/binl:/home/sparky4/p/osfree/bin/tools:/home/sparky4/p/osfree:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
GDMSESSION=xfce
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
OWTOOLS=GCC
LIB=
_=/usr/bin/printenv
```

tyama501, 2024-10-03:
Hi @sparky4 

Are you in master branch?
If you got trouble after git pull you might not in master.
git branch
can check it.

Also
git diff HEAD 
show nothing right?

tyama501, 2024-10-03:
Have you tried this?
cp -a ibmpc-1440.config .config
make menuconfig
make clean
make

tyama501, 2024-10-03:
One more thing.

git status

can show modified files.

Some files may failed while merging when git pull.

sparky4, 2024-10-03:
yeah i did the copy the config file i am compiling it rn!

ghaerr, 2024-10-03:
Thanks for the `printenv` output @sparky4. I was hoping to see something abnormal in PATH=, but it looks OK.

> yeah i did the copy the config file i am compiling it rn!

I take that to mean you can compile with `make kimage` but not `make clean`?

```
clean:
        rm -f *.o *.a *.d
```
Did you try the above change, detailed in my post above?

Can you try changing your prompt using `export PS1=sparky ` and then running `make clean`? 

The only other item that is a unfamiliar is SHELL=Bourne, rather than SHELL=/bin/sh or something like that. Can you change to BASH Shell somehow, or is Bourne the WSL option?

sparky4, 2024-10-04:
hmmm i think it is make itself being problematic.. i tried on both machines

sparky4, 2024-10-04:
ill try sh... but i think it's make itself

sparky4, 2024-10-05:
oh
```
cd /home/sparky4/p/elks; \
( \
	echo CREATE libc/build-ml/elkslibc/medium/regparmcall/libc.a.tmp; \
	for s in libc/build-ml/elkslibc/medium/regparmcall/asm/out.a libc/build-ml/elkslibc/medium/regparmcall/ctype/out.a libc/build-ml/elkslibc/medium/regparmcall/debug/out.a libc/build-ml/elkslibc/medium/regparmcall/error/out.a libc/build-ml/elkslibc/medium/regparmcall/gcc/out.a libc/build-ml/elkslibc/medium/regparmcall/getent/out.a libc/build-ml/elkslibc/medium/regparmcall/malloc/out.a libc/build-ml/elkslibc/medium/regparmcall/math/out.a libc/build-ml/elkslibc/medium/regparmcall/misc/out.a libc/build-ml/elkslibc/medium/regparmcall/net/out.a libc/build-ml/elkslibc/medium/regparmcall/regex/out.a libc/build-ml/elkslibc/medium/regparmcall/stdio/out.a libc/build-ml/elkslibc/medium/regparmcall/string/out.a libc/build-ml/elkslibc/medium/regparmcall/system/out.a libc/build-ml/elkslibc/medium/regparmcall/termcap/out.a libc/build-ml/elkslibc/medium/regparmcall/termios/out.a libc/build-ml/elkslibc/medium/regparmcall/time/out.a; \
		do echo ADDLIB "$s"; done; \
	echo SAVE \
) | ia16-elf-ar -M
ia16-elf-ar: libc/build-ml/elkslibc/medium/regparmcall/asm/out.a: No such file or directory
make[2]: *** [Makefile:123: /home/sparky4/p/elks/libc/build-ml/elkslibc/medium/regparmcall/libc.a] Error 9
make[2]: Leaving directory '/home/sparky4/p/elks/libc'
make[1]: *** [Makefile:200: all] Error 2
make[1]: Leaving directory '/home/sparky4/p/elks/libc'
make: *** [Makefile:11: all] Error 2
Build script has terminated with error 5
```

sparky4, 2024-10-05:
i think the issue is i think how libc make files are kinda strange

sparky4, 2024-10-06:
make clean works now but make fails we are making progress! <3

sparky4, 2024-10-06:
nvm there is issues with the libc makefiles

tyama501, 2024-10-08:
Hello @ghaerr 

The default mformat in Debian 12.7 is 4.0.32, so it seems that current fd1232.img cannot be created (as we already know).
Not easy to install older version so I tried modifying some parameters.
It seems that there is no error if -L is removed. (-M might be better for -S since M is for logical sector size.)
I haven't  looked difference or tried the image since there is no PC-98 emulator on this PC yet.
I will do that later.
 
[image]

ghaerr, 2024-10-08:
Hello @tyama501,

Thanks for your report. I can't quite remember exactly why we couldn't use one of the later versions of `mformat` for ELKS, but if you learn another way of formatting a floppy, even just for 1232k, that would be great, as that could be changed in image/Make.defs for PC-98 which would then work on Debian 12.7 and elsewhere. I'll look further into this after you have learned more, thank you!

tyama501, 2024-10-10:
The fd1232.img created by mformat 4.0.32 without -L could boot,
but the second FAT position and the root directory position moved.
The total 2464k, and df Free 1528 38% are not correct, so it seems that I need to try other values.

(Above one is the new image)
[image]

[image]

tyama501, 2024-10-12:
I manually edited the BPB Total Sector16bits from 0x09A0 to 0x04D0 (0x09A0 / 2) by the binary editor.
The elks could recognize total 1232K, and df value looks correct. Does it use this value to calculate free space?
[image]

ghaerr, 2024-10-12:
> The elks could recognize total 1232K, and df value looks correct. Does it use this value to calculate free space?

Yes, the FAT12/FAT16 filesystem will use the Total Sector 16-bit value as the maximum number of sectors contained in the filesystem (FAT32 uses the later 32-bit total sector field).

I haven't had time to look into the differences yet between the mformat versions, sorry. What way are you currently thinking might work to use a later version of mformat with options, or would there need to be a separate "mformat-fix" program required to make this work, at this point?

tyama501, 2024-10-13:
Hello @ghaerr 

I have been trying to replace the value by dd after the image is created but the somehow escape sequence in the make.image does not work.

Neither
echo -en "\xD0\x04"
nor
printf "\xD0\x04"
does not work.

Do you have idea to create the binary data?

ghaerr, 2024-10-13:
> Neither
echo -en "\xD0\x04"
nor
printf "\xD0\x04"
does not work.

When you say "does not work" - what does the binary map show at the data location you're trying to modify? I presume you're trying something like `printf "..." | dd skip=... of=fd1232.img`?

> Do you have idea to create the binary data?

It would be quite easy to write a C program to do so, much like how setboot.c modifies the BPB. I suppose you could add an option to setboot that would correct the problem.

Actually, I see Make.images uses `setboot $(TARGET_FILE) -K -F $(BPB) $(FD_FAT_BOOT)` for FD1232, and the -K option specifics 1K block size. This (or another new option like `-T 1232`) option could also (re)write the BPB total sectors16 field, and then the problem would be automatically taken care of for FD1232.

If we go with a setboot.c modification, are the mformat options the same or different than 4.0.23? That is, can we use the 4.0.23 paramaters with a later version of mformat and then just run setboot -K to fix it, or will the mformat options have to be different for later version?

tyama501, 2024-10-13:
In printf case the character \ (5C) and x (78) are used. I forgot about echo case but it is strange. Probably some built in echo -e cannot be used.

I will look setboot later.

-L 2 need to be changed to -L4 (or removing the option)
That will take unnecessary 2 additional sectors for each FAT but it would work for the older version too(I think...

sparky4, 2024-10-13:
```[10:59:46][~/p/elks]
[sparky4@lappy4]$ make --version
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

ghaerr, 2024-10-14:
Hello @tyama501,

> The default mformat in Debian 12.7 is 4.0.32, so it seems that current fd1232.img cannot be created 

I have been looking into this, and `mtools` is quite complicated, and still haven't found out exactly why later versions than mtools-4.0.23 don't work for our project. I remember experimenting years ago but of course it is time-consuming to figure out exactly where the mtools maintainers break image formats that used to work.

So I a thinking that, instead of hacking our own tools like `setboot` to fix mtools-4.0.32 etc, maybe a better solution would be to build mtools-4.0.23 from source within ELKS elks/tools/mformat locally, and thus a working version always be available when building ELKS. While Debian 12.7 has chosen 4.0.32, I see that we are dependent on the automated Github CI build using 4.0.23. Should that be updated, our automated build would break.

What do you think? The potential risk is that mtools may not build on certain systems without changes, but we can always manage that with bug reports. This would also save you from diving into hex debuggers trying to (also) figure out exactly what is wrong with newer versions of mformat for PC-98 1232k disk image creation.

Thank you!

[14 more replies]

closed 2024-12-29 by ghaerr