ghaerr/elks #2043: mknod really working?

floriangit, 2024-09-28:
I re-installed ELKS on my HDD with MINIX FS with `sys`. After reboot it won't let me login, since tty1 is missing:

[image]

I looked around and found tty4 was added:
https://github.com/ghaerr/elks/pull/1980/commits/8b5ef832f7ea4f9836d521e28d2bb4bdb7b93e2a
But that reads like a pure addition, nothing that should break here.

Then I booted from /dev/fd0, mounted the HDD, looked at `/bin/sys` and verified that `mknod /dev/tty1 c 4 0` line is there.
So why it did not run on my HDD? Others like tty2, 3 etc. are there!

I try manually and it's getting more confusing:
[image]

Any hints appreciated!

floriangit, 2024-09-28:
(I am able to `touch` files in above directory, so it does not look like a read-only issue)

floriangit, 2024-09-28:
Retried with official 0.8.0 and same issue of first screenshot. It may be my HDD. But maybe not.

ghaerr, 2024-09-28:
Hello @floriangit,

Well, its seems I'm batting zero for two on the latest release. I have duplicated the problem over here and will have a PR fix shortly. I've attached a working `mknod`, or you can recompile with the following fix:
```
diff --git a/elkscmd/file_utils/mknod.c b/elkscmd/file_utils/mknod.c
index 01ff0390..611246e2 100644
--- a/elkscmd/file_utils/mknod.c
+++ b/elkscmd/file_utils/mknod.c
@@ -28,13 +28,13 @@ int main(int argc, char **argv)
                major = atoi(argv[3]);
                minor = atoi(argv[4]);
                
-               if (major > 0 && minor > 0) {
+               if (major > 0 && minor >= 0) {
                        if (mknod (argv[1], newmode | filetype, MKDEV(major, minor))) {
                                errstr(argv[1]);
                                errmsg(": cannot make device\n");
                                return 1;
                        }
-               }
+               } else goto usage;
        } else if ((argc == 3) && (argv[2][0] == 'p')) {
                if (mknod (argv[1],newmode | S_IFIFO, 0)) {
                        errstr(argv[1]);
```
This was broken by a commit of mine just before release, I am sorry. I think we should probably release a v0.8.1 with a fix for this and `passwd`.
[mknod.zip](https://github.com/user-attachments/files/17175623/mknod.zip)

floriangit, 2024-09-28:
Thanks @ghaerr, looks you nailed it real quickly, I can confirm tomorrow.

> commit [...] just before release.

Haha, it's the same story we do since humans do any kind of "releases". We never learn what _not_ to do before we "release", right? :sweat_smile:

closed 2024-09-28 by ghaerr

floriangit, 2024-09-29:
thanks, confirmed to be fixed in git/master for me.

tyama501, 2024-10-01:
Great. Almost ready for 0.8.1?

ghaerr, 2024-10-01:
> Almost ready for 0.8.1?

What do you think, have you tested more on PC-98 and are happy with results?

tyama501, 2024-10-01:
Only with the emulator but I also installed it to the hard disk.
So far it is good.
Thanks.

PR #2045 (2024-09-28): [cmds] Fix mknod when used with minor == 0
Fixes #2043.