ghaerr/elks #2062: /bootopts on FAT hard disk

tyama501, 2024-10-04:
Hello @ghaerr 

I could update my PC-9801RX hard disk (/dev/hdc1) to 0.8.0 (+bug fix).

I took following steps, and it could boot,
but I noticed the bootopts file cannot be recognized.

sys /dev/hdc1
makeboot -f fat.bin (since it needs special boot code)
mount /dev/hdc1 /mnt
cp /bootopts /mnt/

I haven't used bootopts before on the hard disk, so I don't know it could recognize before or not.
I can open the bootopts file and read it.

Where the bootopts should be on the hard disk?
Is there some FAT limitation like it should be within 8 entries? 

[image]

Thank you.

ghaerr, 2024-10-05:
Hello @tyama501,

> Where the bootopts should be on the hard disk?
Is there some FAT limitation like it should be within 8 entries?

Yes, setup.S line 1041 shows that /bootopts needs to be in the first 8 root directory entries on FAT disks (only). However, looking through image/Make.image, I don't see anywhere where this is enforced, including in bin/sys. So that is likely the problem.

In order to check, you can use `ls -laU /mnt` to view the unsorted root directory on your mounted hard drive. Remember that /dev is faked and is not actually present. You may have to delete /tmp or another one of the first files/directories, copy /bootopts, then mkdir the deleted directory again.

Thank you!

tyama501, 2024-10-05:
I see. Thank you.

I will try it later.
Is there any reason sys does not copy the bootopts?
I think it can avoid issue it copies bootopts after copying LINUX before copying applications.
(If the FAT is initially formatted)

toncho11, 2024-10-05:
So @tyama501 you tried the installation on an existing ELKS installation (not a formatted disk) and this resulted in /bootopts not being in the top 8 entries?

ghaerr, 2024-10-05:
I think `sys` should copy /bootopts, but does not since it might not exist. I will look into that change.

tyama501, 2024-10-05:
Hello @toncho11 

I formatted it but currently the sys does not copy bootopts so I copied it after the install.

tyama501, 2024-10-05:
It seems that my power supply on PC-98 is broken, and I cannot test with it anymore...
Probably it will occur on the emulator too.

toncho11, 2024-10-06:
I test on real hardware with MINIX and it is OK.
On fat I keep getting an error from makeboot that the target filesystem must be the same, but they should both be fat. I could not overcome this error. I tried changing the partition type, but it makeboot keeps on thinking that the hdd partition is Minix.

ghaerr, 2024-10-06:
Thanks @toncho11, I'll take a look at this additional problem you've turned up: I assume you're running from a booted FAT floppy and trying to `sys` a FAT HD partition, which is then giving the error you've described?

I'm looking in to the originally described problem now. I'm glad you've confirmed that MINIX has no issues; adding the /bootopts copy will always work since for MINIX, the directory location of /bootopts is not restricted to the first 8 entries like it is on FAT.

toncho11, 2024-10-06:
Yes for the fat setup. I tried to create a new fat partition as well, but then which partition code should I assign for fat? Also how is the check working in makeboot, probably not by partition code. Maybe it is reading MBR and some old information is stored there. Before this partition hda1 was minix with ID 80.

ghaerr, 2024-10-06:
The `fdisk` program will list the partition types using the `l` command. Here's the internal list:
```
void list_types(void)   /* FIXME - Should make this more flexible */
{
    printf(
    " 0 Empty                  3c PartitionMagic recovery 85 Linux extended\n"
    " 1 FAT12                  40 Venix 80286             86 NTFS volume set\n"
    " 2 XENIX root             41 PPC PReP Boot           87 NTFS volume set\n"
    " 3 XENIX usr              42 SFS                     93 Amoeba\n"
    " 4 FAT16 <32M             4d QNX4.x                  94 Amoeba BBT\n"
    " 5 Extended               4e QNX4.x 2nd part         a0 IBM Thinkpad hibernate\n"
    " 6 FAT16                  4f QNX4.x 3rd part         a5 BSD/386\n"
    " 7 HPFS/NTFS              50 OnTrack DM              a6 OpenBSD\n"
    " 8 AIX                    51 OnTrack DM6 Aux1        a7 NeXTSTEP\n"
    " 9 AIX bootable           52 CP/M                    b7 BSDI fs\n"
    " a OS/2 Boot Manager      53 OnTrack DM6 Aux3        b8 BSDI swap\n"
    " b Win9x FAT32            54 OnTrack DM6             c1 DR-DOS/sec FAT-12\n"
    " c Win9x FAT32 (LBA)      55 EZ-Drive                c4 DR-DOS/sec FAT-16 <32M\n"
    " e Win9x FAT16 (LBA)      56 Golden Bow              c6 DR-DOS/sec FAT-16\n"
    " f Win9x Extended (LBA)   5c Priam Edisk             c7 Syrinx\n"
    "10 OPUS                   61 SpeedStor               db CP/M / CTOS / ...\n"
    "11 Hide FAT12             63 GNU HURD or SysV        e1 DOS access\n"
    "12 Compaq diagnostics     64 Novell Netware 286      e3 DOS R/O\n"
    "14 Hide FAT16 <32M        65 Novell Netware 386      e4 SpeedStor\n"
    "16 Hide FAT16             70 DiskSecure Multi-Boot   eb BeOS fs\n"
    "17 Hide HPFS/NTFS         75 PC/IX                   f1 SpeedStor\n"
    "18 AST Windows swapfile   80 ELKS / Old Minix        f2 DOS secondary\n"
    "1b Hide Win9x FAT32       81 New Minix / Old Linux   f4 SpeedStor\n"
    "1c Hide Win9x FAT32 (LBA) 82 Linux swap              fd Linux raid autodetect\n"
    "1e Hide Win9x FAT16 (LBA) 83 New Linux               fe LANstep\n"
    "24 NEC DOS                84 OS/2 Hide C:            ff BBT\n"
    );
}
```
I'll have to check how/where `makeboot` gets the partition type.

tyama501, 2024-10-07:
Hi @ghaerr and @toncho11 
Thank you for fixing it and testing!