ghaerr/elks #2070: panic: divide fault

toncho11, 2024-10-07:
Not sure it is a problem or it should be like that but if you enable/uncomment both:
* the memory reduction line with "n" at the end
* init=/bin/sash
you can not complete the boot process successfully. It halts with "panic: divide fault" and then SYSTEM HALTED

ghaerr, 2024-10-07:
You mean like this:
```
task=6 buf=8 cache=4 file=20 inode=24 heap=15000 n # min w/no rc.sys
#sync=30
#init=/bin/init 3 n # muser serial no rc.sys
init=/bin/sash
```
I can't get this to repeat on QEMU. It tries to run /bin/sash with `n` as an argument and says "n: no such file or directory" after mounting VFS root, which although not booting is correct with /bin/sash. Are you running on real hardware? Which format disk/.config?

toncho11, 2024-10-07:
Yes, that's it on real 8Mhz CPU.

ghaerr, 2024-10-07:
Which CPU and system exactly? 8088 IBM 5150?

Just tried the same test on QEMU w/FAT and same result for me, just "n: no such file or directory" correct behavior.

You might want to try just the line `init=/bin/sash n` with the other commented out, and various variations to see when the error goes away, since I can't repeat it over here. Thank you.

toncho11, 2024-10-07:
I replicated the same setup in 86box emulator: Amstrad 1640, 8Mhz , VGA card and I get the error.
[image]
[86box.zip](https://github.com/user-attachments/files/17284974/86box.zip)
This is the config file for 86box. I used a fat hdd image for ELKS.

toncho11, 2024-10-07:
And this is the fat image:
[hd32-fat_kernel_panic.zip](https://github.com/user-attachments/files/17285000/hd32-fat_kernel_panic.zip)
with panic already, but you can mounted.

toncho11, 2024-10-07:
86box has MacOS version: https://github.com/86Box/86Box/releases/tag/v4.2.1
Hope you can run it.

ghaerr, 2024-10-08:
I downloaded the macOS 86box as well as your 86box.cfg, but it is saying that I need to download ROM files in order to run the emulator. I downloaded the ROM set which is 25+ files. Do you have any instructions as to how to select one of the many ROMs in order to get it to run?

ghaerr, 2024-10-08:
@toncho11, actually it just started running after renaming the ROM set to "roms". However, it seems to only want to boot a floppy image. I will play around with figuring out how to configure it to boot your image.

ghaerr, 2024-10-08:
Ok, it somehow created a completely blank hd32-fat.img. I replace that with yours and the emulator booted up ELKS. However, 86box is showing exactly what QEMU is: "n: no such file or directory"!!!
[image]

ghaerr, 2024-10-08:
I am guessing perhaps the ROM used for emulation may be why the results are different? I am not yet sure how to specify the working directory of the emulator and am running it from my desktop, along with the 86box.cfg and hd32-fat.img, all on the desktop. The 86box.cfg specifies bios = xt. I can't see ways to verify this information when the emulator is running.

toncho11, 2024-10-08:
Select machine type 8086 and then Amstrad 1640.

toncho11, 2024-10-08:
Select XTIDE as hard drive.

toncho11, 2024-10-08:
I use the same image and it is as you reported. 
Now download this image from: [hd32-fat.img](https://github.com/ghaerr/elks/actions/runs/11219801823/artifacts/2024989307)
[hd32-fat.img](https://github.com/ghaerr/elks/actions/runs/11219801823/artifacts/2024989307)

Change the two lines.
Use "shutdown -r".

Ant then it does the panic.

toncho11, 2024-10-08:
After a hard reset ... the problem is not as you say. 
So it must be the shutdown command????

ghaerr, 2024-10-08:
> Change the two lines.
Use "shutdown -r".
Ant then it does the panic.

Ok, I got the panic, finally.

> After a hard reset ... the problem is as you say (no panic).

Yes, upon rebooting the same image just modified above, no panic.

> So it must be the shutdown command????

Not sure yet. I'm thinking it might be something to do with the kernel exit procedure when `sash n` exits with an error, but still works after cold boot. I'll look into this further.

Does your real hardware always panic, or can you get it to give the "n: no such file or directory" ever?

toncho11, 2024-10-08:
Far fetched but: maybe shutdown sends park heads which does something that survives the soft reboot. 
Maybe something that stays in the XTIDE during a soft reboot.

toncho11, 2024-10-08:
Also shutdown should be doing the sync, right? I am not doing it because I have assumed that it does.

toncho11, 2024-10-08:
This time I did sync and ctrl+alt+del. 
I got panic.
Power on/off does render the computer in the "No such file" state. It looks permanent.
Tested on real Amstrad.

ghaerr, 2024-10-08:
I can only get this to fail using your hd32-fat.img. Did you build that yourself? Which commit is it from? When I build the latest and uncomment the two lines then shutdown -r, it doesn't fail with panic!

ghaerr, 2024-10-08:
I am thinking the reason this is happening has something to do with the /bootopts file: on a second (warm) boot, the memory segment that /bootopts is loaded into is NOT zeroed, but instead the new copy of /bootopts is read on top of the old one. If the /bootopts were shorter (which it is with the removal of two '#' chars), there is a possibility for some garbage at the end, particularly the last two chars of the previous /bootopts; this could be causing a crash. This is a possibility since it appears that /bin/sash never runs, which is why we don't see the "no such file" message.

In order to test this theory, after uncommenting the two lines, go to the end of the /bootopts and add one more line commented out with whatever you want in it - this would make the /bootopts file larger than before. If this changes the outcome, then we are on to something.

ghaerr, 2024-10-08:
> In order to test this theory

Tested - and no change, still panic. So I can still only get this to fail using your hd32-fat.img. When I create hd32-fat.img using "make images" using the ibmpc-1440-nc.config, I can't get it to panic.

This isn't a high priority bug, since `/bin/sash n` will always fail to boot anyways, but it'd be nice to know why its happening. We need to find a way to create a failure from a standard build, or a known commit.

toncho11, 2024-10-08:
I am using the hd32-fat image from your fdisk PULL request https://github.com/ghaerr/elks/actions/runs/11219801823

toncho11, 2024-10-08:
The general understanding is that softboot clear the memory. No doubt about it. 
Some initialization steps of the hardware might be skipped though.

ghaerr, 2024-10-08:
@toncho11:

> I am using the hd32-fat image from your fdisk PULL request https://github.com/ghaerr/elks/actions/runs/11219801823

That's the latest version - very strange: when I build that version using ibmpc-1440-nc.config, which is the same as used for the automated CI build, I can't get the image to fail! There's something going on here which we still haven't uncovered...

ghaerr, 2024-10-08:
@toncho11,

After playing around several hours (far?) too long with this, I am thinking the issue could well be a bug in the Amstrad PC 1640 BIOS. This is because, for instance, when I change the machine ROMs from Amstrad 1640 to Compaq DeskPro, use your hd32-fat.img and make the same changes you've made, the emulation works with Compaq DeskPro BIOS, whereas the emulation using Amstrad BIOS fails. 

Since no other emulator actually uses the Amstrad ROM BIOS, this could very well be the reason for the real bug - the Amstrad BIOS may be buggy after a warm start, and cause a Divide Error. Note that only in recent version of ELKS did we add the Divide Error trap, for which previously no error was reported by the BIOS nor ELKS. Thus previously, this could have been happening on warm starts and there would never be an error shown.

So, unless we can come up with another emulator that shows this error, or you can show it on a non-Amstrad BIOS being emulated by 86Box, I will stop work on this, since it takes quite a bit of time 2-3 minutes for each boot run (and still can't reproduce from my build, for yet another unknown reason).

Thank you!

toncho11, 2024-10-09:
Thank you! I was thinking in the beginning that it might be linked to a bigger issue like a new divide code or something, but now I agree with you. 

Thank you!!!

Enjoy your trip! :)

closed 2024-10-09 by toncho11

ghaerr, 2024-10-09:
@toncho11,

> that it might be linked to a bigger issue like a new divide code 

The panic is brought about because ELKS now traps HW INT 0 which is produced on divide by zero or a divide overflow. Thus, the likely Amstrad BIOS bug is exposed and system operation is discontinued since a numeric result is incorrect (somewhere, likely in BIOS ASM code). There are some who think that a system panic might not be the best option, however.

For the 8086 ad 8088 CPUs only, which includes your Amstrad 1640, on divide fault the chip pushes the address *past* the DIV instruction, allowing for a simple IRET instruction to resume operation. Most BIOSes fill all 256 interrupt vectors with an IRET, and that's likely what the Amstrad BIOS does, so you'd never see anything (and neither did Amstrad). This CPU behavior was changed in the 80186 and up to push the beginning of the DIV instruction, and a simple IRET would end up in a silent busy loop.

For your hardware, if you decide that you don't want to see a panic on (some strangely configured?) reboots, you can disable the ELKS divide fault handler by commenting out the following line in elks/arch/i86/kernel/irq.c:
```
    int_handler_add(IDX_DIVZERO, 0x00, div0_handler_panic);
```

This was also debated at the time of adding the handler, but ultimately it was thought best to show an error rather than continue without knowing.

Vutshi, 2025-05-07:
> So, unless we can come up with another emulator that shows this error, or you can show it on a non-Amstrad BIOS being emulated by 86Box, I will stop work on this

@ghaerr I’ve found a similar issue on a different emulator/BIOS combination:

**Setup**: MartyPC emulating a Compaq Deskpro (8086 CPU)
**With Compaq BIOS or Glabios**: ELKS crashes with a “divide by zero” fault

[image]

**With Sergey Kiselev’s bios-xt**: ELKS boots normally
**Key difference**: Under Compaq BIOS and Glabios, ELKS detects zero floppy drives:

[image]

After a night of debugging, @640‑KB arrived at the following conclusion:

> anywho... that's the trick. ELKS is treating bit 0 of the INT 11h aka `40:10` config byte as no floppy drives installed if `0` but if it's set that way then it later tries to divide by that number.
> 
> DOS generally doesn't care about that bit because what it means varies a lot
> 
> 8088_bios hard-codes the number and type of floppy drives and probably completely ignores most motherboard switches anyway (which makes total sense because neither the FE2010A or nor the Xi8088 have all of them) and make DeskproMarty's PPI work like 5150, and have bit 0 always `1`
> 
> GLaBIOS (XT version) was reading the DIP switches as if they were an XT, but it was getting bit 0 as a `0` from the PPI so once again, ELKS thought 0 floppy drives

ghaerr, 2025-05-07:
@Vutshi, thanks for posting this! And here I'd been thinking the problem was in @toncho11's AMI BIOS (which is probably causing the problem in the same way) the whole time.

Hello @640-KB, and thank you for spending possibly too much time debugging this! I have some additional questions about exactly how the INT 11/equipment config byte at 40:10 should be handled (see below).

Ok, I think I've (finally) figured out what the real problem is: long story short, yes - the kernel initializes thinking there's no floppy drives, but since the system was booted from floppy, the floppy device driver gets called by default, and then when it's about to do a full track cache read, it divide faults when trying to see if the cache is large enough, with the (non-initialized) drive's sector size = 0:
```
    if (num_sectors > (TRACKSEGSZ / drivep->sector_size))
        num_sectors = TRACKSEGSZ / drivep->sector_size;
```

That can all be fixed, but I'm wondering what the best way to determine the number of floppy drives is, especially if/when there's some debate about whether the contents can be relied upon. Everything I've read says that that bit 0 specifies the number of drives less one in bits 6-7. What value does GLaBIOS put in those bits?

Some background: (I think) this problem only manifests itself on older BIOSes that don't support INT 13h function AH=8 Get Disk Params. The full sequence is a bit convoluted so I'll post the source below. Normally, equipment config byte correctly identifies the number of drives, but the INT 13 AH=8 function is also called, which returns the number of drives in DL, and that is used instead. So it appears that neither method works on Compaq BIOS or GLaBIOS. I'm not sure if this is the result of an accurate emulation of very old BIOS behavior, or something else. INT 13 AH=8 is a pretty old function, but it might not exist pre-PC/AT.

We can make a quick fix and assume there's always one floppy in this case, and then set it's floppy type to 360k. But we kind of need to know whether GLaBIOS implements INT 13 AH=8 and something else is failing. If neither are present, the best answer would probably be to assume 1 floppy drive.

What is the method supposed to be used by programs to determine the number of floppies on GLaBIOS?

Here's the pertinent source for discussion:
```
int INITPROC bios_getfdinfo(struct drive_infot *drivep)
{
    int drive, ndrives = 0;

    /*
     * The INT 13h floppy query will fail on IBM XT v1 BIOS and earlier,
     * so default to # drives from the BIOS data area at 0x040:0x0010 (INT 11h).
     */
    unsigned char equip_flags = peekb(0x10, 0x40);
    if (equip_flags & 0x01)
        ndrives = (equip_flags >> 6) + 1;
        
    /* Floppy query may fail if not PC/AT */
    BD_AX = BIOSHD_DRIVE_PARMS; // =0x0800
    BD_DX = 0;                          /* query floppies only*/
    BD_ES = BD_DI = BD_SI = 0;          /* guard against BIOS bugs*/
    if (!call_bios(&bdt)) {
        int drives = BD_DX & 0xff;      /* floppy drive count */
        if (!drives && ndrives) {       /* handle Toshiba T1100 BIOS returning 0 drives */
            for (drive = 0; drive < ndrives; drive++) {
                printk("fd%d: default 720k\n", drive);
                *drivep++ = fd_types[FD720];
            }
            return ndrives;
        } else ndrives = drives;
    } else
        printk("fd: no get drive fn, ndrives %d\n", ndrives); // <-- we get here so INT 13 FN 8 seems missing

    /* set drive type for floppies*/
    for (drive = 0; drive < ndrives; drive++) {
        /*
         * If type cannot be determined using BIOSHD_DRIVE_PARMS,
         * set drive type to 1.4MM on AT systems, and 360K for XT.
         */
        BD_AX = BIOSHD_DRIVE_PARMS;
        BD_DX = drive;
        BD_ES = BD_DI = BD_SI = 0;      /* guard against BIOS bugs*/
        if (!call_bios(&bdt)) {         /* returns drive type in BL*/
            *drivep = fd_types[(BD_BX & 0xFF) - 1];
        } else { // <-- this is the default floppy setup, but not executed because both equip byte and FN 8 missing
            int type = (sys_caps & CAP_PC_AT) ? FD1440 : FD360;
            *drivep = fd_types[type];
            printk("fd%d: default %s\n", drive, type? "1440k": "360k");
        }
        drivep++;
    }
    return ndrives;
}
```
Note we are already handling a special case for Toshiba T1100 where the equip byte returns an accurate drive count but is missing INT 13 AH=8 function.

An easy fix would be to initialize `ndrives` to 1 at the top of the function, but if GLABios doesn't implement INT 13 AH=8 function, the code path will go into the T1100 section, setting the drive types to 720k, which we don't want.

@640-KB, what do you think?

640-KB, 2025-05-07:
Hello!  On the issue of bit 0 on the BDA (40:10), you probably know this just for the sake of background (warning TL;DR ahead)...

The 5150 originally used that bit (and corresponding DIP switch) to signify that there were no floppies installed, since the two floppy count DIP switches represent 1-4. On the 5160 that bit/switch was changed to represent the computer being in a manufacturer testing mode, and then some later clones dropped that altogether and just told motherboard owners to always leave it in the on position.  At that point, in 5160 BIOS and clones there was no direct way to configure the computer for 0 drives - only 1-4 was technically valid.  

And yes, later on when CMOS configured PCs introduced high density drives and made having 0 drives installed a valid configuration again.  Many (but not all) floppy BIOSes reintroduced using that bit 0 as a way to signify no drives installed, though at that point it was pretty inconsistent, so DOS tended to not put much meaning into it.  Some BIOSes with HD support might choose to read the bit from the PPI/DIPs as a way to decide whether or not there were drives and then could either choose to keep the bit as a 0 in the BDA or make it a 1 anyway for compatibility's sake (the latter is what the 8088_bios does).  Traditionally though the XT BIOS would just read the value of that switch and pass it through to the BDA 40:10 equipment bytes, which is what you're seeing there with Compaq, GLaBIOS, IBM and most other "IBM compatible" clones.

What actually triggered this issue to pop up in this testing was due to small issue in MartyPC where the for the Deskpro machine profile had implemented the 5160 PPI configuration, however the Deskpro 8086 actually uses the 5150's PPI configuration, so the BIOS's attempt to read the DIP switches resulted in the BDA config byte being `00`, resulting in that bit 0 potentially signaling 0 drives.

As for INT 13h AH=8, as you said that was not available on pre-AT machines, and only introduced for floppies for high density support.  There were no 8K XT BIOSes (that I'm aware of) that implemented it, and only the 16K 1986 IBM 5160 V3 did (and only partially so), as they only (officially) supported DD 360K drives.  For that generation of PCs, there wasn't a clean way to signify 0 drives, as most just used the position of the first switch to fill bit 0 without any specific meaning.

(end TL;DR)

I'd suggest that the lack of AH=8 being implemented for a floppy drive indicates that bit 0 does not carry any meaning for the drive count.  As in, if AH=8 returns a result of CF and AH=1 (bad command) you can assume that there can only be 1-4 drives installed and should use bits 6 and 7 of the BDA/INT 11h byte to get the drive count. Not having INT 13h, AH=8 and possibly seeing bit 0 of 40:10 being a 0 (potentially due to misconfigured motherboard DIP switches) is a valid scenario for the majority of XT-class machines.

Sorry for the long reply - let me know if that helps!

640KB

ghaerr, 2025-05-07:
Thanks @640-KB, I've fixed this with PR #2336.