ghaerr/elks #2117: Booting ELKS on recent x86 PC

rafael2k, 2024-11-28:
I'm trying to boot ELKS in my Thinkpad P70 (which is my main PC indeed) to use it bare metal for experimenting. As I don't want to write to my HD yet, I'm trying to boot using a pendrive (tried both floppy and HD images) and MEMDISK (to load the image from the pendrive).
I get:
[image]

Should I just dd the image to a proper SATA harddrive and try to boot, or I might encounter other problems too?

ghaerr, 2024-11-28:
I don't know what the boot problem is yet... it seems that hda may be proper, but hdb and hdc are probably incorrect. Not sure whether fd0 is correct, but looks like 1440k. Are you trying to boot from floppy, HD, or USB/pendrive?

If the latter, we've seen a number of issues where a CF card wouldn't work, long story short turned out to the the brand of CF card, strange.

> Should I just dd the image to a proper SATA harddrive and try to boot

That would be a clear path to see whether the kernel you've built will in fact boot on your PC, before having to debug pendrive issues, so yes.

rafael2k, 2024-11-28:
Right. I have my previous PC, a Thinkpad T430, I'll burn the image to HD and give a try.

rafael2k, 2024-11-28:
btw, hdb and hdc are kind of correct, I do have them (2 nvme drives), and they have many TB storage and I think CHS limited them to 8GB, which is OK.

toncho11, 2024-11-29:
This is the procedure for the HDD images on real hardware:
https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media

rafael2k, 2024-11-29:
Thanks @toncho11 !

ghaerr, 2024-11-29:
> The source of the DIVIDE by 0 detected by the kernel is not clear. In my case it was my own Amstrad BIOS. So long story short, if nothing else helps, the halt on divide by zero exception from the CPU can disabled and the kernel could with luck continue executing normally.

Thanks @toncho11, I forgot about that. Perhaps the recent Divide by 0 checking should be disabled.

@rafael2k, in order to disable it, comment out the line below in elks/arch/i86/kernel/irq.c:
```
void INITPROC irq_init(void)
{
    /* use INT 0x80h for system calls */
    int_handler_add(IDX_SYSCALL, 0x80, _irqit);

#ifdef CONFIG_ARCH_IBMPC
    /* catch INT 0 divide by zero/divide overflow hardware fault */
#if 1
    /* install direct panic-only DIV fault handler until known that
     * the _irqit version doesn't overwrite the stack
     */
    //int_handler_add(IDX_DIVZERO, 0x00, div0_handler_panic); // <--- comment this line out
#else
```

BTW @toncho11, your comment above seems to be missing from the GitHub web page on my system... is it still showing on yours?

rafael2k, 2024-12-04:
So, I found a harddrive and put it inside the notebook. DD'ed it to the hd (the hd32mbr-minix) and ELKS just booted in the speed of the light!
[image]

rafael2k, 2024-12-04:
[image]

rafael2k, 2024-12-04:
[image]

Now I'll go assembly the C toolchain to copy to the ELKS (appears as Old Minix!) partition.

ghaerr, 2024-12-04:
> I found a harddrive and put it inside the notebook. DD'ed it to the hd (the hd32mbr-minix) and ELKS just booted in the speed of the light 

Nice @rafael2k! So this is without the Divide by zero path above, right? That means that the original problem is likely something related to do with the pendrive or its related BIOS code, not ELKS. This isn't the first time there's been issues with getting ELKS booted on CF, pendrive or USB devices.

Sounds like this might be one of the fastest systems ELKS has been booted on yet! Glad its working for you.

rafael2k, 2024-12-04:
Indeed! This is stock ELKS, without any patch.

And yes, the problem is most likely related to using a USB pendrive and MEMDISK to load the image.

Yay! No need to spend 200 bucks on the Pocket/Book 8088 to run ELKS bare metal.
: )

bocke, 2024-12-04:
> Yay! No need to spend 200 bucks on the Pocket/Book 8088 to run ELKS bare metal.

I have 3 of those. Only one completely working. :( The other is half-working. The third was dead on arrival. My experience is you should really be careful what seller you order from. 

I am weak with electronics so I couldn't really fix them up even if I tried (I did). :)

But they are small and portable. So kinda useful.

I also have hp 200lx, but that can't boot external OSes. 

I've seen a utility called elksboot from the early 00s which allowed to boot elks from real mode DOS. I think I have a copy somewhere on my portable HDD where I archive software and docs. I wonder if it would work with modern ELKS?

rafael2k, 2024-12-04:
Hi @bocke. Didn't they change the defective unit for you? I read comments that the seller usually is open to solve problems. And indeed, these small 8088 notebooks seem very nice, apart of small and portable.

Good point about elksboot. I wonder if I can chain load ELKS from LILO or other boot loader. Did not try yet.

rafael2k, 2024-12-04:
btw, nxterm display garbled characters on real hardware here, while in qemu emulator all runs well.

rafael2k, 2024-12-04:
Some pics of nx apps. It seems the font rendering is broken somehow (the terminal works, but fonts are not ok).
[image]
[image]
[image]

ghaerr, 2024-12-04:
> I've seen a utility called elksboot from the early 00s which allowed to boot elks from real mode DOS.
> I wonder if it would work with modern ELKS?

Yes, it probably would work. All it would need to do is act like the BIOS boot loader: read the boot sector from a drive into RAM location 0:7C00 and jump there. 

> nxterm display garbled characters on real hardware here, while in qemu emulator all runs well.

Thanks for the pictures. The characters themselves are proper characters, but the wrong glyphs. This indicates that its not trying to paint garbage, but rather that the ROM font its getting the glyphs from is possibly mismatched.

We can look into the nano-X source to see where it thinks its getting the ROM characters from, IIRC from the BIOS or from a BIOS call.

ghaerr, 2024-12-04:
> is there an image (jpeg would be nice) viewer in any of the default images?

No. But if you'd like to write one, @tyama501 and myself are familiar with Nano-X. You'd want to use the Gr API and start with the viewer that's in the Microwindows repo (it was written later).

ghaerr, 2024-12-04:
> the  path of the viewer in Microwindows repo?

Here you go: https://github.com/ghaerr/microwindows/blob/master/src/demos/nanox/nxview.c.

Taking a quick look at it, I can see it uses some of the higher-level Gr helper routines for image loading, which aren't in 16-bit Nano-X. Those would also have to be ported, and they would be in src/nanox and/or engine/devimage.c. It'll get a little complicated possibly.

rafael2k, 2024-12-04:
Thanks. I'll take a look on it!
Btw, do you think I should open an issue about nx using incorrect glyphs in font rendering in my computer? Does this issue ever showed up in original computers with i86 and VGA?

tyama501, 2024-12-04:
Hi @rafael2k 

You can try ramfont in the makefile instead of romfont.

https://github.com/ghaerr/elks/blob/master/elkscmd/nano-X/Makefile

rafael2k, 2024-12-04:
Thanks @tyama501! It worked!
I just did as you said:
```
--- a/elkscmd/nano-X/Makefile
+++ b/elkscmd/nano-X/Makefile
@@ -53,7 +53,8 @@ ifeq ($(CONFIG_ARCH_PC98), y)
 DRIVERS += drivers/ramfont.o drivers/X6x13.o drivers/elksutilasm.o
 else
 ifeq ($(CONFIG_HW_VGA), y)
-DRIVERS += drivers/romfont.o drivers/elksutilasm.o
+# DRIVERS += drivers/romfont.o drivers/elksutilasm.o
+DRIVERS += drivers/ramfont.o drivers/X6x13.o drivers/elksutilasm.o
 else
 DRIVERS += drivers/ramfont.o drivers/X6x13.o drivers/elksutilasm.o
 endif
```
[image]
[image]

rafael2k, 2024-12-04:
> > So, I found a harddrive and put it inside the notebook. DD'ed it to the hd (the hd32mbr-minix) and ELKS just booted in the speed of the light (Not any operating system boots in 0.05s)!
> 
> One explanation of why you did not need any of https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media is that the default CHS in ELKS did match you hard drive. But if you start getting some read errors then you might want to verify this. It could be that the fonts are not read correctly (unlikely if you have only this problem).

Tks @toncho11. I think I'll put in the config  1023,255,63. But so far so good, the system seams to behave nice even without changing the CHS of the HD.

ghaerr, 2024-12-04:
> do you think I should open an issue about nx using incorrect glyphs in font rendering in my computer? Does this issue ever showed up in "from good ol' times" computers with i86 and VGA?

I'm glad @tyama501's suggestion worked. No need to open an issue. I've never seen this before, but I'm pretty sure the reason is that the character ROM in newer PCs isn't codepage 437, or has been updated. This version of Nano-X is 25 years old, and in the "good old days" the character ROMs were preferred since memory was so tight, but now can be overridden with RAM based fonts.

ghaerr, 2024-12-04:
> One explanation of why you did not need any of...  is that the default CHS in ELKS did match you hard drive. 
> But if you start getting some read errors then you might want to verify this.

> I think I'll put in the config 1023,255,63.

On anything but very old IBM PCs running ancient BIOSes, ELKS doesn't need a CHS specification, as it asks the BIOS for it, using two ways (BIOS and direct IDE probe), as @toncho11 points out in his wiki post. So it is not necessary to (re)configure ELKS for a specific HD in anything but ancient systems. However, for ancient systems that don't support BIOS INT 13h function 8 "Get Drive Parameters", the system can appear to be fine but then get nasty read (or other) errors, as @toncho11 also points out...

rafael2k, 2024-12-05:
Indeed, ELKS is correctly identifying the disks (even thou they are > 8 GB, and of course we can not access it entirely).
Is there a way I can dump the rom?

ghaerr, 2024-12-05:
> is there a way I can dump the characters rom?

Yes - I'm going to leave that to you, by finding out how Nano-X does it, and writing a program which does the same thing, writing output however you want rather than accessing it for the internal nano-X glyph display routines.

toncho11, 2024-12-05:
> On anything but very old IBM PCs running ancient BIOSes, ELKS doesn't need a CHS specification, as it asks the BIOS for it, using two ways (BIOS and direct IDE probe), as @toncho11 points out in his [...]

I updated the https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media

bocke, 2024-12-05:
> Hi @bocke. Didn't they change the defective unit for you? I read comments that the seller usually is open to solve problems.

Well, let's leave that story for another time. :)

> Good point about elksboot. I wonder if I can chain load ELKS from LILO or other boot loader. Did not try yet.

> Yes, it probably would work. All it would need to do is act like the BIOS boot loader: read the boot sector from a drive into RAM location 0:7C00 and jump there.

Oh, nice. I'll test it on 200 LX whe I get time. But I'm not sure it will work on palmtops. They sometimes have a bit specific hardware that is not completely IBM PC compatible.

I'll dig out the program and upload it somewhere if anyone is interested.

rafael2k, 2024-12-05:
> I'll dig out the program and upload it somewhere if anyone is interested.

At least I am.
: )

bocke, 2024-12-05:
> > I'll dig out the program and upload it somewhere if anyone is interested.
> 
> At least I am. : )

Uploaded:
https://archive.org/details/bootelks

rafael2k, 2024-12-05:
Thanks!
Committed to github here: https://github.com/rafael2k/bootelks

ghaerr, 2024-12-06:
> I'll dig out the program and upload it somewhere if anyone is interested.
> Committed to github here: https://github.com/rafael2k/bootelks

> Yes, it probably would work. All it would need to do is act like the BIOS boot loader:

A quick look at the source shows this probably won't work, my early guess was wrong. It looks for the boot image as "Image", instead of "linux", but that's easy to fix. It then reads that into memory, but looks like it is using old fashioned fixed-value, now incorrect, values for the locations of the INIT seg, SETUP seg, and SYS seg. So the kernel pieces will be loaded into the wrong locations in memory. (It doesn't try to read an HD boot sector which would be easier, but tries to boot an image called "Image" on the DOS filesystem instead. This allows for booting ELKS without a bootable ELKS filesystem somewhere already. Kind of neat idea, except that one needs an ELKS filesystem, even in DOS FAT format, in order for ELKS to boot, so one might as well make that filesystem bootable, and use the BIOS to boot, not DOS).

But the biggest issue is that entire approach it uses is non-portable: it uses an auxiliary program GETINT.ASM which reads the low-core interrupt vectors and saves them into a binary file, which is then blasted into low-core on the running DOS system before attempting the kernel transfer. The current ELKS needs/wants the low core vectors, which are essential to proper BIOS operation, left alone. Even if one created a special the .BIN file did happen to work on their one system, it would need to be reassembled and run again on any new system before elksboot would work on that system. 

The tools required look like they're all borland: TASM and TLINK, so the 8086-toolchain or NASM can't even be used to assemble bootelks. You gotta love the mess of incompatible assemblers and linkers out there, very few ASM programs are portable outside the assembler (and linker) they were using when written.

So - lots of problems. But when somebody wants to actually boot ELKS from DOS and dive into this, I'll be here to answer questions.

bocke, 2024-12-06:
> But when somebody wants to actually boot ELKS from DOS and dive into this, I'll be here to answer questions.

That's a bit out of my scope of my current knowledge and experience, but I would be interested in reading more if you catch time to write something up. Or/and know of any good references.

tyama501, 2024-12-06:
Hi @ghaerr 

BTW,
I think the released elks 0.8.1 is quite stable now.
Aren't we going to update the version number to 0.9.0-dev?

Thank you.

ghaerr, 2024-12-07:
Hello @tyama501,

> Aren't we going to update the version number to 0.9.0-dev?

Thanks for the reminder. Done in #2127.

rafael2k, 2024-12-13:
So, after some trial and error, I managed to boot ELKS on my ThinkPad T430 (which will be my ELKS machine), after changing SATA mode to Compatibility (instead of AHCI).
[image]

rafael2k, 2024-12-13:
And, for me to be able to upgrade ELKS, I'm dual-booting with Linux using Grub (pc-bios) and MEMDISK:
http://syslinux.zytor.com/wiki/index.php/MEMDISK

rafael2k, 2024-12-13:
For the record, here is where the system freezes when SATA is in AHCI mode:
[image]

And I'm not sure what I'm seeing, but with hd32mbr, I can see now about half of the files, many things missing, and I get very strange geometry at boot:
[image]

rafael2k, 2024-12-13:
Found. Just disabled "IDE hard drive CHS probe" and now I have ELKS up and running on the T430.

rafael2k, 2024-12-13:
[ELKS running native C toolchain](https://youtu.be/jYDMLP9tAq0)

rafael2k, 2024-12-13:
I also tested Nano-X in the T430, and the same issue with the ROM fonts is present. But no problem, as I switched back to using fonts in RAM.

ghaerr, 2024-12-14:
> I also tested Nano-X in the T430, and the same issue with the ROM fonts is present. But no problem, as I switched back to using fonts in RAM.

Since this has happened on both your systems, and apparently was required with @tyama501 as well, perhaps the change to RAM fonts should be made permanent. You're welcome to submit a PR with the Nano-X Makefile change, thanks!

ghaerr, 2024-12-14:
@rafael2k,

> Just disabled "IDE hard drive CHS probe" and now I have ELKS up and running on the T430.

Glad you were able to debug the boot! 

The history on this is that the BIOS is consulted for the CHS drive values, but some BIOSes report one cylinder lower than actual for various reasons. So a contribution was made that probes the IDE interface, if present, and uses that value instead. What seems to be happening on your system is that the IDE probe is returning 16,383 cylinders, which is way too high for ELKS to handle, our max is limited by ancient BIOS and architecture and is limited to max 1023 cylinders due to the way the BIOS INT 13h functions encode cylinders in the CX register.

> here is where the system freezes when SATA is in AHCI mode:

That looks like right where the IDE probe is taking place. I am betting it's hanging due to IDE incompatibility. 

Did disabling the IDE probe completely with ' # CONFIG_IDE_PROBE is not set` fix the problem, or something else? It looks like you're still getting an IDE probe message displayed during boot, so I'm a bit confused as to what exactly is going on.

We could add a check after the IDE probe routine that discards the IDE probe CHS if cylinders > 1023, which should solve this problem and remain compatible with older hard drives. I'm pretty sure that the newer HD BIOSes handle the business of faking up the CHS values anyways. However, if the IDE probe itself hangs your system, then this won't work.

The other alternative would be to add a 'ideprobe=0' option to /bootopts. I don't like this idea as much, as it isn't automatic and one can't edit /bootopts easily if they can't boot. If on the other hand you have to set your BIOS to Compatibility mode rather than AHCI, then perhaps no change is needed.

What do you think?

> How to force using BIOS geometry?

There is no way for ELKS to force BIOS geometry - you have to (re)configure the BIOS with a different CHS for that. At this time, ELKS is fairly dependent on BIOS for hard drive I/O, so ELKS takes what the BIOS reports, with the exception of the IDE probe above.

ghaerr, 2024-12-14:
> [ELKS running native C toolchain](https://youtu.be/jYDMLP9tAq0)

That is an amazing accomplishment, getting a full C toolchain running on ELKS :)

Has anyone else tried downloading and running the version 0.1 dev disk?

rafael2k, 2024-12-14:
@ghaerr,

> The history on this is that the BIOS is consulted for the CHS drive values, but some BIOSes report one cylinder lower than actual for various reasons. So a contribution was made that probes the IDE [...]

Got it. Keeping with BIOS value here.

> > here is where the system freezes when SATA is in AHCI mode:
> 
> That looks like right where the IDE probe is taking place. I am betting it's hanging due to IDE incompatibility.

Indeed, now with IDE probing disabled, I reverted the BIOS setting to AHCI and the system booted fine too.

> Did disabling the IDE probe completely with ' # CONFIG_IDE_PROBE is not set` fix the problem, or something else? It looks like you're still getting an IDE probe message displayed during boot, so I'm a bit confused as to what exactly is going on.

I did not post no picture of the boot after disabling IDE probe.  This fixed the issue, yes.

> We could add a check after the IDE probe routine that discards the IDE probe CHS if cylinders > 1023, which should solve this problem and remain compatible with older hard drives. I'm pretty sure [...]

That is a good idea. Then we limit ourselves to 8GB max HD right? Which is totally fine of course.

> The other alternative would be to add a 'ideprobe=0' option to /bootopts. I don't like this idea as much, as it isn't automatic and one can't edit /bootopts easily if they can't boot. If on the [...]

For practical purposes it makes no difference the mode -  in Compatibility mode the system boots and the fs is totally trashed (using that high count cylinder), and in AHCI the system hangs at IDE probe. With IDE probe disabled, just boots fine in both modes. Having this "ideprobe" option is good for users which don't want to recompile the kernel (I believe the pre-build images has IDE probe enabled, right?) , just a matter of mounting the image and change the bootopts.

> > How to force using BIOS geometry?
> 
> There is no way for ELKS to force BIOS geometry - you have to (re)configure the BIOS with a different CHS for that. At this time, ELKS is fairly dependent on BIOS for hard drive I/O, so ELKS takes what the BIOS reports, with the exception of the IDE probe above.

Got it. No need indeed, at least for both of my systems.

rafael2k, 2024-12-14:
> Since this has happened on both your systems, and apparently was required with @tyama501 as well, perhaps the change to RAM fonts should be made permanent. You're welcome to submit a PR with the [...]

Will do it

bocke, 2024-12-14:
Uhm, I was recently reminded of DOS [FIX8x14 utility.](https://www.bttr-software.de/products/fix8x14/). 

Apparently, VESA organization recommended for GPU producers to drop 8x14 fonts to make space for VBE3 and many did. So, the result of that was garbled fonts in some DOS applications with newer GPUs. Can this be connected to the same problem?

This utility was written to fix this. It comes with the source, but I haven't looked at it. I don't have currently where to test. :)

bocke, 2024-12-14:
Oh, it's a DOS device driver. Well, at least you might get an idea how to solve this, if it's really the same problem. :)

bocke, 2024-12-14:
> Since this has happened on both your systems, and apparently was required with @tyama501 as well, perhaps the change to RAM fonts should be made permanent. You're welcome to submit a PR with the Nano-X Makefile change, thanks!

If it's related to misssing 8x14 ROM font I mentioned above, I would recommend to leave option in, even if the ohter option is set to default. ROM fonts should work for older graphic adapters.

tyama501, 2024-12-14:
RAM font version is slightly bigger since it includes the font in every application.
I know these can fit in compressed FD1440 version but I'm not sure the other images afford these.  (It may be ok since there are no much nx application included in the other images

rafael2k, 2024-12-14:
Thanks for finding this information @bocke ! May be we can expose this finding with an option through some define - IF PC98 || VBE3: RAM, IF IBM_XT/AT (I think anything 286 or below): ROM.

ghaerr, 2024-12-14:
> So, the result of that was garbled fonts in some DOS applications with newer GPUs. Can this be connected to the same problem?

That is interesting information about shortening the height of the ROM charset. However, I don't think that is the problem here, since @rafael2k's screen shots do not show any garbled characters - only the wrong characters, but each glyph is drawn correctly. That seems to be a code page issue. Of course, loading an entirely different font into each executable makes it work, since both the font glyphs and codepage match the application(s).
 
> RAM font version is slightly bigger since it includes the font in every application.

I hadn't thought of that - it probably isn't a great idea to include a full ROM character set in each NX application, unless truly needed. We are almost completely out space on the distribution disks.

> May be we can expose this finding with an option through some define - IF PC98 || VBE3: RAM, IF IBM_XT/AT (I think anything 286 or below): ROM.

> we could place the font in the filesystem and open and read it when needed, so we just need one copy of it.

Contributions welcome - this would need to be done in the Nano-x engine, below the NX API layer. I'm thinking that perhaps the answer here is to have a configuration option for RAM fonts, CONFIG_NANOX_RAMFONT, so the Makefile doesn't have to edited manually. Just like the IDE probe issue discussed elsewhere, the reason for the (too many IMO) options in the config file is to allow the system to be (re)configured so that it can be made to work for *your* system, rather than trying to build a distribution that somehow is supposed to work on *all* systems.

rafael2k, 2024-12-14:
I like the CONFIG_NANOX_RAMFONT (and we put somewhere that information on VBE 3.0+ ROM fonts with Nano-X are known to be broken, vide @bocke link).

rafael2k, 2024-12-14:
btw, if I'm to send a PR to ELKS, which is the code style to follow? I need to properly set my emacs.

rafael2k, 2024-12-14:
So, as far as I understood, all GPUs that follow:

VESA BIOS EXTENSION (VBE) Core Functions Standard Version: 3.0 Date: September 16, 1998

or later have a higher chance of not having those fonts in the address int 10h returns anymore.

rafael2k, 2024-12-14:
I placed on github the fix8x14. The code to check the issue (in C-- !!):
https://github.com/rafael2k/fix8x14/blob/main/SRC/DIAG8X14/DIAG8X14.C--

ghaerr, 2024-12-14:
> if I'm to send a PR to ELKS, which is the code style to follow?

Currently, there are quite a few code styles in ELKS, due to the large number of contributed programs and other code over the last 20+ years. My best advice is to try to match the style of the code or subproject you are enhancing, especially including tabs vs spaces and 4 vs 8 column tab widths. For C code, the style varies, but most source files at least have a consistent style within each file. It's still a mess, but I'm working on it.

With regards to the kernel, I'm still in the process of "retabbing" various source files, as it is currently a big mess with 4 character spaces used followed by 8-column tabs, all mixed together. In fact, I have my editor set to move quickly between 4 and 8 column tabs, as well as insert spaces vs real tabs, all with 4 different keystrokes just to keep things readable. The retabbing process is slow, as I want to be able to track historical changes with other projects, and wholesale retabbing makes that a lot harder. This will be getting better, but is still a work in process since I took over!

For config.in and .config files, it's pretty easy to add yet another configuration option. However, over the last four years, I've been trying to remove configuration options, rather than add them. This is because of the many contributions that add a single option used once, then everyone forgets about the option and it piles up with tons of others, making maintenance harder with the contributor long gone, and no way to test. By whittling down the options tremendously over the last couple of years, I think we've achieved some success by making the system simpler, as well as more automatic. So I suppose we should continue to discuss the idea of whether we really need a config option for RAM fonts, it's pro's and con's, and also whether there might be a way to make it all automatically work instead of requiring (re)configuration for different users.

rafael2k, 2024-12-15:
reading the c-- code, it seems just call int 10 to get two different fonts, if there have the same address, bang, can not use the fonts in ROM:
  /* evaluate font addresses and display result */
  IF ((font8x14[2] == font8x16[2]) && (font8x14 == font8x16))
    DOSWRITESTRZ(, STDOUT, , #sameaddressmsg); // WE USE RAM
  ELSE
    DOSWRITESTRZ(, STDOUT, , #diffaddressmsg); // CAN USE ROM

ghaerr, 2024-12-15:
If it turns out that there's either an 8x14 font, or, instead oral a 8x16 font available via the BIOS, we might be able to get Nano-X to recognize either one and display whichever font happens to be present. This should be pretty straightforward. However, before going through that, we should make sure that the code page for the font includes the ASCII characters in their normal positions. 

I'm pretty sure our older version of Nano-X has a font header struct that lets the font engine know what the font glyph width and height is. So, if there were a way to determine the W/H of the BIOS font (possibly using the trick @rafael2k mentions that the C-- code is using), then this might be a good option to automatically use whatever font the BIOS actually has, without any configuration options.

bocke, 2024-12-15:
> Currently, there are quite a few code styles in ELKS, due to the large number of contributed programs and other code over the last 20+ years.

Can a code indenter/beautifier help? Like GNU Indent or AStyle? These kind of beautifers usually have prebuilt presets that can be used - which might be easier than setting every option. Other than that, it's likely some code still use K&R. Also, not suer how usable code formatters are for assembly files or files tha include inline assembly.

Just a food for tought for times you have a spare time to think about misc. stuff.

[14 more replies]

closed 2024-12-29 by ghaerr