ghaerr/elks #2156: The header asm-offsets.h is generated after it's first included

bocke, 2024-12-28:
I just checked out the last version of elks code from Github and the cross-compiler toolchain seems to build fine.

But the kernel build breaks while trying to compile some code dependant on asm-offsets.h. 

I redirected the stdout and stderr into a file called make.log and I see that asm-offsets.h is successfully generated on line 7032 by using this command:

```
awk -f asm-offsets.awk asm-offsets.s > ../../include/arch/asm-offsets.h
```

But the error happens because it's included before that line (before it's created).

This is line 7016 in the make.log:

```
ne2k-asm.S:33:0: fatal error: arch/asm-offsets.h: No such file or directory
 #include <arch/asm-offsets.h>
```

Make log is included here:
[make.log.gz](https://github.com/user-attachments/files/18265904/make.log.gz)

My host platform is Slackware Linux 15.0 64-bit.

bocke, 2024-12-28:
More info:

These are the contents of generated asm-offsets.h (elks/include/arch/asm-offsets.h):

```c
// This file is automatically generated from asm-offsets.c
// See asm-offsets.awk for details

#pragma once

#define TASK_KRNL_SP 2
#define TASK_USER_DS 876
#define TASK_USER_AX 860
#define TASK_USER_SS 880
#define TASK_USER_BX 862
#define TASK_USER_SI 870
#define TASK_USER_DI 868
```

toncho11, 2024-12-28:
I am able to build ELKS without any problems.
I updated the build documentation here: https://github.com/ghaerr/elks/blob/c241108ebf96f5ca8a9242a7e9b9f96e5a655060/BUILD.md

I am on ubuntu 24.04 on WSL.

bocke, 2024-12-28:
I'll look at it.  Will get back at you if problems persist.

Thanks. :)

bocke, 2024-12-28:
Unfortunately nothing really new in the updated docs.

bocke, 2024-12-28:
I found one thing that might be at fault. Just grepped my env:

```
env | grep -i make
```

And found I have MAKEFLAGS variable defined in my config files. It's useful for compiling distro packages. But on rare ocassions this can create problems with other software. I'll try to undefine this or start the kernel build in the clean environment.

ghaerr, 2024-12-28:
> But the error happens because it's included before that line (before it's created).

This sounds like one of those chicken and egg problems: the file ought to be in git, but then when its made, and differs, git shows a file difference after the build.

Sometimes this can be corrected by doing the build, then restarting the build without a "make clean" in between.

Have you got the ELKS build working yet, or is this issue still stopping the build on your system? It has been a long while since I built the system on a fresh download of the repo, but this problem is not occurring with subsequent builds on my system. Another thought is to save a copy of it once it's built, then remove the Makefile line that deletes this file on make clean. The file contents itself won't change on kernel builds unless major changes are made in the task structure in sched.h.

bocke, 2024-12-28:
Im currently building it from the start without MAKEFLAGS environment variable. I think existence of MAKEFLAGS forces make to do paralel build, which might not work with our current build system. :)

But will see. I'll get back to you when it's done.

ghaerr, 2024-12-28:
> I'll get back to you when it's done.

The build should only take 2-3 minutes - no need to rebuild the cross compiler every time.

bocke, 2024-12-28:
That was it.  Now it built fine. Only floppy disk creation failed. The built binaries are too big for fd1440.img:

```
mfs  /home/bocke/Downloads/elks/elks/image/fd1440.img genfs -a -1 -n14 -i256 -s1440 /home/bocke/Downloads/elks/elks/target
Blocks required 1456, available 1440
```

This is on clean tree, without my cc65-chess port. :)

bocke, 2024-12-28:
I'm marking this as resolved as the new problem doesn't have anything to do with the original problem.

ghaerr, 2024-12-28:
> The built binaries are too big for fd1440.img:

It seems that @toncho11 found the solution: Use "Select applications by image size" in .config. I'll make a note of that, as it seems the default configuration on a clean build is too big for a 1.44M floppy otherwise?

bocke, 2024-12-28:
Yeah, this works. 

It seems it's too big indeed.