ghaerr/elks #2170: elksdoom

parabyte, 2025-01-02:
im building elksdoom against latest elks sources and encountering this problem? 
[image]

ghaerr, 2025-01-02:
I'm unsure what the problem is, other than there might not be enough memory to run it, and then the error message given is not displayed correctly?

There is more information on how to increase the amount of memory available for ELKS to run Doom on https://github.com/ghaerr/elks/wiki/Increase-available-memory.

General discussion of running Doom on ELKS at https://github.com/FrenkelS/elksdoom/issues/2#issuecomment-2331922934, and some more here https://github.com/ghaerr/elks/discussions/1343.

parabyte, 2025-01-03:
built the elk libs for large memory model, thought i could just change the build process for elksdoom to be large model instead of the supplied medium memory model, and everything be super!

closed 2025-01-03 by parabyte