ghaerr/elks #2175: Wiki: non-working links

ghoffart, 2025-01-05:
**Description**
- When clicking on a sub-head-line in the TOC of the wiki, it should jump to the respective section. But, the sub-head-lines all have the link-target of the main page of the wiki.

**How to reproduce ?**
- Does the problem always happen ? Yes
- Click a sub-head-line-link in the wiki, e.g. »Serial Logins«, which links to https://github.com/ghaerr/elks/wiki/

ghaerr, 2025-01-06:
Thanks for pointing out this problem @ghoffart. I seem to remember running into this before and having trouble getting internal Markup tags to work properly, for some reason. IIRC it had something to do with section headers not working with spaces in them, requiring the section header itself to be renamed, which I didn't think was a great solution.

Are you aware of how this might be fixed without renaming the sections?

termdew, 2025-01-09:
I tried to fix it, please have a look if it's better now.

ghoffart, 2025-01-09:
Thanks! It’s muuuuch better now (e.g. https://github.com/ghaerr/elks/wiki/#boot-options), but 1 still doesn’t work (https://github.com/ghaerr/elks/wiki/#serial-console). I don’t see the heading for this, so I guess it’s a leftover from an older version?