ghaerr/elks #2182: Unused option in file copy

floriangit, 2025-01-12:
https://github.com/ghaerr/elks/blob/e9e445d6ecbf98fbd9c81d6a590b9d1c8a7016b5/elkscmd/file_utils/cp.c#L30

The option is evaluated but never set. Might either implement the option or delete the code.

ghaerr, 2025-01-12:
Nice find! I remember there was at one time a need for this, let me take a look to see what that might have been for. There was a time when we used zero-length '.keep' files in elks/target during the image creation process, but that was moved to image/Makefile handling.

floriangit, 2025-01-12:
Yes, it looks like it's getting propagated `external` in `elks/tools/mfs/protos.h` and (potentially) used by mkfs.c and genfs.c from the same directory.
I tested prefixing it with `static` in the original found location just to learn:
- the build went fine nonetheless - indicating I don't use mfs (which seems to be host tools to work on Minix FS)?
- why the compiler cannot emit a warning since it is set-once-but-never-changed now that I made it static?

floriangit, 2025-01-16:
thanks, let me check a bit more.

floriangit, 2025-01-18:
> I highly suspect that after you declared the variable static the code was silently deleted by ia16-elf-gcc;

Correct. The only way I could have the compiler keep the static-made opt_nocopyzero in the `.bss` section was to compile with `-O0`...I'll delete the unused code in a PR.

ghaerr, 2025-01-19:
Fixed by #2187.

closed 2025-01-19 by ghaerr