ghaerr/elks #2188: Elks Basic documentation

jgmy, 2025-01-19:
Hello!

Yesterday i tried to use Elks basic to do a simple calculation and i couldn't even exit from it.

Elks basic is not documented at all, and I don't even know where its source is. Is it based in another basic whose documentation we can read?

Yours,
José.

tyama501, 2025-01-19:
Hi @jgmy 
You can look README and source here.

https://github.com/ghaerr/elks/tree/master/elkscmd/basic

Exit is CTRL-D
Quit basic program is CTRL-C

Thank you.

tyama501, 2025-01-19:
snake.bas and test98.bas example only work for PC-98.
Other examples work for IBM too.

Thanks.

ghaerr, 2025-01-19:
ELKS Basic came from [Arduino Basic](https://github.com/robinhedwards/ArduinoBASIC), with a number of functions added.
Overall, it's fairly close to [Sinclair and ZX81 Basic](https://worldofspectrum.net/legacy-info/sinclair-basic/), with some variations.

I thought there might be an EXIT or QUIT command, but @tyama501's correct it seems the only way out is ^D.

jgmy, 2025-01-19:
Many thanks!
I swear I looked into the git tree, bit it seems I missed that folder.

I think a circle function should be both easy and nice to implement.

Also, why does it have a decimal-to-hex function (hex$) but not a
hex-to-decimal (&h or dec(a$) ?

Well, (sinclair basic had hex$ and bin$ but not reverse)

ghaerr, 2025-01-19:
> I think a circle function should be both easy and nice to implement.

Yes - you're welcome to contribute if you like. Shared code goes in basic.c and specialized host-specific code is in host-\*.c.

> Also, why does it have a decimal-to-hex function (hex$) but not a
hex-to-decimal (&h or dec(a$) ?

I'm not looking at the source right now, but sometimes there's a not-so-aptly-named function that does the reverse, and other times likely because the original Arduino Basic didn't have it and nobody needed it :) I was originally going to make Basic much more Sinclair-compatible but my other TODO-list got too long. You're also welcome to suggest functions which should be added or implement them also.

toncho11, 2025-01-19:
Here is my doc: https://github.com/ghaerr/elks/wiki/Coding-games-for-ELKS#3-using-basic

You can use basic in graphics mode on IBM class PC only in CGA mode.

toncho11, 2025-01-19:
@tyama501 I think on IBM PC the circle function is missing. I am looking at https://github.com/ghaerr/elks/blob/1fb2a47aa43dfbf5b77b82ccba0195d67860944f/elkscmd/basic/host-ibmpc.c#L285C1-L286C2

So this leads to partial support?

toncho11, 2025-01-19:
@tyama501 I just tested. One can draw lines, but no circles.

toncho11, 2025-01-19:
I am proposing an implementation:

```
void host_circle(int xc, int yc, int radius) {
    int x = 0, y = radius;
    int d = 1 - radius;

    // Draw initial points
    while (x <= y) {
        host_plot(xc + x, yc + y);
        host_plot(xc - x, yc + y);
        host_plot(xc + x, yc - y);
        host_plot(xc - x, yc - y);
        host_plot(xc + y, yc + x);
        host_plot(xc - y, yc + x);
        host_plot(xc + y, yc - x);
        host_plot(xc - y, yc - x);

        if (d < 0) {
            d += 2 * x + 3;
        } else {
            d += 2 * (x - y) + 5;
            y--;
        }
        x++;
    }
}
```

toncho11, 2025-01-19:
Now I saw that the README explicitly says that `circle` is not supported on IBM PC!

ghaerr, 2025-01-19:
> I am proposing an implementation based on code I found. 
> Now I saw that the README explicitly says that circle is not supported on IBM PC!

Yes, the `host_circle` function is empty in host-ibmpc.c - add that code and see how it works! :)

toncho11, 2025-01-19:
Tested. It works:

```
void host_circle(int xc, int yc, int r) {
    int x = 0;
    int y = r;
    int d = 1 - r;

    // Draw initial points
    while (x <= y) {
        host_plot(xc + x, yc + y);
        host_plot(xc - x, yc + y);
        host_plot(xc + x, yc - y);
        host_plot(xc - x, yc - y);
        host_plot(xc + y, yc + x);
        host_plot(xc - y, yc + x);
        host_plot(xc + y, yc - x);
        host_plot(xc - y, yc - x);

        if (d < 0) {
            d += 2 * x + 3;
        } else {
            d += 2 * (x - y) + 5;
            y--;
        }
        x++;
    }
}
```

I will make a PR later.

ghaerr, 2025-01-20:
> Tested. It works:
> I will make a PR later.

Cool! Post a screenshot too please, I'd like to see the quality of the circle using this simple integer algorithm!

toncho11, 2025-01-21:
Also the POINT(x,y) can be implemented for IBM/CGA for: https://github.com/ghaerr/elks/blob/master/elkscmd/basic/host-ibmpc.c

POINT(x,y) - checks whether the pixel at screen coordinates (x, y) is ON or OFF

```
// Function to get the color of a pixel at (x, y) using INT 10h
unsigned char get_pixel_color_bios(int x, int y) {
    unsigned int ax = 0x0D00;  // Function 0Dh in AH (read pixel), AL = 0 (unused)
    unsigned int bx = 0x0000;  // BH = display page (0 for Mode 13h)
    unsigned int cx = (unsigned int)x;  // X-coordinate in CX
    unsigned int dx = (unsigned int)y;  // Y-coordinate in DX

    int_10(ax, bx, cx, dx);  // Call the BIOS interrupt

    // The color is returned in the lower 8 bits of AX (AL)
    return ax & 0xFF;
}
```

If the result of the function is different from 0 than it means that this pixel is ON.

jgmy, 2025-01-21:
So screen drawing uses BIOS functions? I thought the 286 port used
protected mode.

ghaerr, 2025-01-22:
> So screen drawing uses BIOS functions?

Yes, currently ELKS Basic uses INT 10h for IBM PC and INT A0h-A8h for PC-98; graphics output is definitely SLOW as a result. However, the benefit is the same code works on CGA or EGA systems, as the hardware differences are handled by the BIOS, as is the mode switch in the mode switch in and out of graphics mode.

>  I thought the 286 port used protected mode.

There is no 286 port for ELKS - all binaries and the OS are compiled for the 8086 CPU, which allows the same distribution to run on 8088/8086, 80186, 80286, and 80386+ CPUs. The kernel never uses protected mode, although when XMS is configured and enabled, we take advantage of the "unreal mode" on 386+ CPUs for accessing memory > 1MB for use as system buffers.

ghaerr, 2025-01-22:
> Also the POINT(x,y) can be implemented for IBM/CGA

You're welcome to add the POINT function into basic.c, but when doing so, support should be provided for not only IBM PC, but PC-98 and host-stubs.c also (used for host basic).

Also, unless the compiler thoroughly optimizes the code written, I would suggest a speedy one-liner for get_pixel_color_bios as:
```
int host_get_pixel(int x, int y)
{
    return int_10(0x0D00, 0, x, y) & 0xff;
}
```

toncho11, 2025-01-31:
@jgmy Did you manage to try ELKS BASIC?
The documentation has been updated: https://github.com/ghaerr/elks/blob/master/elkscmd/basic/README.md

jgmy, 2025-01-31:
Hi!
I should recompile kernel first so I can use a Spanish keyboard.

closed 2025-02-24 by ghaerr