ghaerr/elks #2189: HD reading problem. CHS/LBA? or unhandled type 5 partition?

jgmy, 2025-01-19:
Various hd issues.

**Description**
Hi!
I tried to install elks in my 1990 AT computer in order to use the entire hard drive (more than 1024 cylinders) ,using a unix-like, free from bios, approach.

- First of all, fdisk detects the real disk size, but dd is unable of reading lt if /dev/hdaX is pointed to a region below the BIOS int13 CHS limit of 1024 cylinders.
- But, if selecting a space in the first 1024 cylinders aa /dev/hda2, doing dd if=hdimage of=/dev/hda2, it seems the disk is being written in LBA mode (instead of bios CHS).
- When using DOS utils to write the partition inside D: (/dev/hda2), it can't be mounted from Elks. It seems Elks can't handle fat disks inside type 5 (dos extended) partitions.(* note: I only copied the fat part of the elks 32 meg image. I skipped the 64k partition part).
- Changing partition type to 0x6 (fat) does not help (maybe because start of DOS logical unit is not he same as the start of partition).

**Configuration**
- Elks 0.8.1 commit 9bd1e045 15 oct 2024 22:40:41 -0600
- Hardware. 80286 processor,  1990 American Megatrend BIOS, VLSI motherboard, 1024kb memory,, booting from floppy, 

**How to reproduce ?**
- using fdisk, create /dev/hda3 starting on sector 1024, write partition. 
- Try to write to it-will fail.
- using fdisk, create /dev/hda1 type 06 from cyl 0 head 1 sector 1 to cyl 511 head 15 sect 63, and /dev/hda2 type 05 or 06 from cyl 512 head 0 sect 1 to cyl 1024 head 15 sect 63.
-  copy the elks hd32 fat image to a mounted /mnt (i did it putting my drive in another computer running windows)
- dd if=/mnt/hd32mb~1.img bs=512 skip=63 of=/dev/hda2 
- process fails.
- do the same from msdos (example: truncate image with qbasic, then write over D: with Norton Utilities or similar).
- reboot into ELS.
- mkdir /mnt/hda2; mount /dev/hda2 /mnt/hda2
Error:
Bioshd sector 2 access beyond partition.
Bioshd: I/O read error dev 302 lba sector 2
Minix: unable to read sb
FAT: me=0,csz=0,#f=0,floc=0,....
FAT: Unsupported format
Mount failed: unsupported argument.

**Raw data**
- Screenshot(s):
[image]

- Log file:
(Sorry I can't transfer log from 286 to my mobile phone without a hell of screwdriving horrors).

**Additional information**
Add any other context information about the problem here.

toncho11, 2025-01-19:
Installation with sys: https://github.com/ghaerr/elks/wiki#Installation
Writing and ELKS HDD image to real HDD: https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media

And I think I had this problem where I just did not know which partition type to set or it was useless to set the type of partition for FAT. I think it is checking the beginning of the partition, so the type was not used, there was something like that.

toncho11, 2025-01-19:
Mounting usually requires setting the type of the target partition. For FAT on ELKS use:

`mount -t msdos /dev/fd1 /mnt`

or 

`mount -t vfat /dev/fd1 /mnt`

One of them should work.

floriangit, 2025-01-19:
> [Writing and ELKS HDD image to real HDD: https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media](https://github.com/ghaerr/elks/wiki#Installation)

I've had issues with this part recently as well. This description just did not work for me anymore. It barked with `mkfs` that size 100000 (don't remember) either does not fit into 32k (but I did deploy fdisk before with more than that size in the partition) or it said it cannot do more than 0, which also gave no sense to me. Sorry, I don't remember, but can check if needed. I'm also not sure really, this helps with the original problem since I surely used everything possible without an offset or without even thinking about LBA/CHS etc. I do remember it worked in the past though ;-) When things are sorted once, `sys` and its `makeboot` is now a breath to use </ oops marketing detected.>

toncho11, 2025-01-19:
You did check: https://github.com/ghaerr/elks/wiki#remarks ?

So `sys` does work for you?

floriangit, 2025-01-19:
Yes and yes, I edited `sys` with help of @ghaerr in recent pull requests ;-) so I know it (now) well.
My comments only relate to Minix FS, I hope they don't distract the original issues...Chicken & Egg is also, that I'm conservative to power on my HDD and debug things, since it's ancient and I fear it breaks any given day.

toncho11, 2025-01-19:
Also try to limit the space you would like to use. Even if you have a lot of space, use mkfs or mkfat with less. For example 33000 which is 33MBs. If it complains, just adjust it. On ELKS you will never need more than 33 MBs. Big sizes might not be supported.

ghaerr, 2025-01-19:
> in order to use the entire hard drive (more than 1024 cylinders) ,using a unix-like, free from bios, approach.

ELKS doesn't (yet) support HD access free from BIOS, unfortunately. The BIOS is always used to access a hard drive currently. In addition, ELKS uses only the ancient CHS (cylinder/head/sector) BIOS method to access drives. The new LBA (logical block address) BIOS sector addressing method is not supported. This means that disks with more than 1024 cylinders are supported, since the original BIOS INT 13h function(s) are used.

I suspect this is the reason you're disk isn't working. The `sys` comments for initializing the disk contents, while correct, run at a higher level than the low level CHS disk addressing scheme used internal to the ELKS kernel.

> how to reproduce ?

I haven't studied your reproduction directions with too much detail just yet, because if your disk has more than 1024 cylinders, it can't be made to work at the moment until we 1) add a direct-access (non-BIOS) HD driver, or 2) your PC BIOS can be setup to specially map Heads or Sectors in a way that allows access past 1024 real cylinders while only using 1024 logical cylinders.

> It seems Elks can't handle fat disks inside type 5 (dos extended) partitions.

ELKS can handle DOS extended partitions of type 5, as well as Linux extended partitions of type 0x85, but the same restrictions apply - max 1024 cylinders for the disk in total.

> Changing partition type to 0x6 (fat) does not help (maybe because start of DOS logical unit is not he same as the start of partition).

The partition type is only used when checking for DOS or Linux extended partitions; it is not used for determining FAT vs MINIX filesystem types.

> mount -t msdos /dev/fd1 /mnt

This will work.

> mount -t vfat /dev/fd1 /mnt

Use "msdos" not "vfat" for mounting a FAT filesystem.

> [Writing and ELKS HDD image to real HDD: https://github.com/ghaerr/elks/wiki/Installing-HD-image-on-physical-media](https://github.com/ghaerr/elks/wiki#Installation)
> It barked with mkfs that size 100000 (don't remember) either does not fit into 32k 

The max size able to be given to mkfs is 65535.

> (but I did deploy fdisk before with more than that size in the partition) or it said it cannot do more than 0, which also gave no sense to me. 

I suspect the reason for saying "cannot do more than 0" is because of integer overflow on trying to convert an ascii number larger than 65535, and returning "0" instead.

> This description just did not work for me anymore. 

We should probably have somebody update the description in the Wiki if it is incorrect.

toncho11, 2025-01-19:
> The max size able to be given to mkfs is 65535.

But our example in wiki installation is: `mkfs /dev/hda1 180000`. So this must be reduced?

ghaerr, 2025-01-19:
> But our example in wiki installation is: mkfs /dev/hda1 180000. So this must be reduced?

Yes, the maximum that will work is 65535 (64MB). I just checked the mkfs source and it should give an error if 180000 is specified, but the Wiki should still be updated.

The mkfs usage line, which is displayed on error is:
```
Usage: mkfs /dev/name blocks (Max blocks=65535)
```

jgmy, 2025-01-22:
Finally! I think I have it now:
1) delete hda2 and re-create it as type 6 partition. This way I have 2 type
6 (FAT16 primary) partitions.
2) from DOS, format D:
3) from DOS, copy hd32fatimg.img to D, as sectors, skipping first cylinder
(starting at the second block that contains AA55 as boot4) signature).
4) reboot DOS. Test D: can be read.
5) using fdisk, set D: as boot.
6) reboot. Elks tries to boot
7) reboot from floppy.
8) mount /dev/hda2 into /mnt. Now, ELKS can mount /dev/hda2.
9) makeboot -M /dev/hda2
10) umount /mnt
11) sys /dev/hda2

In my computer, this does not work 100% ok, but I think it is due I made a
mistake in the size of the D: partition (I made it 120Mb long instead of
32Mb, so weird messages start to appear when system is copied to /mnt/bin).
But at last I could open /dev/hda2 from elks.

floriangit, 2025-01-22:
Your latest comment reads a bit like the dual-boot adventure I had to play some time back:
https://github.com/ghaerr/elks/wiki/Dual-boot-ELKS-and-DOS#ms-dos-50

jgmy, 2025-01-23:
Yes, the problem is I don't have a network card in this computer, so if I
want to install a system bigger than the floppy (for example, the man
pages) I have to copy from the hd32mbrfat.img to the harddrive.
And there is a mistery dd does not seem to work with partition 2, so I have
to switch to DOS and use Norton utilities instead.
If I could easily transfer files from a working linux machine (e.g. using a
network card or even a serial null-modem cable) I'll try to patch mount to
allow mounting of "loop" files (so I could mount my truncated version of
hd32mbrfat.img).
I have freedos utilities that can "loop" mount files, but I think most of
my freedos utilities are 32-bit, so there is no way I can use them.

Regards!

jgmy, 2025-01-23:
I finally understand what happens with dd.

dd if=file of=block_device returns "error, writing past end of partition"
if your file does not have the same size than the block device (being
either bigger, as the message suggests, or smaller).

I finally managed to pick a right size for my partition: ending in start+63
cylinders, head 15, sector 63, wich makes 63504 sectors, which is the size
of my trucated hd image (it seems I downloaded hd32fatmbr,img, so I
stripped out the first cylinder / first 32768 bytes).
The partition has to be class 6, so you have to make it from ELKS. and
there is the problem ELKS shell can't do calculations (to get the size of
the image) and the size of the image is too long to make calculations in
elksbasic (at least I get an error message when I try). I don't know if
UBASIC (a 80286 basic aimed at maths that could handle big numbers) can be
ported to elks

Many thanks for all your help.

ghaerr, 2025-01-23:
Hello @jgmy,

> I finally managed to pick a right size for my partition: ending in start+63
cylinders, head 15, sector 63, wich makes 63504 sectors,

Yes - from what I've read, certain MSDOS tools require that a partition start on a cylinder boundary, although I'm not aware that's a strict requirement in ELKS. Nonetheless, good idea - the ELKS fdisk does the same thing.

I am a bit surprised your HD required a heads setting of 15, rather than 16, though. 63504 sectors is 63 \* 16 \* 63, which would be 63 sectors, 16 heads and 63 cylinders, still skipping the first cylinder for partition start.

> The partition has to be class 6, so you have to make it from ELKS.

I have had an idea of making the ELKS `fdisk` program available as a host program. Would that have helped in this case? This would set the OS partition type as well as the CHS in the partition table. There are `fdisk` programs available on Linux and macOS, but I'm not sure whether they are able to set OS type 6 or not?

> and the size of the image is too long to make calculations in
elksbasic (at least I get an error message when I try).

ELKS basic calculates everything in floating point, so this should have worked. Do you have hardware floating point on your computer/CPU? I'd like to see the error message you got so we can chase this down.

Thank you!

sraase, 2025-02-05:
> I am a bit surprised your HD required a heads setting of 15, rather than 16, though.

Some drives use a head for servo information (or bad sector mapping). Such drives expose an odd number of heads in their physical geometry.

closed 2025-02-24 by ghaerr