ghaerr/elks #2198: Update binutils-ia16 and gcc-ia16 to latest tags ?

mfld-fr, 2025-01-22:
Hello, is there any reason to keep ELKS on some old versions of `binutils-ia16` or `gcc-ia16` ?

I made a try to build the cross tools from the latest tags available from @tkchia, with a host GCC version 14 in strict mode, and I ended in a script simpler than current one in `tools`, with a single patch applied to the GCC sources : https://github.com/mfld-fr/build86

By the way, it is a good thing to see that ELKS now relies on others cross build chains, because the pause in gcc-ia16 project is at risk.

ghaerr, 2025-01-22:
Hello @mfld-fr!

> is there any reason to keep ELKS on some previous versions of binutils-ia16 or gcc-ia16 ?

In general, since there are no known bugs or new features with the IA16 toolchain for ELKS, and rebuilding takes so much time and has introduced problems for some developers getting the compiler built, we haven't pushed to run "the latest versions". I'll take a look at your build86 script and patch to see what you've done. I'm personally still using macOS and clang, not host GCC v14.

> By the way, it is a good thing to see that ELKS now relies on other cross build tools, because the pause in the gcc-ia16 project is at risk.

Thanks. We're not yet using OpenWatcom for building any "standard" applications or the kernel, but of course having added the 16-bit OS/2 NE executable format supporting any number of code and data segments has greatly expanded the capability of programs runnable on ELKS. 

I had thought for a bit about using OpenWatcom to build the kernel, that would require rewriting all the assembly to yet a third ASM format, a quite laborious project for sure, which you know all about as you converted all of ELKS assembly to IA16 from BCC five years ago. (Our new C86 compiler toolchain started using NASM, a fourth ASM format, but we switched to AS86 w/enhancements, so now we're back to a "mere" three ASM formats for the project: GAS, WASM and AS86, and three object file formats. And there's no good tools for converting back and forth, and IA16's Intel-format output switch is completely buggy).

I have been becoming concerned that the IA16 GCC is appearing unmaintained, even on the new master at GitLab. What are you thinking about that?

mfld-fr, 2025-01-23:
> I have been becoming concerned that the IA16 GCC is appearing unmaintained, even on the new master at GitLab. What are you thinking about that?

I think we should avoid to get stuck as we has been with `bcc`. I tried to contribute to maintain `bcc` in @jbruchon 's `dev86` fork, at least fixing the build with a a modern compiler, but it quickly appeared to be a spaghetti plate, without the essential features to generate optimized IA16 code, and with a monolithic design that makes the addition of such features very difficult without rewriting whole parts.

I see that the story is starting to repeat with the recent build problems with `gcc-ia16`... it still does a good job, but how long will we be able to rebuilt it from sources ?

I am afraid I have no obvious solution for now 😕 ...

Oichkatzelesfrettschen, 2025-01-24:
> I think we should avoid to get stuck as we has been with `bcc`. I tried to contribute to maintain `bcc` in [@jbruchon](https://github.com/jbruchon) 's `dev86` fork, at least fixing the build with a [...]

If you can get to Belgium on February 1st it looks like you could do a presentation at FOSDEM. 
[GCC Mailing List Post Regarding](https://inbox.sourceware.org/gcc/875xparmhn.fsf@kataplop.net/)
[FOSDEM 2025](https://fosdem.org/2025/)

mfld-fr, 2025-01-28:
@Oichkatzelesfrettschen Agree `gcc-ia16` maintenance is worth to lobby for all the listed projects (and probably more, including mine 😄), but good luck to find a maintainer! I think better place to discuss that topic should be one of the `gcc-ia16` project mirrors.

Closing this issue as I got my answer to my initial question.

closed 2025-01-28 by mfld-fr