ghaerr/elks #2214: ELKS Web Browser

rafael2k, 2025-02-02:
I remember in the old times a friend Ricbit wrote FudeBrowser, a web browser which used per-rendered pages (from a proxy fudebrowser server) to use access web sites on Uzix (a Unix for MSX - a Z80 computer, also partly written by Adriano, which also studied at Campinas University some years before me and we talked sometime about it) - all in Z80 assembly:
https://github.com/ricbit/Oldies/tree/master/1999-05-fudebrowser

Then I of course remember about Arachne, a web graphical web browser for DOS 8086, which also got Linux support (SGVALib and SDL2). I was looking at its source code, and now had a brief talk about its port to ELKS:
https://github.com/sraase/arachne-browser/issues/3
https://www.glennmcc.org/

Also check (on how to compile):
https://github.com/sraase/arachne-browser/issues/2

Arachne on ELKS - it is trouble I know - remove all watt-tcp stuff - and what more to port - buuut, it would be cool - and it would certainly run - as it compiles for large memory model with Borland for DOS (and flat memory too on linux, but there are different code paths) and supports CGA, EGA and VGA.

[image]

sraase, 2025-02-03:
Have you looked at https://github.com/jhhoward/MicroWeb ?

In my opinion, Arachne isn't a good fit for ELKS.

rafael2k, 2025-02-03:
Very interesting the MicroWeb! Thanks @sraase. Indeed, it seems a better fit for ELKS.

toncho11, 2025-02-03:
Related to: https://github.com/ghaerr/elks/discussions/2185

rafael2k, 2025-02-03:
Thanks @toncho11. May be I close this one and we keep just the discussions page for this topic.

rafael2k, 2025-02-06:
I'll close this one, lets keep the conversion in "discussions" page.

closed 2025-02-06 by rafael2k