ghaerr/elks #2229: string.h issue with C86

rafael2k, 2025-02-17:
I have this here to be able to include string.h with C86:

'''
diff --git a/libc/include/string.h b/libc/include/string.h
index 0eac14ff..cd7285c6 100644
--- a/libc/include/string.h
+++ b/libc/include/string.h
@@ -28,8 +28,10 @@ void * memset(void*, int, size_t);
int memcmp(const void*, const void*, size_t);
void * memmove(void*, const void*, size_t);

+#if 0
void __far *fmemset(void __far *buf, int c, size_t l);
int fmemcmp(void __far *s1, void __far s2, size_t n); / Watcom C only, in ASM */
+#endif

/* Error messages */
char * strerror(int);
'''

ghaerr, 2025-02-17:
Wha's the problem is the  \_\_far keyword giving an error?

Are you running CPP86 with the -0 option, this is required to automatically set -D__C86__=1, which then defines \_\_ to be nothing in elks/include/arch/cdefs.h.

I can compile programs including <string.h> just fine using `ecc`. Make sure you're running the same options in your Makefile, if running each toolchain component manually, which isn't recommended. See examples/Makefile.elks for a standardized approach to this.

rafael2k, 2025-02-17:
Thanks @ghaerr. I'm "de-stashing" some changes here I dunno the reason to exist, and indeed, I re-compiled elks-viewer and it is all good. Bug on me.
: )

closed 2025-02-17 by rafael2k