ghaerr/elks #2231: fd360-minix.img build fail

rafael2k, 2025-02-17:
I'm building latest ELKS, and I get a build error for the small 360k floppy. If I understand well the error, we are running out of space.

```
rm -f /home/rafael2k/programs/devel/elks/image/fd360-minix.img
mfs  /home/rafael2k/programs/devel/elks/image/fd360-minix.img mkfs -1 -n14 -i128 -s360
mfs -v /home/rafael2k/programs/devel/elks/image/fd360-minix.img addfs Image.all /home/rafael2k/programs/devel/elks/target
Adding files from Image.all to /home/rafael2k/programs/devel/elks/target
image 1K blocks: boot 1, super 1, inodemap 1, filemap 1, inode 4, data 352
inodes inuse 1, free 126, overhead 1, total 128
blocks inuse 1, free 351, overhead 8, total 360
adding inodes: 20, blocks 180
mkdir /dev
mkdir /etc
mkdir /bin
cp /home/rafael2k/programs/devel/elks/target/bin/init /bin/init
cp /home/rafael2k/programs/devel/elks/target/etc/inittab /etc/inittab
cp /home/rafael2k/programs/devel/elks/target/etc/rc.sys /etc/rc.sys
cp /home/rafael2k/programs/devel/elks/target/bin/sh /bin/sh
cp /home/rafael2k/programs/devel/elks/target/etc/profile /etc/profile
cp /home/rafael2k/programs/devel/elks/target/bin/clock /bin/clock
cp /home/rafael2k/programs/devel/elks/target/etc/mount.cfg /etc/mount.cfg
cp /home/rafael2k/programs/devel/elks/target/bin/date /bin/date
cp /home/rafael2k/programs/devel/elks/target/bin/getty /bin/getty
cp /home/rafael2k/programs/devel/elks/target/etc/issue /etc/issue
cp /home/rafael2k/programs/devel/elks/target/bootopts /bootopts
cp /home/rafael2k/programs/devel/elks/target/linux /linux
cp /home/rafael2k/programs/devel/elks/target/bin/login /bin/login
cp /home/rafael2k/programs/devel/elks/target/etc/passwd /etc/passwd
mkdir /root
image 1K blocks: boot 1, super 1, inodemap 1, filemap 1, inode 4, data 352
inodes inuse 19, free 108, overhead 1, total 128
blocks inuse 179, free 173, overhead 8, total 360
mfs  /home/rafael2k/programs/devel/elks/image/fd360-minix.img genfs -a -1 -n14 -i128 -s360 /home/rafael2k/programs/devel/elks/target
Blocks required 361, available 360

make[2]: *** [Make.image:82: minixfs] Error 255
make[2]: Leaving directory '/home/rafael2k/programs/devel/elks/image'
make[1]: *** [Makefile:60: fd360-minix] Error 2
make[1]: Leaving directory '/home/rafael2k/programs/devel/elks/image'
make: *** [Makefile:17: all] Error 2
```

ghaerr, 2025-02-17:
Yeah, looks like it's one block too big. I'll take a look and post a fix shortly. In the meantime, just edit elkscmd/Applications and take something out that's got :360k in it. Also be sure your ELKS target /root (elkscmd/rootfs_template/root) is empty, since some of the C86 make stuff copies files over there for the time being.

rafael2k, 2025-02-17:
I already commented out from images Makefile the 360k disk, no problem at all.
: )

And indeed, it could be my fault. I'll start again from a clean tree.

rafael2k, 2025-02-17:
Uff, I'll boot ELKS on my notebook and keep focus there. 

I built from a clean tree - and all good.

closed 2025-02-17 by rafael2k

rafael2k, 2025-02-17:
ps: and you were right - it was C86 stuff...