ghaerr/elks #2236: suitability for HP 200LX?

ishotjr, 2025-02-22:
Has anyone run ELKS on the V20-based HP 200LX? I'd like to give it a try but the lack of floppy disk and requirement to instead write to PCMCIA (or CF) from Linux will be a bit more of a challenge, so I wanted to get a feel for feasibility first? 💗

ghaerr, 2025-02-22:
Hi @ishotjr, I should think ELKS will come up and run on the HP 200LX without problems. For your first pass, perhaps just grab a bootable image from the 0.8.0 Release page. Should there be issues, please report them and we might have to have you pull the repo and do a custom build, but we'll only know what is needed after seeing the boot screen.

Thank you!

fehertamaskyndryl, 2025-03-07:
An educational purpose Uni*x clone called MINIX is available for HP 100/200/700LX and 1000 CX palmtops. It requires significant amount of tweaking to run on the unusual 80186 CPU + PCMCIA hardware.

ishotjr, 2025-07-16:
there's up to 4MB of RAM (depending on model/minus whatever is allocated to the system) that's bootable, but I'm really interested to learn about the CF support and will gladly give that a try once merged! ❣️

toncho11, 2025-07-16:
> there's up to 4MB of RAM (depending on model/minus whatever is allocated to the system) that's bootable, but I'm really interested to learn about the CF support and will gladly give that a try once merged! ❣️

How would you boot ELKS? Or maybe an image of ELKS can stored in RAM and somehow ELKS can be booted from there?

ishotjr, 2025-07-16:
the device will boot to a PCMCIA card first if inserted, but I'm not sure if this is at the OS or BIOS level...

fehertamaskyndryl, 2025-07-16:
I remember running Minix on my 200LX, using a 24MB CF card in a PCMCIA adapter, before the LCD's flat cable was destroyed by an unrelated battery leak. Minix was fairly complete, it even had wheatstone, etc. performance benchmarks and text-based games, like advent. 

If I had a PCMCIA slot doubler for the palmtop, even TCP/IP could have been used in Minix, with the Accton EN-2216 or Silicom ethernet cards. I think the lack of convenient networking would be a big discouragement for trying to port Linux to 200LX, because the palmtop only has a non-standard infrared port and 115kbps RS-232 built in.

toncho11, 2025-07-16:
Here we are not talking about Linux, but ELKS. ELKS runs on anything from the original 8088 and above.
The tool needed is called elksboot or bootelks and it is discussed here: https://github.com/ghaerr/elks/issues/2117#issuecomment-2516516525 It even mentions hp 200LX. 

The tool is available here: https://github.com/rafael2k/bootelks
Honestly the procedure is not very clear. Maybe @rafael2k can elaborate more on that.
At least in theory it is doable.

toncho11, 2025-07-17:
BOOTELKS - Boot ELKS from DOS
=============================

This setup allows you to load the ELKS kernel from a DOS environment using BOOTELKS.COM.

REQUIRED FILES
--------------

1. BOOTELKS.COM
   - The DOS program that loads and starts the ELKS kernel.

2. IMAGE
   - The ELKS kernel image file (usually named 'Image').
   - Must be the raw kernel binary, not a floppy or disk image.
   - Can be compiled from ELKS source or obtained prebuilt.

3. INTS.BIN
   - A binary dump of the system's Interrupt Vector Table (IVT) and BIOS data.
   - Must be exactly 0x500 (1280) bytes, copied from RAM starting at segment 0x0000.
   - Created using the GETINTS utility described below.

4. GETINTS.COM
   - A small DOS program to dump the first 0x500 bytes of memory.
   - Run it under a clean DOS environment to generate INTS.BIN.

     Example usage:
     > GETINTS > INTS.BIN

SETUP INSTRUCTIONS
------------------

1. Boot into a clean DOS environment (MS-DOS or FreeDOS). Avoid loading memory managers or TSRs.

2. Place the following files in the same directory:
   - BOOTELKS.COM
   - IMAGE
   - INTS.BIN

   Example directory layout:

     A:\ELKS\
     ├── BOOTELKS.COM
     ├── IMAGE
     └── INTS.BIN

3. Run BOOTELKS:

     > BOOTELKS

   This will:
   - Load the IMAGE (ELKS kernel) into memory.
   - Restore BIOS state from INTS.BIN.
   - Transfer execution to ELKS.

NOTES
-----

- You must create INTS.BIN only once per machine (or BIOS environment).
- BOOTELKS is not compatible with compressed kernel images.

TROUBLESHOOTING
---------------

- If BOOTELKS hangs or resets, ensure:
  - The IMAGE is a raw, valid ELKS kernel.
  - INTS.BIN is correct and generated in a clean DOS state.
  - No drivers or memory managers (like HIMEM.SYS or EMM386) are loaded.

@rafael2k is this correct?

toncho11, 2025-07-17:
So you need a fat partition on the CF card which will contain the bootelks files and the ELKS' files needed to continue after the kernel is loaded such as /bin/bash. Kind of crazy.  And the kernel must know somehow (precompiled in it) to mount as root the CF card. As I said - kind of crazy.

ghaerr, 2025-07-17:
> So you need a fat partition on the CF card which will contain the bootelks files and the ELKS' files needed to continue after the kernel is loaded such as /bin/bash. Kind of crazy. 

Pretty straightforward - take an ELKS distribution in FAT format, and add bootelks.com to it. When running DOS, the CF card is readable, one would run bootelks.com. After booting, ELKS comes up and mounts the same CF card as a FAT filesystem, which the runs normally, producing a login: prompt and after a bash prompt.

> And the kernel must know to mount as root the CF card when its load and execution is done. 

I've added that capability in PR #2357, which is working on QEMU. However, that will ultimately need to be tested on real hardware, such as your Amstradt or any other modern PC with a CF card accessible via an ATA (or XTIDE or XTCF) interface.

> Quite a lot of things must go right.

I think the biggest issue is bootelks.com - no-one has run it for years. It may expect an older format of ELKS, I have no idea. I don't run DOS at all, and somebody needs to look at bootelks source code to understand what its internal loader is expecting. Not a big deal, just takes someone with a DOS box and some knowledge of compiling and playing around with ELKS. 

> This could first be tested on a regular desktop with c: and d:

Definitely try getting bootelks.com running on a desktop floppy DOS system first, using an ELKS distribution floppy with bootelks.com added to it.

rafael2k, 2025-07-17:
> # BOOTELKS - Boot ELKS from DOS
> 
> This setup allows you to load the ELKS kernel from a DOS environment using BOOTELKS.COM.
> ## REQUIRED FILES
> 
>     1. BOOTELKS.COM
>        
>        * The DOS program that loads and starts the ELKS kernel.
> 
>     2. IMAGE
>        
>        * The ELKS kernel image file. The ELKS' kernel is called "kernel" and must be renamed to "image".
>        * Must be the raw kernel binary, not a floppy or disk image.
>        * Can be compiled from ELKS source or obtained prebuilt.
> 
>     3. INTS.BIN
>        
>        * A binary dump of the system's Interrupt Vector Table (IVT) and BIOS data.
>        * Must be exactly 0x500 (1280) bytes, copied from RAM starting at segment 0x0000.
>        * Created using the GETINTS utility described below.
> 
>     4. GETINTS.COM
>        
>        * A small DOS program to dump the first 0x500 bytes of memory.
>        * Run it under a clean DOS environment to generate INTS.BIN.
>          Example usage:
>          > GETINTS > INTS.BIN
> 
> 
> ## SETUP INSTRUCTIONS
> 
>     1. Boot into a clean DOS environment (MS-DOS or FreeDOS). Avoid loading memory managers or TSRs.
> 
>     2. Place the following files in the same directory:
>        
>        * BOOTELKS.COM
>        * IMAGE
>        * INTS.BIN
>        
>        Example directory layout:
>        A:\ELKS
>        ├── BOOTELKS.COM
>        ├── IMAGE
>        └── INTS.BIN
> 
>     3. Run BOOTELKS:
>        > BOOTELKS
>        
>        
>        This will:
>        
>        * Load the IMAGE (ELKS kernel) into memory.
>        * Restore BIOS state from INTS.BIN.
>        * Transfer execution to ELKS.
> 
> 
> ## NOTES
> 
>     * You must create INTS.BIN only once per machine (or BIOS environment).
> 
>     * BOOTELKS is not compatible with compressed kernel images.
> 
> 
> ## TROUBLESHOOTING
> 
>     * If BOOTELKS hangs or resets, ensure:
>       
>       * The IMAGE is a raw, valid ELKS kernel.
>       * INTS.BIN is correct and generated in a clean DOS state.
>       * No drivers or memory managers (like HIMEM.SYS or EMM386) are loaded.
> 
> 
> [@rafael2k](https://github.com/rafael2k) is this correct?

I'm not sure, but if it works, lets add this to the README.

ghaerr, 2026-05-31:
It's possible the issue isn't INT 16h after all?

> what would you suggest instrumenting next? 

Did you try adding printk statements in the kbd-poll.c code?

The keyboard polling routine in kbd-poll.c uses a kernel timer whose callback actually calls INT 16 AH=1 (status) every 80ms (8 hardware timer ticks). Adding a printk in kbd_timer() before conio_poll() would allow you to see whether the hardware timer and kernel timer mechanisms are working. If so, then reworking conio_poll() in conio-bios.S would be the next step, to verify exactly what is returned from the dual calls to INT 16 (status, then read).

> Timer/debug variants showed the system can continue printing periodic/debug output after the shell appears, so the kernel is not simply frozen at the prompt.

For example, it may be that the INT 16 status call is hanging and never returns. It's also possible that the kernel timer callback is running every 10ms tick and interrupting a prior INT 16h call. So counting stack depth levels may also be needed (hopefully not).

> Would it be useful to compare ELKS’s BIOS console/timer path against Minix 2.0.x keyboard handling, since Minix keyboard input works on this machine?

I would suggest the above instrumentation first, to better determine which rabbit hole to jump into. There is the possibility that ELKS' messing around with the 8042 port addresses might be hanging the 200LX (see comments above for info on which files touch the 8042, they might also need instrumentation to track this down).

> Direct/scancode keyboard experiments did not successfully deliver input either. Some raw port tests saw stale/noisy-looking values such as port 64h=53 and port 60h=9C, but those did not translate into usable ELKS input.

Can the 200LX be setup for keyboard hardware interrupts? What does it use for a keyboard controller? What technical documentation is available for hardware access to the keyboard? The ELKS scancode driver assumes an IBM 8042 at the standard addresses, but it appears you're getting garbage returns, correct? The BIOS 16h works in DOS, but it may be that a DOS driver is emulating  INT 16. So I need to learn more about the actual 200LX hardware supporting the keyboard to be of more help.

Finally, yes, it would be a good idea to look at the MINIX keyboard driver to see what they're doing.

We'll get this working yet!

l00nix, 2026-09-28:
## Release 3.5: HP firmware keyboard support on a clean ELKS baseline

Thanks @ghaerr and @rafael2k for the encouragement and interest in upstreaming. Following the Release 3 CF-root update above, I've published [Release 3.5](https://github.com/l00nix/elks-on-hp200lx/releases/tag/hp200lx-release-3.5), with the HP changes separated onto a clean upstream-based source branch for review.

The main change is the keyboard approach. The working [GentleOS HP 200LX port](https://github.com/l00nix/gentleos-hp200lx) and historical MINIX-on-HP-200LX work encouraged keeping the HP firmware services alive. Release 3.5 preserves the firmware NMI vector and dispatches IRQ2 to the original HP ROM service. The firmware handles scanning and debounce, then produces PC-style scan codes that ELKS consumes through its IRQ1 keyboard driver and normal TTY path. The direct matrix scanner and idle-loop keyboard polling are no longer used.

Some relevant findings from the earlier interrupt investigation:

- **Firmware compatibility:** the ROM owns IRQ2 acknowledgement, so ELKS avoids a duplicate EOI. HP-specific guards suppress AT keyboard LED commands and XT port-61 acknowledgement strobes. Firmware RAM at physical `0x90000–0x9FFFF` is reserved, and the interrupt stack is increased to 1 KiB.
- **IRQ0 handoff:** a one-time guarded acknowledgement clears an inherited IRQ0 in-service state only when the PIC ISR is exactly `01` and none of the ELKS IRQ0/1/2 handlers has run. This happens before storage initialization. The original cause of that inherited state remains under investigation; this is not a claim that all timer questions are resolved.
- **CF support:** the HP byte-through-word transfer and socket initialization quirks remain isolated in the ATA patch. Upstream jiffy-based waits and LBA capability validation are restored.
- **Memory and loading:** the standard ELKS allocator is used with the firmware reservation. The DOS kernel load segment must remain `0340h` for this boot path: the initial extraction selected `00B0h` and failed early in boot; restoring `0340h` passed that failure on hardware. The precise underlying low-memory conflict is still unproven.

This uses BIOS/ROM services, but keyboard input is **not INT 16h polling**. ELKS retains its scheduler, filesystems and scan-code-to-TTY handling. No Mack MINIX source was copied into the new platform code; the reused DOS boot/loader chain retains its provenance in Richard L. Dubs's MINIX-on-HP-200LX work.

**Hardware results:** boot to a shell, Minix root mounting on `/dev/cfa1`, basic built-in keyboard input, display zoom and zoom panning are confirmed on the development candidate. The Release 3.5 kernel was rebuilt and byte-compared against that candidate: only two same-length version strings differ. The existing Release 3/3.1 userland is reused.

Modifiers/repeat, Ctrl-C, exact timer frequency, sustained CF writes, RTC initialization, shutdown/suspend and battery behavior still need validation for this implementation. Busy idle remains enabled. Build, platform mock, firmware-bridge emulation, DOS copy/error and load-address regression checks passed; those do not substitute for the remaining hardware tests.

The pinned upstream base is `69dfd4f274139ef1f533c646711db84902b0cfe4`, not a claim to track current upstream HEAD. The kernel and diagnostic utility changes touch 20 files, with 423 added and 5 removed lines, before documentation and packaging. HP runtime adaptations are guarded by `CONFIG_HP200LX_FW`.

- [Release 3.5 downloads and complete source kit](https://github.com/l00nix/elks-on-hp200lx/releases/tag/hp200lx-release-3.5)
- [Source branch](https://github.com/l00nix/elks-on-hp200lx/tree/hp200lx-release-3.5)
- [Changes from the upstream base](https://github.com/l00nix/elks-on-hp200lx/compare/69dfd4f274139ef1f533c646711db84902b0cfe4...hp200lx-release-3.5)
- [Separate platform, CF and load-address patches](https://github.com/l00nix/elks-on-hp200lx/tree/hp200lx-release-3.5/ports/hp200lx/release3.5/patches)
- [Architecture, build instructions and validation notes](https://github.com/l00nix/elks-on-hp200lx/blob/hp200lx-release-3.5/docs/hp200lx/RELEASE3.5.md)

Thanks again for pointing the investigation toward the timer and interrupt controller. This is now a much more focused set of platform changes to discuss for upstreaming.

ghaerr, 2026-09-28:
Hello @l00nix,

Thank you for the detailed report for your Version 3.5 implementation of ELKS on HP 200LX. Very nice work and I enjoy the technical information and discussion. I'll have to spend some time looking at your current source and can then give comments when you're ready for details on the best way we might incorporate into upstream.

> the interrupt stack is increased to 1 KiB.

Can you detail more specifics on the requirement found for a 1K interrupt stack? I would guess this is a result of the HP ROM and its requirements? If so, this poses a possible additional problem, which is, that not all hardware interrupts occur on the interrupt stack. That is, hardware interrupts occurring during user mode (application) execution will occur on the kernel stack associated with the running task - and that stack is only 640-700 bytes, so if 1K is actually needed, then a kernel stack overflow could occur with just a single interrupt, depending on details of HP ROM stack usage. 

> The DOS kernel load segment must remain 0340h for this boot path: the initial extraction selected 00B0h and failed early in boot; restoring 0340h passed that failure on hardware. The precise underlying low-memory conflict is still unproven.

I'm not sure I understand exactly what you're saying here, nor whether it's worth tracking anything down. The ELKS early boot sequence is a bit complex, but most load and copy addresses are configuration in the config.h file, which I'm sure are already modified for this port. Let me know if you'd like to discuss further.

Thank you!

l00nix, 2026-09-28:
Hello @ghaerr,

Thank you—your stack point is important, and I need to qualify the earlier report.

**The 1 KiB interrupt stack was a precaution, not a measured HP ROM requirement.** It was increased from 512 bytes to allow headroom for the ROM call and possible nested service during bring-up. I do not have a measured worst-case stack depth, a demonstrated 512-byte overflow, or an HP specification establishing that 1 KiB is necessary. The earlier development notes explicitly left worst-case stack usage unvalidated; I should have carried that qualification into the issue update.

I checked the current code against your observation:

- Only `INTRSTACK_BYTES` was increased. `KSTACK_BYTES` remains **700 bytes** in this configuration because `CONFIG_ASYNCIO=y`.
- `_irqit` retains the upstream stack selection: an interrupt from user mode uses the current task's kernel stack; an interrupt from kernel mode switches to the shared interrupt stack.
- The HP IRQ2 bridge saves registers, constructs the interrupt return frame and calls the saved ROM handler. It does **not** switch to a separate firmware stack.

So you are right: increasing the shared interrupt stack does not address the user-mode case, and the current successful hardware runs do not establish adequate task-stack headroom. The emulator tests use a synthetic ROM handler to check the calling convention and return behavior; they do not measure the real HP ROM's stack consumption.

[Stack definitions](https://github.com/l00nix/elks-on-hp200lx/blob/ea6d21377481de4ac11f01b7bedc181e6675afd6/elks/include/linuxmt/limits.h), [interrupt entry](https://github.com/l00nix/elks-on-hp200lx/blob/ea6d21377481de4ac11f01b7bedc181e6675afd6/elks/arch/i86/kernel/irqtab.S), and [ROM bridge](https://github.com/l00nix/elks-on-hp200lx/blob/ea6d21377481de4ac11f01b7bedc181e6675afd6/elks/arch/i86/kernel/hp200lx-fw-asm.S).

The next diagnostic I propose is to measure high-water usage and check guards on both stack types, including IRQ2 during a CPU-bound user process, kernel activity, typing and zoom/panning. I would want that evidence before proposing stack sizes upstream. Guidance on the preferred instrumentation, and whether a dedicated firmware stack would be appropriate if the measured demand is too high, would be very welcome.

**On `0340h`, my wording was imprecise:** I meant `REL_SYSSEG`, the final kernel relocation segment used by `setup.S`, not the initial DOS image load address.

When extracting the clean upstream-based port, I omitted the previous HP override in `config.h`. With this driver's configuration, upstream `REL_SYSSEG = DMASEGEND` evaluated to `00B0h`. That build reached early ELKS startup but printed `invalidate_buffers: skipping active block ...` and `open: No files`, before the HP firmware banner. Restoring the HP-only `REL_SYSSEG=0x0340` override, with no other functional kernel changes, allowed boot and CF-root mounting with the same DOS loader and root setup. Non-HP configurations still use `DMASEGEND`.

[The isolated correction](https://github.com/l00nix/elks-on-hp200lx/blob/ea6d21377481de4ac11f01b7bedc181e6675afd6/ports/hp200lx/release3.5/patches/0003-hp200lx-dos-load-base.patch) shows the configuration change plus identification strings. I have not identified an overwritten region or proved a low-memory conflict. “Retain the known-working relocation segment for this loader/platform combination” is the justified conclusion; “must remain” overstated it as a general requirement. I am happy to keep this as an explicit platform configuration while we establish which boot-layout details merit further investigation.

Separately, the 3.5.1 beta now boots with the RAMdisk driver compiled out, and `ls`, `cat`, `meminfo`, basic keyboard input, zoom and panning work on hardware. That is useful progress, but does not resolve the stack concern above.

Thanks again for catching this and for being willing to review the upstreaming approach.

ghaerr, 2026-09-28:
> Guidance on the preferred instrumentation,

I have prepared instrumentation to help you determine very specific max usages of both kernel and interrupt stacks. Start by re-compiling the kernel with the general CONFIG_TRACE=y trace options, which enable a number of interesting instrumentation capabilities (see include/trace.h for more details in general). Then, uncomment `kstack` and `istack` both or separately in /bootopts enable the instrumentation for usage of the kernel and interrupt stacks. This works by filling the kernel or interrupt stacks with 0x55 and then counting changed values at various times of interrupt processing; see elks/arch/i86/kernel/strace.c for more details.

With either option enabled, the system console will display line streams of information showing stack usage values, along with \* when a new high has been reached. Another useful option, but quite verbose, is `strace`, which instruments all syscalls and will also show the max stack used during each, which can be useful for determining system ROM usage for disk I/O or other services.

> KSTACK_BYTES remains 700 bytes in this configuration because CONFIG_ASYNCIO=y.

You may not need CONFIG_ASYNCIO set, which requires slightly more stack usage because of some ELKS drivers. Which driver are you using that requires async I/O?

> The emulator tests use a synthetic ROM handler to check the calling convention and return behavior; they do not measure the real HP ROM's stack consumption.

Hopefully the debug instrumentation can help you determine HP ROM stack usage, if the BIOS calls are performed during syscalls. Otherwise, you may want to use a similar technique used in strace.c::check_kstack() or check_istack() in your own instrumentation.

> whether a dedicated firmware stack would be appropriate 

Off the cuff, I would probably not recommend the complexity introduced by implementing a separate firmware stack; there are a few places in the kernel that are coded quite explicitly around kernel vs interrupt stacks (e.g. \_irqit of course), and introducing a third will require some serious inspection of kernel routines, which IMO isn't probably worth it. Better to determine required stack usages during various syscalls and other system operation and then increase kernel stack if needed, even at the expense of max tasks having to be decreased (because all kernel stacks have to live in 64K limited kernel data segment). The max tasks can also be dynamically configured with tasks= /bootopts.

Overall, it has been found that the highest kernel stack usage happens during execution of MSDOS FAT filesystem I/O requests, in particularly when longer multi-directory pathnames are traversed and I/O is required instead of retreiving from a kernel block I/O buffer.

> With this driver's configuration, upstream REL_SYSSEG = DMASEGEND evaluated to 00B0h.

I haven't had time to review exact changes/issues with changing REL_SYSSEG for your port, but yes, the setup.S boot loader is complicated with its requirement for copying the kernel a few times. I don't recall whether your boot process loads the boot sector and then executes the ELKS setup loader, or whether that's done separately. If you're running the ELKS setup, it also can be configured to display precisely each and every load and copy: this is enabled by setting debug_output=1 in elks/arch/i86/boot/setup.S. That output will then also stream to the console. (For any of the instrumentation options discussed other than setup.S, I suggest running serial console to capture the data displayed, since it scrolls quickly otherwise).

> ls, cat, meminfo, basic keyboard input, zoom and panning work on hardware.

Nice! Did you write a separate console driver to achieve that?

l00nix, 2026-09-28:
Thanks Greg, that's helpful. I've now done another run on the 200LX with 3.5.2c, including `while :; do :; done`, zoom/panning while it ran, and Ctrl-C to stop it. Ctrl-C worked and screen output still feels normal.

The probe around the ROM IRQ2 call recorded a peak of 202 bytes on the 700-byte task stack and 224 bytes on the 1,024-byte shared interrupt stack. It measured over 1,000 task-stack calls and 481 shared-stack calls, with no guard errors or rejected stack bounds. Every valid task-stack call is measured; the shared stack is sampled once in 32 calls. Measuring every shared-stack call had made screen output noticeably slow.

These are written-watermark depths, including the probe's six bytes of metadata and the existing interrupt/bridge overhead. They aren't measurements of the ROM alone or a proven worst case. The task peak went from 82 to 202 bytes with the longer run, so the extra exercise was worthwhile. I've left both stack sizes unchanged, and agree that a third stack would add complexity we don't need at this point.

I'll try `CONFIG_TRACE` with `kstack` and `istack` as you suggest, to get a broader picture of ELKS stack use. The HP ROM service here is called from IRQ2 rather than a syscall, which is why I put the small probe directly around that call.

For storage, I'm using `CONFIG_BLK_DEV_ATA_CF` with a Minix root on `/dev/cfa1`, and no floppy driver. `CONFIG_ASYNCIO` is enabled, but I haven't established that this driver needs it. I'll test disabling it separately.

The display uses the existing ELKS BIOS console driver (`CONFIG_CONSOLE_BIOS`), without a separate HP console driver. The HP changes keep the firmware keyboard service alive: ROM IRQ2 handles scanning/debounce and feeds scan codes into ELKS's normal IRQ1/TTY path. Zoom and panning work with that arrangement.

The boot path starts in DOS with `BTGVDX.COM`, loading the ELKS Image and its setup stage. It isn't a normal disk-sector boot. I've kept `REL_SYSSEG=0340h`; I'll use the `debug_output=1` trace before trying another relocation address.

I've put the [hardware results](https://github.com/l00nix/elks-on-hp200lx/blob/codex/hp200lx-release-3.5.2c/docs/hp200lx/hardware-results/2026-09-28-3.5.2c-meminfo.md) and [probe source](https://github.com/l00nix/elks-on-hp200lx/blob/hp200lx-release-3.5.2c/elks/arch/i86/kernel/hp200lx-fw-asm.S) up for review. Thanks for the pointers on where to look next.

ghaerr, 2026-09-28:
Interesting results on ROM IRQ2 stack usage, it'll be interesting to see how/whether that compounds usage shown via `kstack` and `istack`.

The CONFIG_BLK_DEV_ATA_CF driver doesn't require CONFIG_ASYNCIO - the lowest level ATA routines use programmed I/O for data transfer, and don't return until complete or error. The primary device driver requiring async I/O on ELKS is the direct floppy driver, which is required to be fully interrupt-driven. With your port only using the ATA block driver, disabling CONFIG_ASYNCIO will save a little bit of kernel data segment storage for the block I/O queue, but most importantly allow the kernel task stack size to safely be reduced to 640, which then allows for more kernel resources (nodes, file descriptors, heap) within the kernel data segment.

> ROM IRQ2 handles scanning/debounce and feeds scan codes into ELKS's normal IRQ1/TTY path. Zoom and panning work with that arrangement.

So the IRQ2 handler performs the hardware zoom/panning underneath the BIOS console driver, while also handling the keyboard hardware?

l00nix, 2026-09-29:
Thanks Greg. I checked the HP developer guide, and it answers the zoom/panning
question quite directly. The "Hotkey Sequences" section says those keys are
handled by the firmware code that emulates a keyboard controller, outside
INT 09h. It specifically lists Zoom and Menu+Arrow window movement.

The preceding section describes KBD and TIMER1 as IRQ2 sources. TIMER1 handles
debounce, builds the scan codes and triggers IRQ1 through the Hornet hardware.
That fits what we're seeing: keeping ROM IRQ2 running preserves the display
hotkeys, while ELKS uses its existing scan-code keyboard and BIOS console.

There is one distinction worth making. For automatic cursor tracking, the guide
says IRQ2 sets CurFlag and the BIOS INT 08h timer handler does the deferred
screen adjustment. We use ELKS's native IRQ0, so working manual panning doesn't
establish that the BIOS cursor-tracking path is fully maintained.

The original SDK sections are C3000011.HTM (keyboard preprocessing),
C3000012.HTM (hotkeys), and C3000023.HTM (display cursor interrupt). INT 10h
AH=D0h documents text zoom, and INT 15h AH=45h controls window keys and cursor
tracking separately. I haven't found a worst-case IRQ2 stack requirement in
those sections.

3.5.2d has now been tested on the palmtop with ASYNCIO disabled. Keyboard,
zoom/panning, screen speed and Ctrl-C during the busy loop all worked. The
small CF write/read test also survived a reboot, and cleanup worked afterward.
`shutdown -r` also works and returns the palmtop to DOS.

Task stacks are now 640 bytes, with the shared stack still at 1024. The busy
loop run measured 930 task calls, with a 202-byte written peak; the shared
peak was 222 bytes from 276 samples across 8,826 calls. Both guards and the
unknown-bounds count stayed zero. A second run after reboot gave the same
peaks with no guard errors. These are still workload measurements, not a
worst-case bound. The TASK allocation dropped by the expected 960 bytes.

One detail I caught is that the old CF copies of meminfo and ps use a compiled-in
task structure size. I used meminfo -s for these readings; full process listings
will need matching rebuilt tools. Broader kstack/istack tracing is still to do,
with the custom ROM probe disabled or made compatible with its marker scheme.

I've put the [guide findings and section references](https://github.com/l00nix/elks-on-hp200lx/blob/hp200lx-release-3.5.2d/docs/hp200lx/BIOS-KEYBOARD-DISPLAY.md)
alongside the [hardware results](https://github.com/l00nix/elks-on-hp200lx/blob/codex/hp200lx-release-3.5.2d/docs/hp200lx/hardware-results/2026-09-29-3.5.2d-meminfo.md).

l00nix, 2026-10-22:
Hi @ghaerr and everyone,

**[HP 200LX package 4.0 · based on ELKS 0.9.2](https://github.com/l00nix/elks-on-hp200lx/releases/tag/hp200lx-release-4.0) has been released.**

The update includes enhanced HP BIOS integration for keyboard input, zoom, panning, cursor focus and screen inversion, plus guarded sleep/wake. A dual-kernel DOS launcher supports either a CF root filesystem or a local filesystem image on C:, leaving the PCMCIA slot available for Ethernet.

Networking with the Accton EN2216 R02 and the native ELKS `ktcp` stack has been tested on both single- and double-speed palmtops, including DHCP and Internet access. **[ELKSWeb](https://github.com/l00nix/ELKSWeb)** successfully loaded and rendered [68k.news](http://68k.news) and [FrogFind](http://frogfind.8088retro.com), including GIF images. The package also includes the [ELX utilities](https://github.com/l00nix/elks-on-hp200lx/blob/hp200lx/docs/hp200lx/ELX-TOOLS-4.0.md), including the ELXdash dashboard shown below. A separate [ELX tools download](https://github.com/l00nix/elks-on-hp200lx/releases/download/hp200lx-release-4.0/hp200lx-4.0-elx-tools.zip) is available with the release.

More detailed updates, installation instructions and the [4.0 changelog](https://github.com/l00nix/elks-on-hp200lx/blob/hp200lx/docs/hp200lx/RELEASE-4.0.md) are available in the [ELKS on HP 200LX repository](https://github.com/l00nix/elks-on-hp200lx).

Thank you for the guidance and support during development!

*ELXdash on the actual HP 200LX — development photograph.*

[image]

*ELKSWeb welcome page on the actual HP 200LX — development photograph.*

[image]