ghaerr/elks #2237: [OpenWatcom] Can not link with libc compiled in small memory model

rafael2k, 2025-02-23:
Trying to link software using the small-memory model libc gives me:
```
owcc -c -O2 -mcmodel=s -bnone -march=i86 -std=c99 -fno-stack-check -fnostdlib -Wc,-fpi87 -Wc,-zev -Wc,-zls -Wc,-x -Wc,-wcd=303 -Wall -Wextra -I/home/rafael2k/programs/devel/elks/libc/include -I/home/rafael2k/programs/devel/elks/elks/include -I/home/rafael2k/programs/devel/elks/libc/include/watcom -D__ELKS__ -DNDEBUG -o ppmview.obj ppmview.c
owcc -bos2 -s -Wl,option -Wl,dosseg -Wl,option -Wl,start=_start -Wl,option -Wl,nodefaultlibs -Wl,option -Wl,stack=0x1000 -Wl,option -Wl,heapsize=0x1000 -o ppmview ppmview.obj graphics.obj utils.obj -Wl,library -Wl,/home/rafael2k/programs/devel/elks/libc/libc.lib
Error! E2052: file /home/rafael2k/programs/devel/elks/libc/libc.lib(/home/rafael2k/programs/devel/elks/libc/stdio/vfprintf.c): relocation at 0001:14bf not in the same segment
Error! E2052: file /home/rafael2k/programs/devel/elks/libc/libc.lib(/home/rafael2k/programs/devel/elks/libc/stdio/vfprintf.c): relocation at 0001:15aa not in the same segment
Error: Linker returned a bad status
make: *** [Makefile.owc:51: ppmview] Error 1
```

ps: I build the watcom libc after setting watcom.model to "MODEL=s".

ghaerr, 2025-02-23:
Can you try commenting out the following line in libc/stdio/Makefile and then rerunning the entire OWC libc make, then your own link? I would like to know if this changes the error message or makes it go away:
```
$(LIB): $(LIBOBJS)
    $(RM) $@   
ifeq "$(COMPILER)" "watcom"
ifdef FIXOMF
    # fixomf -w dtostr_,ptostr_ vfprintf.obj <--- ADD '#' to front of line here
endif
endif
    $(AR) $(ARFLAGS_SUB) $@ $(LIBOBJS)
```
Make sure to do a full clean:
```
make -f watcom.mk clean
make -f watcom.mk
```

ghaerr, 2025-02-23:
@rafael2k: never mind the comment out above, I have this problem duplicating already and am looking into it.

rafael2k, 2025-02-23:
It linked!

ghaerr, 2025-02-24:
> It linked!

Nice! Looks like I have a problem in the magic OBJ file post processor `fixomf`. Go ahead and check whether the programs runs without issues while I figure out what to do. What is likely happening at this point is that vfprintf is dragging in dtostr and ptostr which are floating point and ptick conversion routines, which aren't needed but just make the program bigger. All of this is part of trying to get weak symbol support working, which looks like it works in large model only.

ghaerr, 2025-02-24:
The problem turned out to be inherent to the way that "weak" symbols are generated by OWC, not a fixomf problem. With #2238, all models should link, with the caveat that small and compact model programs can't automatically display floating point.

This shouldn't be a big issue because OWC floating point support is limited to machines that have an FPU, with no soft float emulation like that provided by ia16-elf-gcc. So any OWC programs that use floating point may not run on all machines and won't give an error either.

closed 2025-02-24 by ghaerr

rafael2k, 2025-02-24:
Thanks for this. Now ELKS-VIEWER compiles and run using owc small memory model. As the jpgview still has problems with C86, I released version 0.2.1 of elks-viewer with official binaries now compiled with owc small model.
https://github.com/rafael2k/elks-viewer/releases/tag/0.2.1

PR #2238 (2025-02-24): [owc libc] Fix OWC link errors with small and compact models using _weaken
Fixes #2237.

OWC wlink was failing to link vfprintf.obj in small and compact models due to the mechanism used to make "weak" symbols work using the \_weaken macro, because the OWC compiler generates a far segment relocation entry which causes wlink to fail with a "relocation not in same segment" error.

This all happens because vfprintf attempts to be smart about dragging in the large dtostr and ptostr routines into printf when floating point/ptick display aren't used. One fix would be to always link in dtostr in small [... cut]