ghaerr/elks #2249: No graphics support for BASIC when compiled with WATCOM

toncho11, 2025-03-03:
https://github.com/ghaerr/elks/blob/6b091b73c216d7ae1e68804b76e83942819fe993/elkscmd/basic/host-stubs.c#L31C1-L32C2

I have not checked executing but when compiling with openwatcom: https://github.com/ghaerr/elks/wiki/Using-OpenWatcom-C-with-ELKS
there is no watcom compatible assembler for switching to a VGA model.

toncho11, 2025-03-03:
It should be something like that in host-ibmpc.c. I am not sure what is the objective of this gmode and exit code. 

```
#ifdef __WATCOMC__

void host_mode(int mode) 
{
    gmode = mode;

    if (gmode && !exit_on) {
        atexit(mode_reset);
        exit_on = 1;
    }

    if (gmode)
        int_10(VIDEO_05_G320x200, 0, 0, 0);
    else
        int_10(VIDEO_03_T80x25, 0, 0, 0);
}
#else
int host_mode(int mode)
{
_asm{
                push si
                push di
                push bp
                push es
                cli
                mov ax,[mode]
                int 0x10
                sti
                pop es
                pop bp
                pop di
                pop si
        }
}
#endif
```

toncho11, 2025-03-03:
Or better:

#ifdef __WATCOMC__
#pragma aux int_10 = \
    "cli" \
    "int 0x10"      \
    "sti" \ 
    parm [ax] [bx] [cx] [dx] \
    modify [ax bx cx dx];
#else
extern void int_10(unsigned int ax, unsigned int bx,
                   unsigned int cx, unsigned int dx);
#endif

ghaerr, 2025-03-03:
FYI: there is no point in adding `CLI` before INT 10h in all cases - all interrupt routines start with interrupts disabled, and all return to the previous interrupt state on the RETI instruction.

Adding CLI/STI doesn't "save" the ELKS kernel from a multitasking switch in the middle of the call, if (as is the case in almost all BIOS INT calls) the BIOS reenables interrupts with an STI in the routine. 

Regardless, the CLI/STI pairs should be removed from all BIOS INT calls. You'll notice there are none in the C86 graphics routines I wrote last month.

ghaerr, 2025-03-03:
> It should be something like that in host-ibmpc.c

Again, I'm not seeing what purpose there is in this, since basic is already working with graphics. But if you desire to add additional compiler support, I recommend that instead of polluting the files with hard-to-follow multiple ifdefs and internal ASM language, all of this get moved to separate C (or ASM) files, each for a different compiler or assembler. In this way, we "write once" the proper routine for the compiler, then link it with the associated programs "library-style", rather than continuing to hack ifdef code into each and every separate program that needs the functions.

closed 2025-03-03 by toncho11