ghaerr/elks #2264: XMS RAM drive

toncho11, 2025-03-19:
I autogenerated the following code for elks/arch/i86/drivers/block/xmsram.c:

```
#include <linuxmt/device.h>
#include <linuxmt/fs.h>
#include <linuxmt/sched.h>
#include <linuxmt/errno.h>
#include <linuxmt/string.h>
#include <linuxmt/bios.h>

#define XMS_SECTOR_SIZE  512   // Standard sector size
#define XMS_BLOCKS       2048  // Number of blocks (1MB RAM drive)
#define XMS_BASE_ADDR    0x100000 // Starting address of XMS (1 MB)

static unsigned long xms_base_addr = XMS_BASE_ADDR; // XMS base address for RAM drive

// Function to query available XMS
static unsigned long query_xms_size() {
    struct biosregs r;
    initregs(&r);
    r.ax = 0xE801;  // Query Extended Memory
    intcall(0x15, &r, &r);

    if (r.flags & 1) return 0; // Error
    return ((unsigned long)r.ax) * 1024; // Size in bytes
}

// Read sector from XMS memory
static int xms_read(struct device *dev, char *buf, int block) {
    if (!xms_base_addr) return -ENXIO; // No memory allocated

    unsigned long addr = xms_base_addr + block * XMS_SECTOR_SIZE;
    memcpy(buf, (void*)addr, XMS_SECTOR_SIZE);
    return 0;
}

// Write sector to XMS memory
static int xms_write(struct device *dev, char *buf, int block) {
    if (!xms_base_addr) return -ENXIO; // No memory allocated

    unsigned long addr = xms_base_addr + block * XMS_SECTOR_SIZE;
    memcpy((void*)addr, buf, XMS_SECTOR_SIZE);
    return 0;
}

// Device operations
static struct device xms_device = {
    .name = "xmsram",
    .read = xms_read,
    .write = xms_write
};

// Initialize the XMS RAM drive
static int __init xms_init(void) {
    unsigned long xms_size = query_xms_size();
    if (xms_size < (XMS_BLOCKS * XMS_SECTOR_SIZE)) {
        printk("XMSRAM: Not enough extended memory! Available: %lu bytes\n", xms_size);
        return -ENXIO;
    }

    dev_register(3, &xms_device);  // Register as `/dev/xmsram`
    printk("XMSRAM: Initialized with %lu KB\n", (XMS_BLOCKS * XMS_SECTOR_SIZE) / 1024);
    return 0;
}

module_init(xms_init);

```

ghaerr, 2025-03-19:
@toncho11: what is the purpose in posting this code? Do you understand it? It won't compile and link, and certainly won't work.

toncho11, 2025-03-20:
It is not for compile and run, but rather if this is the right direction to create block device that represent the XMS memory.

toncho11, 2025-03-20:
OK

ghaerr, 2025-03-20:
>  I see I need to use register_blkdev instead and also adapt this structure to the above driver

In general, it seems AI isn't very helpful writing fully usable code unless there's thousands of slightly similar examples of what's being asked to be done in its training set. For code suitable for ELKS, this would be hard to find. Admittedly though there's likely a huge set of code and algorithms being "rewritten" daily by programmers and AI software all doing the same things. IMO, this is a unsolved problem in computer software in general, and might be better solved by redesigning the way software systems are put together, or common design requirements reused. But I suppose it's inevitable that machines start writing their own software.

Back to ELKS, I have been thinking about writing the XMS ram disk driver, and have found that the SSD driver's "test" module ssd-test.c (used to test the block device) should be easily convertible to using XMS instead of main memory. I plan to start on getting a working version very soon.

toncho11, 2025-03-20:
Does the code uses the correct way to access the XMS through BIOS 15H?

toncho11, 2025-03-20:
Looking at https://github.com/ghaerr/elks/blob/9719bd2ceb5ce185c75ba572b389c23ae0265186/elks/arch/i86/mm/xms.c
It seems more complicated.

ghaerr, 2025-03-20:
> Does the code uses the correct way to access the XMS through BIOS 15H?

No. It seems it just kind of made up the 'struct xms_move' structure. The BIOS function requires a GDT pointer. Who knows where it got the ideas for those mixed up.

The first version you posted showed the use of the BIOS INT 15h Function E801h "Query Extended Memory" which could be a good idea, but it got that wrong too, as function E820h supersedes E801h and is used by all modern BIOSes. So basically looking through this kind of AI code is mostly a waste of one's time, and possibly shows people the wrong thing to be learning.

> Maybe these: int15_fmemcpyw and linear32_fmemcpyw routines can be reused?

Definitely. They're already both wrapped in the function xms_fmemcpy which is used throughout the kernel. Since XMS copies are already implemented in the kernel for the buffer subsystem, doing the actual XMS<->RAM copy is the easiest part of enhancing the ram disk driver to work with XMS. Take a look at elks/arch/i86/block/ssd-test.c - it already uses these functions. What's missing is the setup and initialization of the RAM disk parameters from an XMS perspective, rather than main memory. That will be quite straightforward to write by using a new ssd-xms.c which will replace ssd-test.c. Coming shortly.

toncho11, 2025-03-21:
Looking forward :). Thank you.