ghaerr/elks #2266: urlget block in bind() at second run

rafael2k, 2025-03-21:
Running urlget twice to a remote http server, it gets blocked in bind(), and system freezes.

It seems that when the ftpd server is running, urlget keeps working as expected after many runs.

I observe the same behavior in other software, like bobcat.

rafael2k, 2025-03-21:
So, if there is any daemon running (be it telnetd or ftpd), urlget (and other client software) works fine. SO_LINGER and SO_REUSEADDR did not help.

ghaerr, 2025-03-22:
I'll take a look, my initial guess was that SO_REUSEADDR was needed as that's been an issue before, depending on which side of the network connection closed the connection first, and the TCP timer countdowns.

What exact command line are you going `urlget` that experiences the hang?

rafael2k, 2025-03-22:
Thanks @ghaerr. I just do a urlget to the server in the linux host running qemu (stock nginx install).
urlget http://10.0.2.2

ghaerr, 2025-03-22:
> I just do a urlget to the server in the linux host running qemu 

I don't have a host web server, but I was able to duplicate the problem using outgoing telnet to a host telnet daemon. 

How are you certain the program hangs in `bind` and not `connect`? My duplication shows that the system isn't hung, as one can login from another terminal, and things work fine. Permanent network hang issues have been described in #1536 where the application is in a network "hung" process state - unfortunately not easily unhung (read #1536 for details).

Nonetheless, the reason for the bind/connect hang occurring regularly on a 2nd connect should probably be able to be tracked down and fixed, this is definitely a bug.

> ps: @Mellvik is working on a improved ktcp. Could ELKS import the improvements if they provide better stability for networking on ELKS?

Of course. TLVC is an ELKS fork from a couple years ago where all "ELKS" strings and directories were renamed TLVC, but 95% of the code from the time of the fork is identical. The big TLVC improvements are in direct (non-BIOS) HD block and raw device drivers, which unfortunately are not compatible with ELKS block drivers, and also enhancements and maintenance of existing and additional network device drivers, which are likely much more close to compatible. I have not been able to test or import any TLVC NIC drivers because to this day I still don't have a working NIC card on real hardware (@Mellvik and I talked about this years ago and has offered his help, being a huge user on real hardware). I would imagine ktcp is almost identical, unless specialized  improvements were made to support the new network drivers.

closed 2025-03-22 by ghaerr

rafael2k, 2025-03-22:
Got it. Cool, it is nice to have more free unixes on ia16. Thanks for the fix! I'll test and report back. I'm optimistic our bobcat be (the first?) lynx port to ia16 real mode.

Mellvik, 2025-03-23:
I cannot repeat the problem at this end (provided I understand the procedure correctly: telnet from tlvc to any host, disconnect, try again). That may mean the problem is gone in my version of `ktcp` but it may also be that the debug printouts I constantly use, change the timings sufficiently to get around the problem. OTOH - I tried this on a variety of hardware and from QEMU to the Mac host.

About the `ktcp` update: It will drop right into ELKS, no problem. As will most likely the network drivers - new and updated.

ghaerr, 2025-03-23:
> I cannot repeat the problem at this end provided I understand the procedure correctly: telnet from tlvc to any host, disconnect, try again

All network daemons have to be killed first. The problem only occurs when a single network application is running, bind hangs in tcpcb_check_port. When a TCP CB struct is allocated and there are no others on the queue, the prev/next pointers were not initialized, which caused an infinite loop in tcpcb_check_port. 

> It will drop right into ELKS, no problem. As will most likely the network drivers - new and updated

Unfortunately, the network drivers won't just drop right in. The initialization "interface" between /bootopts and each of the NIC drivers was never rigidly defined, and now there's a slew of globals and other dependencies between elks/init/main.c and each of the network drivers preventing "dropping the code in" without more integration work. I have been unable to do this since I can't easily test NIC drivers, and didn't want to break what is working on ELKS without testing.

Mellvik, 2025-03-24:
> Unfortunately, the network drivers won't just drop right in. The initialization "interface" between /bootopts and each of the NIC drivers was never rigidly defined, and now there's a slew of globals [...]

My perspective on this is different. The drivers will indeed drop right in (unless there have been a `slew of` changes in ELKS on that front). Excluding that possibility, the updated ones except the ne2k driver which have had buffering added, can replace the current ones in ELKS directly. Likewise, the ee16 driver needs no changes to work in ELKS, but will obviously need the environment to be prepared since it it a new device, which is the normal for new devices added to a system. There is no slew of globals that I`m aware of, but a number of files need to be adjusted - a less than 5 minute job most likely.

That said, there is apparently little demand for these drivers in the ELKS community since no one is asking. And thus no reason to move in or update. And of course I don`t mind them coming to TLVC instead. :-)

Given the fact that the number of ethernet drivers will likely continue to grow, thinking about how to make it easier to integrate them is indeed useful. Like being able to query the system for a list of available interfaces at runtime time in order to keep `sys` and `net` out of harms way so to speak. Or even simpler,  redoing the device interface to point to the 'active' device instead of having one device per driver. And so on.

rafael2k, 2025-03-24:
> > > I cannot repeat the problem at this end provided I understand the procedure correctly: telnet from tlvc to any host, disconnect, try again
> > 
> > 
> > All network daemons have to be killed first. The problem only occurs when a single network application is running, bind hangs in tcpcb_check_port. When a TCP CB struct is allocated and there are no others on the queue, the prev/next pointers were not initialized, which caused an infinite loop in tcpcb_check_port.
> 
> Ooops; I forgot that part. I still cannot reproduce the problem on my running `ktcp` (real hardware) - and my code is like yours before the fix. So I added the fix - with a slight modification. The `tcpcbs = n; ` statement has been accidentally duplicated - while in fact it might as well be moved outside the `if`. Possibly less readable, potentially more optimal - depending on how smart the gcc optimizer is.
> 
> ```
> /* Link it to the list */
>     if (tcpcbs) {
> 	n->next = tcpcbs;
> 	tcpcbs->prev = n;
>     } else {
> 	n->next = NULL;
>     }
>     n->prev = NULL;
>     tcpcbs = n;
> ```
> 
> Thanks, [@ghaerr](https://github.com/ghaerr).

Hi all,
The code is not the same. TLVC does not have the problem because of the "memset" a little before the patch.

ghaerr, 2025-03-24:
> I still cannot reproduce the problem on my running ktcp (real hardware

> The code is not the same. TLVC does not have the problem because of the "memset" a little before the patch.

I see now. Yes, I must have removed the first memset thinking  the small number of fields were all initialized as part of adding the structure to the list. It's all fixed now on ELKS, without need for the first memset in my case.

However, I see the second memset is commented out on TLVC - not sure this is a good idea, as there are a ton of fields in the tcpcb_s struct that should be zeroed. TLVC now also has an essentially duplicated "tcpcb = n" statement at the end of each portion of the discussed "if" statement, FYI.

Mellvik, 2025-03-24:
> However, I see the second memset is commented out on TLVC - not sure this is a good idea, as there are a ton of fields in the tcpcb_s struct that should be zeroed. TLVC now also has an essentially duplicated "tcpcbs = n" statement at the end of each portion of the discussed "if" statement, FYI.

Thanks - I'll check out that on my branch. Unfortunately the code on github is very much out of sync with 'reality' now. I need to do a checkin even though it isn't finished.

Mellvik, 2025-03-25:
The `memset` is present in the version I'm running, which should explain that the symptom didn't show.
I plan to cerate a 'point release' PR of `ktcp` at its current state later today or tomorrow. That will make both discussions and diagnosing easier.

The state of the 'new' (updated) `ktcp` is that it has been running without bugs (that is, hiccups) for close to a year. Continued development is tuning, not bugfixing. More about that in the release (PR) notes.

PR #2267 (2025-03-22): [net] Fix network hang in bind() when only single TCP socket open
Fixes #2266. Looks like this was in ktcp for a very long time.