ghaerr/elks #2269: Paint in EGA Mode

Vutshi, 2025-03-26:
Hi @ghaerr,

With the recent exciting developments in ELKS, I was eager to try out the new ~~Photoshop~~ Paint program. My favorite emulator, MartyPC, doesn't yet support VGA, but Paint still runs—although it only displays the upper part of the screen:

[image]

Would it be possible to run Paint in EGA mode? Hercules graphics?

Thanks!

Vutshi, 2025-03-26:
By the way, using it on an 8088 is quite a meditative experience.

https://github.com/user-attachments/assets/c5d7ff50-b74a-433d-bf31-677ee70e4934

ghaerr, 2025-03-27:
Hello @Vutshi,

Yes, Paint could be made to run on EGA with a simple change to enable EGA 640x350 mode (instead of VGA 640x480), but the slightly larger issue is re-laying out each of the controls on the right hand control pane, using different Y offsets. I'll take a look at this, unless you'd like to: the buttons x,y values are all in elkscmd/gui/app.c in the A_InitPalette function. To set EGA mode, replace the current graphics_open with `graphics_open(VGA_640x350x16)` and that will change the BIOS mode to EGA 640x350 and you should be good to go.

I can probably make this an option from the command line to run in either VGA or EGA mode, and then all that would be left would be to re-layout the buttons on the right. 

We can't easily support hercules graphics since its mono only, no color. Doing so would involve rewriting the color palette selector on the right side, and some updated low-level graphics draw routines.

> using it on an 8088 is quite a meditative experience.

Geez, that is slow! It seems a large part of the wait is the "screen clearing" to BLACK of the drawing pane, which is already black. Let me take a look at not doing that :)

Can you actually draw anything on the screen? Press 'm' to toggle between the larger and smaller cursor, this will also help on very slow systems. I'd be interested in seeing a painting on your 8088 :)

Vutshi, 2025-03-27:
> Yes, Paint could be made to run on EGA with a simple change to enable EGA 640x350 mode (instead of VGA 640x480), but the slightly larger issue is re-laying out each of the controls on the right hand [...]

Thank you @ghaerr, it worked perfectly. Now I can draw in EGA paint!

[image]

Button positions are not optimized.
[paint_diff.txt](https://github.com/user-attachments/files/19493578/paint_diff.txt)

Vutshi, 2025-03-27:
@ghaerr what do I need to do to make it CGA compatible (320×200, 4 colors)?

ghaerr, 2025-03-28:
Thanks for the EGA diff @Vutshi - I'll apply that and I think use EGA=1 environment variable to have paint initialized EGA 640x350 mode when started. This same EGA=1 could be used for other graphics programs we create for ELKS, including nano-X.

> what do I need to do to make it CGA compatible (320×200, 4 colors)?

This will require a lot more work, as the graphics routines in graphics.c currently are written to use the 4-planar EGA/VGA method of accessing video RAM. We will need a newly-written set of routines for CGA (see elkscmd/gui/graphics.c for more details on EGA/VGA). We do have a CGA screen driver for Nano-X, which could be compared with the Nano-X EGA/VGA driver, to see what the differences look like. However, the NX driver format is pretty different from the current simpler graphics.c library.

The other problem will  be the lack of resolution for the icons on the right pane, as width 200 is too small and the icons would take too much room. Of course, many could be just removed.

While still a lot of work, running in 256-color CGA palette mode would be lots easier for the graphics.c mods (we already have that done in the 8086-toolchain/examples/graphics.c, that's not updated to elkscmd/gui/graphics.c yet); but the color selector and a few other paint internals would have to be changed. I'd be happy to help you if you want to jump into it, but probably not going to develop a CGA version myself for lack of time.

I'll try to get your EGA paint enhancement committed in a PR soon, thank you!

Vutshi, 2025-03-28:
Just for fun I asked chatGPT to draw the paint panel with certain buttons. Here is what it did on the first try:
[image]
Not bad, isn't it? :)

ghaerr, 2025-03-28:
I like the ChatGPT designed layout. It's pretty similar to what we've got but a bit more modernized, with simple graphics that could be drawn with our basic routines (except for text which isn't implemented). The color picker and current color display are drawn by code (they're not images), but the pen size and save/exit buttons are BMP files. Paint won't be able to parse and display SVG files, I wonder if ChatGPT could generate BMP images directly!?

You're welcome to try upgrading the panel, let me know if you want to jump into it! (This would need to be done for any CGA 320x200 layout also).

ghaerr, 2025-03-28:
> One can feel it on 8088.

Paint is a first-cut for ELKS and was put together quickly, since there was so much to do. All the drawing is done pixel-by-pixel, which is quite slow. This can be seen when drawing the grey control panel background. I have some ASM code which isn't included yet, which could be used to do fast horizontal or vertical line draws, which would speed that up greatly. You should comment out the easel background, it's just after the graphics_open call in app.c::TomentPainter(), that'll speed things up immensely. I'll work on speeding things up after you get your improved version running with better graphics.

The color picker squares and/or the pen sizes could be BMP files, but they'll still be about the same drawing speed. For pen sizes, 5 might be small, perhaps two rows of 4 for 8, but it's up to you. 

Yes, having coordinates readable from SVG is great, didn't think of that! Remember we don't have fonts (yet) in graphics.c so any graphics text needs to be embedded in the image.

I'm interested to know whether paint on 8088 can keep up with faster mouse movements, and what that looks like. There is some floating point used to "round out" the mouse motions, but this will be quite slow on 8088. Turn that off by setting USE_FLOATS to 0 in render.c at the top. I'm interested in knowing the difference with this off, as well as mouse movements both when drawing or just moving the cursor. This may be a limiting speed for 8088.

ghaerr, 2025-03-28:
Also a thought on the EGA vs VGA layout: since both share 640 pixels wide, we could just adopt a "shorter" right panel which is always laid out for EGA 640x350, then use the same thing on VGA 640x480. If it doesn't look good, then we can add dynamic setting of the box.y value based on EGAMODE=1 or not.

Vutshi, 2025-03-28:
@ghaerr, this is what it looks like

VGA:
[image]

EGA:
[image]

Vutshi, 2025-03-28:
It still needs some button position optimizations. I will do it tomorrow.

Best

ghaerr, 2025-03-28:
That looks pretty nice! The color picker is much improved, and nice that it runs without Y value changes on EGA.

Are you using BMP images for the brush size?

I notice that anti-aliasing was not turned on when creating the text font glyphs... can that be done?

Also what do you think of the idea of perhaps a "Mouse" button which we can wire up to toggle the mouse cursor size? That could go just next to CLS.

Vutshi, 2025-03-28:
> Are you using BMP images for the brush size?

Yes for now. But I want to just draw white boxes and then black draw circles in the centre by the actual brush such that it is done by code.

> I notice that anti-aliasing was not turned on when creating the text font glyphs... can that be done?

I switched anti-aliasing off during export from Inkscape. I think it is better to draw text by hand or use a proper pixel font, automatic conversion of a vector font to so few pixels is ugly with or without anti-aliasing.

> Also what do you think of the idea of perhaps a "Mouse" button which we can wire up to toggle the mouse cursor size? That could go just next to CLS.

Ok.

Btw, the small cursor looks like it is inverted (black arrow with white outline). Is it intended?

ghaerr, 2025-03-28:
>  But I want to just draw white boxes and then black draw circles in the centre by the actual brush such that it is done by code.

That will be a nice enhancement, and also eliminate needless BMP files.

> I think it is better to draw text by hand or use a proper pixel font, automatic conversion of a vector font to so few pixels is ugly with or without anti-aliasing.

Ok, well perhaps you can find a different program to generate the text then copy it into a final BMP. We do have two shades of gray so it can probably be made to look fairly good, all things considered.

> the small cursor looks like it is inverted (black arrow with white outline). Is it intended?

Lol, that also was a quick hack job. The cursor is drawn using a mask, whose 1 bits are those changed, and the bitmap, which are then changed to white or black (1 or 0). The cursor bits are in cursor.c, perhaps try flipping the X/\_ in cursor.c smcursorbits[] for the following:
```
/* Small 8x8 cursor for machines to slow for 16x16 cursors */
static MWIMAGEBITS smcursormask[8] = {
    MASK(X,X,X,X,X,X,_),
    MASK(X,X,X,X,X,_,_),
    MASK(X,X,X,X,_,_,_),
    MASK(X,X,X,X,X,_,_),
    MASK(X,X,X,X,X,X,_),
    MASK(X,_,_,X,X,X,X),
    MASK(_,_,_,_,X,X,X)
};
static MWIMAGEBITS smcursorbits[8] = {
    MASK(_,_,_,_,_,_,_),
    MASK(_,X,X,X,X,_,_),
    MASK(_,X,X,X,_,_,_),
    MASK(_,X,X,X,_,_,_),
    MASK(_,X,_,X,X,_,_),
    MASK(_,_,_,_,X,X,_),
    MASK(_,_,_,_,_,X,X)
};
```
I think I initially played around with it the opposite but found that the cursor didn't display well when over pure white background.

Vutshi, 2025-03-29:
New iteration. Brush buttons are generated by code
VGA:
[image]

EGA video (it is faster thanks to no clear screen at the start and the new brush buttons):
https://github.com/user-attachments/assets/5a7e7694-4907-4893-804f-3c1a5c427057

Vutshi, 2025-03-29:
The VGA version has enough free space in the panel to accommodate a 128x128 logo. How does this one look?

[image]

ghaerr, 2025-03-29:
> New iteration. Brush buttons are generated by code

Wow, this is really starting to look fantastic! And the new brush buttons are better than the original BMPs. I really like the small separations within the color picker as well as the outline in the current color box.

> The VGA version has enough free space in the panel to accommodate a 128x128 logo. How does this one look?

Looks great, very nice and fancy! Can you have your imager store it in BMP RLE4 or RLE8 compressed format? I haven't tested my BMP decompression code lately, but it should be working in drawbmp.c. This will reduce the file sizes of the .bmp files by quite a bit.

> EGA video (it is faster thanks to no clear screen at the start and the new brush buttons):

Thank you @Vutshi for working on enhancing Paint. I can guess you're having almost as much fun as I had getting it up and running. I have to say, I had always wanted to write (and understand) a paint program, and never had. After writing the first iteration of graphics.c for our C86 compiler and then @rafael2k getting involved with OpenWatcom support and some well written graphics routines, I started looking around for a simple paint program, and finally ran into TomentPaint, from which this is based. It was written by a gamer and designed well, but only using SDL and 24-bit RGB output. So taking it from there and rewriting it for ELKS has been a lot of fun. And now we're finally getting a really fancy version with good looks and a logo - what every OS needs, its own version of good Paint program!

ghaerr, 2025-03-29:
BTW, are you aware that Paint also has a color-fill capability? If you draw a closed shape and hit the 'f' key, it will fill the shape with the current color. Maybe we should have a button for that next to CLS, instead of the Mouse idea I had earlier?

Vutshi, 2025-03-29:
A slight brush improvement makes them rounder by centering the circle on the middle of a pixel to remove rough edges, requiring half-pixel precision.

Before (compare actual drawn brushes with their buttons).

[image]

After

[image]

Vutshi, 2025-03-29:
> BTW, are you aware that Paint also has a color-fill capability? If you draw a closed shape and hit the 'f' key, it will fill the shape with the current color. Maybe we should have a button for that next to CLS, instead of the Mouse idea I had earlier?

I've noticed several hidden features, like AltColor. I'll switch to icons for buttons, allowing us to fit up to 8 buttons for better control of the program.

ghaerr, 2025-03-29:
> A slight brush improvement makes them rounder by centering the circle on the middle of a pixel to remove rough edges, requiring half-pixel precision.

NIce!

> I'll switch to icons for buttons, allowing us to fit up to 8 buttons for better control of the program.

Do we really need switch to icon buttons? Keeping your existing brush drawing instead of icons reduces 8 .bmp files from /lib, which is a pretty nice savings. Also, we don't need a button for AltColor, since it uses the right mouse button. There's only three keyboard features that matter C)lear, F)ill and M)ouse size, right?

Vutshi, 2025-03-29:
> Do we really need switch to icon buttons? Keeping your existing brush drawing instead of icons reduces 8 .bmp files from /lib, which is a pretty nice savings. 

We currently have CLS, Save, and Exit. Adding C, F, and M actions requires smaller buttons—30x30 pixels, like the palette and brush buttons—achievable only with icons. These can be minimalistic 1-bit color and defined directly in code, eliminating the .bmp file.

ghaerr, 2025-03-29:
> Adding C, F, and M actions requires smaller buttons—30x30 pixels, like the palette and brush buttons—achievable only with icons. 

Oh, I see. Good idea, I thought you were talking about replacing the brush buttons with .bmp-based "icons" again. I'm not really sure about the difference between your references of icons vs buttons, but like the idea of removing .bmp files as well as having most functionality available via mouse clicks on the control pane.

> These can be minimalistic 1-bit color and defined directly in code, eliminating the .bmp file.

I had initially figured that internal C representations for icons or buttons would be large, and thus use up a lot of an application's data segment (and TomentPainter originally used bmp files). But for Paint I suppose it doesn't matter whether its data segment becomes larger, and removing the need for external .bmp files makes it much more portable!

Vutshi, 2025-03-30:
Hi @ghaerr, do you know a way to generate RLE4 BMP? Apparently gimp doesn’t know it, Python doesn’t know it, imagemagick doesn’t know it either. I’m lost :)

ghaerr, 2025-03-30:
> Do you know a way to generate RLE4 BMP? Apparently gimp doesn’t know it, Python doesn’t know it, imagemagick doesn’t know it either. 

LOL!! That must mean that only Windows did it! Don't worry about it. Perhaps I should remove the RLE4/8 code from by BMP decoder to save space, since this compression seems to be something from to far too distant past! :)

The reason we're using BMP in the first place is that it's somewhat widely known and easy to decode, after all.

Vutshi, 2025-03-30:
Well, one can indeed view rle4 bmp in safari for example. So it is widely known, it is just hard to create them. 

How do you save BMP in paint, in what kind of BMP?

Vutshi, 2025-03-30:
ChatGPT gave me some Python code to convert images into RLE4. I haven’t test it yet

ghaerr, 2025-03-30:
> How do you save BMP in paint, in what kind of BMP?

I wrote a routine drawbmp.c::save_bmp() which creates a (non RLE-encoded), palletized, 8bpp BMP. This means it carries a color palette along with the image, and the image is written out 8 bits per pixel. This routine could be enhanced to write 24-bit RGB pretty easily.

ghaerr, 2025-03-30:
> This routine could be enhanced to write 24-bit RGB pretty easily.

Now that I (re)think of it, the big problem with images on ELKS is they're huge. Even the small palletized bmp is ~238k, which won't usually fit on a distribution floppy.

Vutshi, 2025-03-30:
> N no ow that I (re)think of it, the big problem with images on ELKS is they're huge. Even the small palletized bmp is ~238k, which won't usually fit on a distribution floppy.

This is why I think keeping some compressed 4-bit image format is important (both decoding and encoding). It could be RLE4 BMP or 4-bit PNG…

ghaerr, 2025-03-30:
> If you started with RLE8 it would be lots simpler, then move to 4bpp and RLE4.

I mean that for the exporting. I'm pretty sure the RLE4 input works, since I took that code from my Microwindows repo where the code was written 30 years ago, when this stuff was in pretty wide use :)

Vutshi, 2025-03-30:
> Yes, the 4-bits will halve the size. If you started with RLE8 it would be lots simpler, then move to 4bpp and RLE4.

I mostly want to use RLE4 for paint button icons. Since paint should already be able to read it I just need to generate them using existing Python infrastructure KU’s some ChatGPT magic. 
After that I will see whether it is hard to introduce encoding to RLE4,8 to ELKS.

Vutshi, 2025-03-30:
Here's how Paint currently looks.

[image]

ghaerr, 2025-03-30:
> Here's how Paint currently looks.

Nice! You used bucket fill but there's no fill bucket icon!!

I'm a bit confused by the trash and X sign being next to each and somewhat close in meaning... I assume trash means clear and X means exit? I suppose if one wants to clear or exit, it doesn't matter if they're close in meaning.

It might be better to keep the "buttons" all black and white, which would mean using a black circle in the exit button? Or are you thinking of making them different colors

We could also have the smaller cursor by used by default if the application is compiled for EGA or running in EGA mode when EGAMODE=1 is set in the environment.

rafael2k, 2025-03-31:
> Hi [@ghaerr](https://github.com/ghaerr), do you know a way to generate RLE4 BMP? Apparently gimp doesn’t know it, Python doesn’t know it, imagemagick doesn’t know it either. I’m lost :)

I @Vutshi, I have a decoder in the elks-viewer: https://github.com/rafael2k/elks-viewer
In the scripts folder, there are examples on how to encode. ImageMagick does support RLE encoding for BMP.

rafael2k, 2025-03-31:
@Vutshi, if you need any help with the RLE4 or RLE8 code, lemme know, I can put an example in elks-viewer. RLE is the real deal, it really compresses a lot images with low entropy, like icons (rle of run-length encoding after all). RLE support in elks-viewer is first class.

Vutshi, 2025-03-31:
> ImageMagick does support RLE encoding for BMP.

This is interesting. I think I see why it didn't work for me when I tried it. MacOS doesn't display the BMP4 version for me. 

So this doesn't work in macOS:
`magick save.png -define bmp:format=bmp4 -type palette -depth 8 -compress RLE save_rle8_ref_bmp4.bmp`
But this works:
`magick save.png -define bmp:format=bmp3 -type palette -depth 8 -compress RLE save_rle8_ref.bmp`

However, space saving is not that great.

- (original) `save.png` = 3230 Bytes
- `save_rle8_ref.bmp` = 1528 Bytes
- (my RLE8 realisation) `save_rle8.bmp` = 568 Bytes

Maybe something can be stripped off the `magick` generated files...

Here are the test files [save_icon_tests.zip](https://github.com/user-attachments/files/19533472/save_icon_tests.zip)

Vutshi, 2025-03-31:
Oh, there are a lot of `0` in the header of `magick` files

Vutshi, 2025-03-31:
>  if you need any help with the RLE4 or RLE8 code, lemme know, I can put an example in elks-viewer. 

@rafael2k how do you actually generate RLE4 with `magick`?

`-depth 4` doesn't help me, the bmp data still looks the same as RLE8. No nibbles

Vutshi, 2025-03-31:
RLE4 is indeed tricky. Simple implementation is larger than a simple RLE8 (without absolute mode).

This is current `paint` with RLE8 icons

[image]

ghaerr, 2025-03-31:
> However, space saving is not that great.
save_rle8_ref.bmp = 1528 Bytes
(my RLE8 realisation) save_rle8.bmp = 568 Bytes

1528 -> 568 bytes? I'd say that's great, 2/3 smaller!

> icon sizes:

I am a little surprised to see that all the bs?.bmp files are the same size - 1954. Shouldn't they be different when encoded RLE8?

Also, I thought you moved to having Paint draw those directly, instead of using .bmps for them? That saves quite a bit of disk space.

> How exactly should I realise this EGAMODE? Do we choose it during compilation or dynamically during program execution? Does ELKS know that it runs in EGA?

No, ELKS isn't (currently) providing any information about EGA vs VGA, but could in the future. I was thinking it or the user would indicate that system-wide with a preset EGAMODE=1 environment variable, set either automatically at boot time in the future or now perhaps in /etc/profile by the user.

Paint would then check for the EGAMODE environment variable dynamically perhaps as follows:
```
    int EGA = (int)getenv("EGAMODE");
    graphics_open(EGA? VGA_640x350x16: VGA_640x480x16);
```
Although you've now got Paint designed so that the box.y values don't have to change (much), the global int EGA variable could be used when required to dynamically operate differently between the larger VGA and smaller EGA layouts, where needed.

Later on, I was thinking of moving the initialization under the hood in graphics.c, so that it could perhaps check the command line and/or environment variables to set not only EGA or VGA, but also a variety of 8bpp or CGA modes and determine the correct argument to graphics_open, etc. It would then set some global variables, and int EGA would move into graphics.c. Graphics.c already has an int VGA global but its used for slightly different purposes, so I thought it easiest for you to keep things mostly in app.c for now.

ghaerr, 2025-03-31:
I'm wondering if all the zeros in the header are the 256-16 = 240 unused palette entries (=4*240 = 960 bytes? I bet so! It's not shrinking the palette size down to the actually used entries!

ghaerr, 2025-03-31:
> The actual RLE8 encoded data looks the same

Sounds like that means it's not encoding RLE then.

ghaerr, 2025-03-31:
I have a [BMP converter](https://github.com/ghaerr/microwindows/blob/master/src/images/tools/convbmp.c) in Microwindows. It isn't what we need right now, since this version converts .bmp files to .c files, to be linked into applications. (I don't think that's necessarily the best general answer since we could quickly exceed the 64k data segment limit, and wanted to support loadable icons).

Nonetheless, that program could be enhanced to read in any .bmp, and output an RLE4 compressed image with a small palette. 

If you can get Paint working the way you like with the .bmp icons you want, we can always add that capability later, to decrease the size of the .bmp files (or link them in, if you think that might be a better option).

Vutshi, 2025-03-31:
> > The actual RLE8 encoded data looks the same
> 
> Sounds like that means it's not encoding RLE then.

No, no. The reference file `save_rle8_ref.bmp = 1528 Bytes` is already RLE compressed by `magick`. The RLE data of this file matches my Python implementation and it is smaller than the raw icon bitmap `32x32x1Byte=1024B`

Vutshi, 2025-03-31:
> I'm wondering if all the zeros in the header are the 256-16 = 240 unused palette entries (=4*240 = 960 bytes? I bet so! It's not shrinking the palette size down to the actually used entries!

That is it!

Vutshi, 2025-03-31:
I haven’t found a way to tell image magick to use 16 color palette as well as RLE4

Vutshi, 2025-03-31:
> Paint would then check for the EGAMODE environment variable dynamically perhaps as follows:
> 
> ```
>     int EGA = (int)getenv("EGAMODE");
>     graphics_open(EGA? VGA_640x350x16: VGA_640x480x16);
> ```

This works! 
When I set `export EGAMODE=1` before running `paint`, the 640x350 mode activates, and the mouse cursor automatically switches to the smaller size.

I guess I can try to make PR now. I wonder how does it work when using `git`.

ghaerr, 2025-03-31:
> I guess I can try to make PR now. I wonder how does it work when using git.

Do you have a cloned version already, that will be easier, as you can use `git diff` and it'll show you the diff.

0. Back up your changes or copy all of elkscmd/gui/* somewhere just in case!
1. Fork the ELKS repo on your own Github account
2. Clone your repo to a new elks master somewhere. Be careful - this could have the same name as the master ELKS repo elks/ and overwrite your change!. You can rename your working copy to elks-org/ beforehand to be safe.
3. cd elks-new/
4. git pull
5. copy all your changes into elkscmd/gui
6. git diff  (to see your changes)
7. git checkout -b paintchanges (adds all changes to branch paintchanges)
8. git commit -a (saves all changes locally)
9. git push (this will give an error, copy and paste git push command below)
10. git push --set-upstream origin paintchanges (this should be the error text above, or something like it)
11. Go go your own ELKS repo on Github and it should automatically open a PR for you on ELKS master
12. Submit it and I should see it

Vutshi, 2025-03-31:
Thank you @ghaerr for the detailed instruction. PR has kind of work, although I managed to do two commits somehow (I tried using `gh` tool).

Vutshi, 2025-03-31:
> I'm interested to know whether paint on 8088 can keep up with faster mouse movements, and what that looks like. There is some floating point used to "round out" the mouse motions, but this will be [...]

Going back to mouse on 8088. It feels very slow. The video I showed above doesn’t convey it. There is now an option in MartyPC to speed up CPU to double its frequency. It makes a huge difference for some reason. It feels as if 8088 is drowning in the stream of mouse events and when I double its speed it suddenly can swim in the stream. I guess rather than explaining it is easier to try for yourself )

ghaerr, 2025-03-31:
> I guess rather than explaining it is easier to try for yourself )

I don't have MartyPC on my system, and long story short don't have a Rust development system on my mac until I upgrade it.

>  It feels as if 8088 is drowning in the stream of mouse events and when I double its speed it suddenly can swim in the stream

I think I know what you mean, however. I ran into this problem with @toncho11 when testing Nano-X on his very old Amstrad, and it was found that the size of the cursor made a huge difference - literally drawing the cursor, which is more work than one would think (see cursor.c) - caused things to slow down considerably.

To test this theory, try first switching between the large and small cursor and noting how much different it makes, versus not.

Then, try creating a cursor mask/bits which is quite a bit smaller - perhaps just a thin X. I did this for Nano-X nxclock when I was getting and it was definitely lighter. Here's some code for that you might try, or even making the X smaller like x or a single point. Using a single point will give you the best possible speed to test whether this problem can be fixed with a smaller cursor size:
```
    bitmap1fg[0] = MASK(_,X,X,X,X,X,_);
    bitmap1fg[1] = MASK(X,_,X,X,X,_,X);
    bitmap1fg[2] = MASK(X,X,_,X,_,X,X);
    bitmap1fg[3] = MASK(X,X,X,_,X,X,X);
    bitmap1fg[4] = MASK(X,X,_,X,_,X,X);
    bitmap1fg[5] = MASK(X,_,X,X,X,_,X);
    bitmap1fg[6] = MASK(_,X,X,X,X,X,_);

    bitmap1bg[0] = MASK(X,_,_,_,_,_,X);
    bitmap1bg[1] = MASK(_,X,_,_,_,X,_);
    bitmap1bg[2] = MASK(_,_,X,_,X,_,_);
    bitmap1bg[3] = MASK(_,_,_,X,_,_,_);
    bitmap1bg[4] = MASK(_,_,X,_,X,_,_);
    bitmap1bg[5] = MASK(_,X,_,_,_,X,_);
    bitmap1bg[6] = MASK(X,_,_,_,_,_,X);
```

Depending on how speeds things up (or not), we may have to look into other ideas, like profiling the event subsystem (I already did that, but could be more to do), or perhaps trying some fanciness with aggregating the mouse input - adding together subsequent mouse move X,Y values into a single event using a timestamp - but this is quite tricky, and may end up adding less resolution to the mouse trail.

Finally, we are not using "mouse acceleration", which is also a bit complicated, but that involves moving the mouse location faster based on how long its been moving in a certain direction. All OSes use it, and so does Nano-X, but I didn't add that code into cursor.c. If you'd like to take a look at that code from Nano-X, [it is here](https://github.com/ghaerr/microwindows/blob/master/src/engine/devmouse.c) (see filter_relative - it takes a scale and threshold and transforms the relative mouse coordinates).

Vutshi, 2025-04-01:
@ghaerr I suspect that the floating point division for mouse is not necessary.
I did the following change to the integer code:
```
#if USE_FLOATS
                float slope = fabs(((float)(y2-y1))/((float)(x2-x1)));
                if(slope > 128.f)      slope = 128.f;
                if(slope < 0.0078125f) slope = 0.0078125f;
                if(slope >= 1)         repy=round(slope);
                else                   repx=round(1.f/slope);
#else
                // repx = 1;
                // repy = 1;
                int deltaY = abs(y2-y1);
                int deltaX = abs(x2-x1);
                if(deltaY>=deltaX)  repy = deltaY/deltaX;
                else                repx = deltaX/deltaY;
#endif
```
and it works quite well, I don't see any difference in line smoothness in comparison with floats.

Paint float division:
[image]
Paint new integer division:
[image]

The main difference is the binary size:
-` float paint.exe = 13163 B`
-` integer paint.exe = 9492 B`

However, I don't really understand what the code is doing or why only one of `repy` or `repx` is updated at a time, rather than both.

Vutshi, 2025-04-01:
Well, well, well I can feel the Terminator judging day is approaching fast.
I copy-pasted the whole `R_Paint` function (integer version) into `chatGPT` and asked to optimize it. This is what I got:
```
void R_Paint(int x1, int y1, int x2, int y2) {
    // Draw initial point
    if (bushSize <= 1) {
        if (x1 >= 0 && x1 < SCREEN_WIDTH && y1 >= 0 && y1 < SCREEN_HEIGHT) {
            drawpixel(x1, y1, drawing ? currentMainColor : currentAltColor);
        }
    } else {
        R_DrawCircle(x1, y1, bushSize);
    }

    // Bresenham's line algorithm for efficient line drawing
    int dx = abs(x2 - x1);
    int dy = abs(y2 - y1);
    int sx = (x1 < x2) ? 1 : -1;
    int sy = (y1 < y2) ? 1 : -1;
    int err = dx - dy;
    
    while (x1 != x2 || y1 != y2) {
        if (bushSize <= 1) {
            if (x1 >= 0 && x1 < SCREEN_WIDTH && y1 >= 0 && y1 < SCREEN_HEIGHT) {
                drawpixel(x1, y1, drawing ? currentMainColor : currentAltColor);
            }
        } else {
            R_DrawCircle(x1, y1, bushSize);
        }

        int e2 = 2 * err;
        if (e2 > -dy) {
            err -= dy;
            x1 += sx;
        }
        if (e2 < dx) {
            err += dx;
            y1 += sy;
        }
    }
}
```
I copy-pasted this AI code into our render.c and it just works. Actually, it seems to work better than before:
[image]

ghaerr, 2025-04-01:
Hello @Vutshi,

Thank you for your analysis and proposed rewrite of the R_Paint algorithm. There is a bit of a history behind this and perhaps more discussion, before we commit #2276 as-is. 

First, the big difference in executable size is the result of the software floating point litbrary being included when any floating point operation is performed - for now, on ELKS, we don't assume an 8087+ FPU is present and thus have the ia16-elf-gcc compiler setup to emulate floating point. This takes ~3.5k of code, all linked in, and the software floating point is slow, probably very slow, on 8088's, and definitely slower than in hardware. All x86 CPUs after the 80486DX have floating point built in, but we can't assume our users have that.

> I did the following change to the integer code:
> and it works quite well, I don't see any difference in line smoothness in comparison with floats.

Thank you for the screenshots and first (human) rewrite. I can't seem to tell a difference. The original TomentPaint also only used integer operations for drawing. However, a [PR to improve interpolation at high speeds](https://github.com/silvematt/TomentPainter/pull/1) was submitted, but not committed. Note this PR states the improvement is only seen at high mouse speeds, and requires floating point. In my first version of Paint, I tool this PR and added the floating point PR code, as I wanted to see the difference.

Take a look at the original version, let me know what you think. It doesn't look like it has Bresenham error correction though.

>  I don't really understand what the code is doing or why only one of repy or repx is updated at a time, rather than both.

I must admit that I don't actually understand the high-speed mouse interpolation code. I hastily added my own ifdefs to create a version of R_Paint that could quickly be compiled with or without FP. Importantly, on QEMU (running at high speed on my mac Laptop) I can definitely see a difference between the FP and integer versions. Also, I never ran the original Toment integer code, just placed some quick ifdefs when integrating the uncommitted PR.

I don't know if there's any displayed difference between the Bresenham and previous integer version(s) of R_Paint, it could be hard to tell without the mouse moving rapidly. I remain concerned that your system may be too slow to see a difference when the floating point interpolation may also provide a benefit.

Can you slightly reformulate your PR so that perhaps two versions of R_Paint remain in the source, with your new integer version being active, but the older floating point version still available via if USE_FLOATS? Since we haven't gotten to the bottom of the higher speed mouse issue(s) yet, I think it would be nice to be able to keep the FP algorithm for future comparison.

What do you think?

Vutshi, 2025-04-01:
After ChatGPT revealed what is going on in the code I googled Bresenham and found this very nice [explanation](http://members.chello.at/easyfilter/bresenham.html). 
The “new” AI code is one-to-one Bresenham algorithm for line drawing from this link. They have other interesting codes for circles, ellipses, etc. 

> I don't know if there's any displayed difference between the Bresenham and previous integer version(s) of R_Paint, it could be hard to tell without the mouse moving rapidly. I remain concerned that [...]

If one zooms into the above screen shots, one can see that AI Bresenham consists of perfect straight line segments. In contrast the float and my integer versions have slightly zigzagging “straight” lines. I would say, after seeing the perfect Bresenham lines I cannot unsee the zigzagging of the float algorithm. It is clearly broken. 

These plots were obtained in Qemu they are unaffected by slow mouse issues. 

> Can you slightly reformulate your PR so that perhaps two versions of R_Paint remain in the source, with your new integer version being active, but the older floating point version still available [...]

I would say the Bresenham algorithm is what is supposed to be in the code from the very beginning. Float version is not well understood most likely because it is broken.

This new Bresenham didn’t solve the problem of slow mouse on 8088. Bottleneck is somewhere else. I noticed that the mouse cursor takes its time to even start moving. So the bottleneck is probably long before any line drawing happens.

ghaerr, 2025-04-01:
> If one zooms into the above screen shots, one can see that AI Bresenham consists of perfect straight line segments. 

> In contrast the float and my integer versions have slightly zigzagging “straight” lines. I

I must say, what you are saying does indeed seem to be the case! 

Thank you for the Bresenham link, I was unaware the algorithm could be simply used for drawing Bezier curves and antialiased lines - I may have to use that in some replacement routines I've been considering for Microwindows and graphics.c.

> These plots were obtained in Qemu they are unaffected by slow mouse issues.

Ok. Well, you (or maybe ChatGPT) have convinced me the new code is superior. Frankly, I'm happy to get rid of FP as well as code we don't understand. From your screenshots it appears neither previous algorithm got it right. I'll commit the PR.

> I can feel the Terminator judging day is approaching fast.

It appears so. It's nice ChatGPT happened to pull from the fantastic link you found as well - those are a nice set of Bresenham algorithms for a number of common requirements. The Nano-X circle/ellipse drawing routine could probably use an update from it.

> I noticed that the mouse cursor takes its time to even start moving. So the bottleneck is probably long before any line drawing happens.

We might want to separate the problem of startup time versus continuous mouse movement. I still suggest trying a small x cursor to see how that helps.

ghaerr, 2025-04-01:
>  It is always a couple of seconds behind my next motion.

A couple seconds? That's huge! There is no mouse buffer, instead the mouse bytes are stored by the kernel serial driver, and then read using `read` in mouse.c::read_mouse() on line 241. Long story short, reading as many bytes as necessary caused a problem where the mouse "button up" was not acted on until the next mouse movement. I "fixed" this in a previous commit. This could be where the lag is.

To play with this, change the MAX_BYTES 3 on line 79 to a larger multiple of 3, like 66. 

If that allows the mouse to fly much quicker, you'll see that on occasion a button up will be missed until the next mouse movement. To fix this, we may need to implement mouse event queues. Try it and we'll go from there.

Vutshi, 2025-04-02:
> A couple seconds? That's huge!

I played with the mouse a bit more and it is not just a couple of seconds. When I just attach mouse to the emulator, the mouse works okayish for a few seconds but the delay grows! It goes up to several seconds then I detach mouse (like Ctrl+alt+G in Qemu) and observe how mouse cursor in paint lives its own life, performing actions and motions I did in the previous 5-10 seconds (I wonder where does it store all this information). After paint runs out of steam and settles into a waiting loop I can attach mouse again and have some relatively responsive cursor for a few seconds and then the cycle repeats. 

If I do the same thing with double speed emulation then the huge queue of events never pile up. It is slow but consistent and doesn’t degrade with time. 

> To play with this, change the MAX_BYTES 3 on line 79 to a larger multiple of 3, like 66.

This change seems to not affect the problem.

ghaerr, 2025-04-02:
>  When I just attach mouse to the emulator, 

Which emulator are you using MartyPC? 

>  I detach mouse (like Ctrl+alt+G in Qemu)

Can you make this happen in QEMU, or is this only happening in MartyPC?

What about real hardware, have you ever seen this growing delay?

> If I do the same thing with double speed emulation then the huge queue of events never pile up.

I'm not aware that QEMU has double speed. Perhaps this is something to do with the way MartyPC is sampling the mouse internally?

> This change seems to not affect the problem.

I used to see an issue like this in QEMU, but can't remember exactly when/how it happened. Let me see if I can try to reproduce that on QEMU to give us more ideas.

[66 more replies]

closed 2025-04-21 by Vutshi