ghaerr/elks #2293: PC-98 image does not boot on Neko Project 21/W emulator

tyama501, 2025-04-11:
Hello @ghaerr ,

Nice to see great improvements recently :) Than you.
I still don't have my build environment in the new house, (mainly because of my PC for the build is still in the box) 
but I downloaded fd1232.img and tried on the Neko Project 21/w emulator.
Unfortunately, it seems that it won't boot on NP21/w although it boots on DOSBox-X.

If the hard disk is not attached,
it stops here.

[image]

If the IDE hard disk is attached,
it stops here.

[image]

Current memory setting is 640KB + 3MB.

Do you have any idea how should we debug this?
It might related to the memory issue,
but I wonder the hard disk issue is coming from the same issue or not.

Thank you.

ghaerr, 2025-04-12:
Hello @tyama501,

> it seems that it won't boot on NP21/w although it boots on DOSBox-X.

That's good news it boots on DosBox-X, this could mean that either XMS or HMA isn't well supported on Neko, or it might be something to do with the hard drive. When a hard disk is seen, early on the kernel reads the MBR to get the partition table, and that means the XMS buffer system is used, so XMS is still possibly related. 

> Do you have any idea how should we debug this?

Can you edit /bootopts on the boot disk?

I would suggest taking two steps: commenting out xms=on in /bootopts will turn off XMS. If that boots, then it would seem that XMS memory, or at least INT 1F BIOS block move isn't supported in Neko. However, HMA also uses memory above 1M, so you may want to comment out hma=kernel as well. If commenting that out, then perhaps the A20 line isn't supported. If this is so, then we'll need you to write some ASM in setup.S to setup A20 for HMA booting.

With both lines commented out, the system should hopefully boot, I don't think we've made any other changes that might affect this.

Thank you!

tyama501, 2025-04-12:
Hello @ghaerr 

Thank you,
I will try that later.
Neko is known as xms well supported and I have implemented the xms buffer to elks with the int1F in it before.

Actually DosBox-X int1F has been fixed referred to NP2 recently.
https://github.com/joncampbell123/dosbox-x/issues/5463

I haven't used that DosBox-X yet, so I may try that as well.

ghaerr, 2025-04-12:
> I haven't used that DosBox-X yet, so I may try that as well.

Yes, that would be good. I tested XMS and HMA on DosBox-X and both worked, so I'm not sure what the issue is yet.

tyama501, 2025-04-12:
Hello @ghaerr ,

It seems that just commenting out xms=on works.

[image]

The log says xms: 3072K, ssd; no XMS
Kernel text ffff
Does this mean Kernel is in HMA?
What does the 541+7+0K free mean?

Thank you.

tyama501, 2025-04-12:
I have confirmed free has been increased about 40K. That's great.

tyama501, 2025-04-18:
Hello @ghaerr ,

I could pull the latest codes and build image.
I have a question about xms=int15.

If I comment out xms=on and enable xms=int15,
it can boot but the log says no xms and
the kernel also stays in the conventional memory even hma=kernel.

What does xms=int15 do?

[image]

ghaerr, 2025-04-18:
> What does xms=int15 do?

That option forces the use of XMS using the INT 15 (for IBM PC) or INT 1F (for PC-98).

The older XMS code always forced the use of INT 15/1F for PC-98. I wanted to test with your PC-98 works with some 386 processor's "unreal mode". That mode is set using xms=on. If that works, then your PC-98 would be using the faster unreal mode XMS access rather than using INT 1F. 

> Oh sorry it says bootopts not found.

Check that the file is not over 1023 bytes long, or possibly over 512 bytes long if we have a bug in the setup.S PC-98 routine that reads /bootopts. The PC-98 1232k sectors are 1K so this should be working automatically.

Once the system does not display the bootopts not found or bad format, you can test xms=int15. If /bootopts is not found, then XMS will default to OFF, so that ELKS can boot on as many systems as possible.

ghaerr, 2025-04-18:
> My 486 machine is broken and I only have the 286,

The 286 CPU can't support unreal mode, and ELKS forces 286 systems to use INT 15/1F even if xms=on is in /bootopts.

> but the Neko supports 486 and the unreal mode.

That means your 286 real hardware should show xms: INT 15/1F, while Neko shows xms: unreal mode.

Is Neko now working with unreal mode and ELKS, different than when you could not get it to boot for this issue?

> ssd: 2508K avail
I assume it is ramdrive but I have never tested the ramdrive on PC-98 before.
Is there way to disable this to distinguish problems?

Yes. Right now, the only way to disable the XMS ramdisk is to reset CONFIG_FS_XMS_RAMDISK in .config and rebuilding. The XMS ramdisk is automatically disabled when xms=off in /bootopts or when there is no /bootopts though. 

When CONFIG_FS_XMS_BUFFERS=y and xms=on or xms=int15, XMS will attempt to initialize very early in boot process. Otherwise, if CONFIG_FS_XMS_RAMDISK=y and xms=on or xms=int15, XMS will attempt to initialize later, during the ssd: displayed portion. But the ramdisk itself isn't actually used until `ramdisk /dev/ssd ...` is executed.

tyama501, 2025-04-18:
No, Neko

tyama501, 2025-04-18:
>

ghaerr, 2025-04-18:
>  I' m not sure why it does not go to the unreal mode.

Over at https://github.com/Mellvik/TLVC/pull/161 is a very long discussion of why this might be happening. We have concluded that some 386 real CPUs as well as emulators don't fully handle "unreal" mode properly; it is an undocumented CPU feature, after all. I'm currently working on porting over @Mellvik's full protected mode switch solution to the problem. You can then test that with Neko once committed.

> Neko is not working. The above result is all about Neko.

Did you say this version of Neko once did work with unreal mode? I forget.

tyama501, 2025-04-18:
@drachen6jp application has used the unreal mode before so I think the Neko worked with it.
https://github.com/ghaerr/elks/discussions/2183#discussioncomment-11905033

However,
I don't think I have used on the xms buffer before.

ghaerr, 2025-04-18:
> I don't think I have used on the xms buffer before.

That's probably good, as out means we haven't probably broken anything, we just need to get it working on Neko.

> @drachen6jp application has used the unreal mode before so I think the Neko worked with it.

If you can locate the function that is used to enter unreal mode in @drachen6jp's application, I can take a look at it and compare that to what we've learned in https://github.com/Mellvik/TLVC/pull/161. My improved version using @Mellvik's finding that full protected mode entry is required on some CPUs is almost done - I will push that shortly and you can test to see if that makes a difference.

ghaerr, 2025-04-18:
@tyama501, I have submitted PR #2308 for you to test whether this changes operation within Neko.

tyama501, 2025-04-19:
Hello @ghaerr 

Thank you.

I have downloaded the image from here and tried.
https://github.com/ghaerr/elks/actions/runs/14544056059

It seems that it still says int15/1F and does not work.
So, the problem is either or both
- It cannot enter to unreal mode
- It does not work with int1F xms, which has been working before (long long time ago...

[image]

I will look it again later.

drachen6jp, 2025-04-19:
I checked My Machines with this fd1232.img
https://github.com/ghaerr/elks/actions/runs/14544056059

crytical?Memory size ?
under 64Mbyte: do not run elks. stop in initialize
[image]

over 64Mbyte: do run elks 
[image]

neko project also can run with over 64Mbyte Memory.
[image]

ghaerr, 2025-04-20:
Hello @drachen6jp,

Thank you very much for testing ELKS! I am very surprised at your finding that ELKS w/XMS does not boot on your system with ~30Mb XMS memory but does boot with 64Mb!! I can't think offhand why this might be, unless the XMS memory does not start at 1MB address. Right now, ELKS always assumes XMS starting just past the lower 1Mb memory, and is not (yet) adjustable. If that memory is not present, crash at boot could happen.

Is your machine a 286 or 386? Is Neko emulating 286 or 386?

If the CPU is 286, then INT 1F will always be used, never unreal mode. This may be why both your machine and Neko are showing INT 1F instead of unreal mode. This assumes /bootopts has xms=on in it. If xms=int15, then INT 1F will always be used.

Is there a PC-98 BIOS location that shows the start address of XMS memory? 

We are currently using the following code to get the PC-98 memory present in lower memory, returned as Kbytes in AX:
```
        xor %ax,%ax
        mov %ax,%es
        mov $0x501,%bx
        mov %es:(%bx),%cl
        and $7,%cl
        inc %cl
        mov $128,%al
        mul %cl
```
The following code is used to get the total (including XMS) memory present, returned in Kbytes in CX (from setup.S):
```
   xor %ax,%ax                  // get xms size in CX
   mov %ax,%ds
   mov $0x401,%bx               // xms_kbytes = *(byte 0x401) * 128
   mov (%bx),%al
   mov $128,%cl
   mul %cl
   mov %ax,%cx
   mov $0x594,%bx               // xms_kbytes += *(word 0x594) * 1024
   mov (%bx),%ax
   cmp $49,%ax                  // don't allow overflow when > 64MB
   jl 1f
   mov $0xffff,%cx              // 64MB - 1024
   jmp 2f
1: mov $1024,%bx
   mul %bx
   add %ax,%cx
```
This code will return 64M - 1K = 65535 as maximum XMS memory even on systems with more memory, since result has to fit in 16-bits.

Thank you!

Hello @tyama501,

> Does the elks support the memory over 64MB?

No, the above code gets BIOS reported memory at 0x401 and 0x594 for total memory, and will return 65535 when larger memory is reported. 64M total  corresponds to 1M lower memory + 63M XMS, and also assuming the extended memory starts at 1M.

> I doubt it is supported since the boot messages show kernel is not in ffff hma.

If /bootopts contains hma=kernel and A20 gate works and XMS memory available at 1M, then kernel will load in HMA and show:
Kernel text ffff ... in boot screen.

I remember you showed that it is possible to boot kernel in HMA, so it seems that XMS memory is available above 1M. Thus I am not sure why system boots w/64M but not with 30M ram.

I will try to think other reasons why this might be.

Thank you!

drachen6jp, 2025-04-20:
Hello @ghaerr. thanks for your reply.
My machine which has dirty LCD is Pentium 133Mhz.and which has 64MB is MMX-Pentium 166.
Neko is emulating 386 machine.

0000:0401h
this memory points memory size under 100000h. (very rare machine (PC-H98 NESA) can use over 100000h ).
0000:0594h
this memory points memory size which starts at 100000h. 
so 0000:0401h was set 70h 0f0000h(15MiB) to 100000h(16MiB) is not RAM.

High memory is divided at this address.

thanks.

ghaerr, 2025-04-20:
> My machine which has dirty LCD is Pentium 133Mhz.and which has 64MB is MMX-Pentium 166.
Neko is emulating 386 machine.

I see. So the system should be trying to use "unreal mode" XMS instead of INT 1F. Can you verify what xms= and hma= statements are in your /bootopts?

> 0000:0401h
this memory points memory size under 100000h. (very rare machine (PC-H98 NESA) can use over 100000h ).

100000h is 1MiB, right? You are saying only rare machines have memory > 1MiB?

> 0000:0594h
this memory points memory size which starts at 100000h.
so 0000:0401h was set 70h, from 0f00000h(15MiB) to 1000000h(16MiB) is not RAM
High memory is divided at this address.

So you are saying that on PC-98 with 0000:0401h set to 70h, low memory goes from 0-15MiB, then has 1MiB of no RAM,
then 0000:0594 will specify how much ram is above 16MiB? There is always a hole in memory between 15 and 16MiB?

What I do not yet understand is that ELKS XMS starts using memory at 1M, and 500 XMS buffers is only 500k. So all XMS memory is only 1M through 1.5M. I don't see why system would be crashing with 30MiB of memory, but working with 64MiB, since ELKS will not be using any memory > 1.5M until ramdisk is used, and it is not being used.

Am I understanding PC-98 correctly? it sounds like I cannot add the two 0401 and 0594 memory sizes together, since there might be hole in between.

drachen6jp, 2025-04-20:
oops.I'm sorry.
0:401 means
under 1000000h(16MiB)　from100000h(1MiB)

>0000:0594 will specify how much ram is above 16MiB? There is always a hole in memory between 15 and 16MiB unless 0401h = 80h?
Yes. but when 0401h=78h,memory hole is not exist.
under100000h(1MiB) memory size is set to 0:501h bit0 1 2.

tyama501, 2025-04-20:
Hello @drachen6jp 

Thank you for testing.

I think
xms=on is unreal/int1f auto mode and
xms=int15 is force int 1f mode (so, there is no xms=unreal.

If xms= is commented out and
just setting hma=kernel,
it will boot and the kernel goes to ffff hma successfully on Neko. (It would be nice if you can try it on the real PC)
(So, I think xms= is something wrong for PC-98)

Hello @ghaerr ,

How does the elks detect 286 and bellow or 386 and up?
IIRC, the cpu detection in the boot code (or setup) has been eliminated because of code space.

Thank you!

drachen6jp, 2025-04-20:
config.h Line58
SETUP_CPU_TYPE 1 //ifdef PC98
system.c Line73
arch_cpu = SETUP_CPU_TYPE
xms.c Line 71
arch_cpu <= 6

so PC-98 goto INT15 routine on XMS.C

thanks.

ghaerr, 2025-04-20:
Hello @drachen6jp,

> memory divide 3 positions.
A. 0-1MB 0:501h, 3bit 128kB size , ordinal pc-98 has 640kB
B.1MB-16MB 0:401h, byte 128kBsize
C.16MiB-4GiB 0:594h, word 1MBsize

Thank you. I now understand how hardware memory is laid out on PC-98. For now, it looks that we will have to limit the size of usable XMS memory for ELKS to the contiguous XMS only; that is, if there is a hole at 15MB with lots of memory above, then ELKS will report 14MB usable XMS and not use the upper XMS memory. I will produce a PR for that, and also check whether that affects our present booting problem.

> xms=unreal

There is no xms=unreal (force unreal mode) option. So this will be acting like no xms= in /bootopts (correctly, in this case).

> config.h Line58
SETUP_CPU_TYPE 1 //ifdef PC98
system.c Line73
arch_cpu = SETUP_CPU_TYPE
xms.c Line 71
arch_cpu <= 6
so PC-98 goto INT15 routine on XMS.C

Wow! You are very good at finding answers! Yes, this is the problem, the CPU type is being forced to non-386 by config.h for PC-98, that is the reason the unreal mode code is never executed. If SET_CPU_TYPE were set to 7, unreal mode could be tested on 386 systems.

Hello @tyama501,

> How does the elks detect 286 and below or 386 and up?
IIRC, the cpu detection in the boot code (or setup) of PC-98 has been eliminated because of code space.

I see that we are still using the getcpu routine in cputype.S, but only for IBM PC. This should probably be added back for PC-98. I will take a look at this and submit a PR for this as well as setting SETUP_CPU_TYPE above from it. This should probably wait until we try xms=on using SETUP_CPU_TYPE 7 in config.h to see whether it works. Otherwise there is no point in having dynamic CPU detection.

Thank you!

drachen6jp, 2025-04-21:
I changed config.h.
CPU_TYPE to 7
and
XMS_START_ADDR to 0x110000 for HMA kernel.
and
delete 0x594 section on Setup.S .
[image]
good.

ghaerr, 2025-04-21:
This is great, thank you!

Do you have any ideas as to why the system did not boot before? 

I will plan a PR with your changes, but probably include a routine to determine CPU type automatically.

drachen6jp, 2025-04-21:
Understood.
If HMA is used, the block_move function is already present in the FFFF segment.

If you call Int1F at that location, internally A20 will be disabled again, and it will not come back from the call destination. It is impossible to re-enable A20 because it will not come back.
So the solution is to leave the block_move function in the low area and modify it to use HMA, or leave HMA out when using XMS with Int1F, or use unreal mode.

Also, XMS_START_ADDR could work with 100000L. Sorry for the extra information.

I translated it into English at DeepL. My English ability is too poor so I rely on tools.

Translated with DeepL.com (free version)

ghaerr, 2025-04-21:
Hello @drachen6jp, thank you for your comments. 

> If you call Int1F at that location, internally A20 will be disabled again, and it will not come back from the call destination.

Yes, agreed. That is the reason why PC-98 was not booting, as CONFIG_CPU_TYPE was set to 1, which then forced the use of INT 15/1F block move, and since the kernel was in HMA, the system crashed after the first disk access.

The unanswered question for me was: why was the system booting at 64M but not booting at 30M? I have finally figured the reason out; the reason has to do with the check for whether the kernel can load into HMA or not, in setup.S. In this code, the hmacheck routine compares the XMS memory available with 64, to see if there is at least 64k of memory available:
```
// check whether HMA can be enabled, return NC if so with A20 enabled
checkhma:
        push    %ds
        mov     $INITSEG,%ax
        mov     %ax,%ds
        mov     xms_kbytes,%ax          //. must have xms size >= 64k
        cmp     $64,%ax
        jl      1f    // <--- BUG IS HERE
        mov     hma_kernel,%al          // and hma=kernel in /bootopts
        test    %al,%al
        jz      1f
        call    enable_a20_gate         // and A20 enabled
        test    %ax,%ax
        jz      1f
        clc                             // success
        jmp     2f
1:      stc                             // fail
2:      pop     %ds
        ret
```
What was happening is that the comparison with 64 was done using the "JL" (jump lower) conditional jump, which is a *signed* comparison. When memory was 32M to 64M, the value xms_size is 32768 (0x8000 = -32768 signed) to 65535 (0xFFFF = -1 signed) and the comparison to 64 being lower would fail, the kernel would not be loaded into HMA, and then the INT 15/1F would work.

The fix is to use an unsigned compare, "JB" (jump below unsigned). I will also fix that.

@Mellvik, I thought you might be interested in this fix.

> delete 0x594 section on Setup.S .(memory hole problem)

Yes, thank you. For the time being, we will limit the usable memory for PC-98 to 16M by not using 0x594 memory size, due to the possibility of there being a hole in between the lower and upper XMS hardware memory.

I will prepare a PR that should address all these issues.

Thank you!

Mellvik, 2025-04-22:
> @Mellvik, I thought you might be interested in this fix.

Thank you @ghaerr. Yes, I'm definitely interested in this fix. As mentioned in a different thread there is an issue here when newer systems with (relatively) huge memories run TLVC or ELKS: The XMS-size BIOS call returns a 16 bit unsigned and maxes out at 64k (indicating 64M). Which most likely means that larger memory still gets reported as 64M. It did admittedly not occur to me that the `jl` instruction was signed, i.e. it means 'jump if less', not 'jump if lower'. The devil is in the details.

In TLVC the XMS size is now being capped at 32M for practical reasons and it occurs to me that that capping could be much more efficiently done in the setup code (by means of a mask) rather than in the C code.

tyama501, 2025-04-22:
Hello @ghaerr and @drachen6jp,

Thank you for the debugs!

I tested with my PC-9801RX, intel 286, 4MB xms.

[image]

I have good news and bad news.

The good news is that I could boot the PC with int1F, after commenting out the hma kernel.

[image]

[image]

I also could make 3000KB ramdisk, and mkfat to the /dev/ssd and mounted it.

[image]

The bad news is that if I make the ramdisk,
the filesystem gets something wrong.
For example, the files in /etc/ disappears. Sometimes cd does not work.
Is there any idea why this happen?

Also, I'm not sure the xms buffer is working or not.
Here is meminfo. Is there way to see where the xms buffer is placed?
[image]

Thank you!

ghaerr, 2025-04-23:
Hello @tyama501,

> I could boot the PC with int1F, after commenting out the hma kernel.

Good news, this confirms what is working on PC-98. I am working on an update for PC-98 for the other items previously mentioned.

> The bad news is that if I make the ramdisk,
the filesystem gets something wrong.
For example, the files in /etc/ disappears. 

What did you use to copy the files over to the mounted filesystem, `cp -R` ? Can you give me the exact commands used to replicate the issue?

> I also could make 3000KB ramdisk, and mkfat to the /dev/ssd and mounted it.

Can you run mkfs instead of mkfat? This will allow us to concentrate on just MINIX filesystems, for which we think the code is very bug-free. Also, MINIX has a filesystem checker.

Also, can you try creating 2000KB or 1000KB  ramdisk instead? There is the possibility that the XMS size is not being set properly, and this would check that.

Also, you can run `fsck -lvf /dev/ssd` to check MINIX filesystem integrity. This should normally be done when unmounted, but can sometimes work when mounted. Please run that to see what it might say. The listing will be long, you may have to pipe through `more`.

> Sometimes cd does not work.

You mean the shell `cd`? That is strange. Can you give me an example to replicate?

Since you are running 4MiB XMS, there should not be a hole anywhere in memory. I do not know what the problem is yet, but will investigate.

> Also, I'm not sure the xms buffer is working or not.

That might be. If the above ideas don't fix anything, try turning off CONFIG_FS_XMS_BUFFER, but keep CONFIG_FS_XMS_RAMDISK. That will enable only XMS ramdisk.

> Here is meminfo. Is there way to see where the xms buffer is placed?

No, we need to add that to meminfo. I have made a note of that. The XMS allocation currently always starts at 1M, but if hma=kernel is set, then 64K is allocated for HMA kernel, which then leaves XMS buffers to start at 1M+64K. The ram disk is allocated after the number of buffers specified. If XMS buffers are turned off, then the RAM disk will start right after HMA kernel, or at 1M if kernel not HMA.

It sounds like it also might be useful to modify `hd` so that it can display XMS RAM, not just main memory. That would allow us to debug more deeply from running system.

Thank you!

tyama501, 2025-04-23:
Hello @ghaerr 

I will try what you mentioned when I have another time,
but I haven't copy any files to the ramdisk yet.

I just created and mounted the ramdisk as the picture, then ls /etc in the floppy disk.
It cannot show the files anymore in /etc.
This can be reproduced multiple times easily.

Thank you.

tyama501, 2025-04-24:
I'm still using the same image as the day before yesterday,
but it seems that the /etc issue after ramdisk creation does not occur on the NP21/w emulator.
(There is no 4MB option on NP21/W, so I used it with 7MB but the ramdisk is created for 3MB as before)

[image]

[image]

BTW, 
When making FAT, it says FAT16 is created.
When mounting it, it says FAT12.
Is this correct behavior?

tyama501, 2025-04-30:
I agree that this NP21/W boot issue has been solved.
I close this now.

It seems that the ramdisk issue is only occurring in my system.
(It might be a hardware issue)
I will create another ticket if needed.
Thanks!

closed 2025-04-30 by tyama501

toncho11, 2025-05-10:
I created a new wiki page for the XMS memory: https://github.com/ghaerr/elks/wiki/Access-to-XMS-memory
It needs to be completed.