ghaerr/elks #2305: Porting Advice

fhendrikx, 2025-04-18:
Hi,

I'm porting ELKS to a new platform. This platform is similar, but not the same as the standard PC platform. I have modified EMU86 appropriately and this is working correctly with a "monitor" in ROM. The monitor is used to load the Elks Image and rootfs to the right segments (0xE000 and 0x8000 respectively).

The following questions are probably somewhat hard for you to answer, but I welcome any thoughts you may have about where the issues may lie or what to look out for:

1. Under EMU86 it boots correctly with no errors. However, I cannot seem to get it to read any characters from my input. Adding debug, I can see that kbd_timer() in kbd-poll.c is not being called. Which probably explains why I don't get any input: conio_poll is never called.

2. On the real hardware, it boots correctly, but seems unable to find init. It returns error code -8. See the attached screenshot.

[image]

3. On the real hardware, it seems to come to a halted state pretty quickly after hitting the login prompt. I thought this might be power saving in the idle loop, but the timer interrupt should keep that from being in that state too long.

Thanks in advance,
Ferry

ghaerr, 2025-04-18:
Hello @Ferry! Thanks for your interest in ELKS. It looks like you've already managed to come pretty far, cool! :)

> I can see that kbd_timer() in kbd-poll.c is not being called

Without special configuration, ELKS requires the hardware INT 0 timer in order to multitask applications, as well as the kernel counting down registered timeouts, which in this case ends up calling kbd_timer. I would bet that's the issue. Can you emulate INT 0? EMU86 by default doesn't, I think. My EMU86 fork may have that added, I can't remember. If not, then we'll need to hack a solution which involves configuring CONFIG_TIMER_0F, which essentially has the idle task in main.c execute a INT 0F, which maps to an INT 7 (normally printer interrupt) that ends up being attached to the original INT 0 timer code...

> On the real hardware, it boots correctly, but seems unable to find init. It returns error code -8.

So the hardware timer's working on real hardware, but /bin/init isn't on the ROM image. If you set CONFIG_SYS_NO_BININIT it will look for and exec /bin/sh instead, which should be on ROMFS. At least I think that's what's happening, errno -8 is ENOEXEC, "Exec format error", which means the executable header isn't proper. Look closely at the end of the build printfs to see what is being built into ROMFS to further debug this.

> On the real hardware, it seems to come to a halted state pretty quickly after hitting the login prompt. I thought this might be power saving in the idle loop, but the timer interrupt should keep that from being in that state too long.

Yes, the idle task executes HLT to save power. If there's no timer interrupt or CONFIG_TIMER_0F isn't set, it'll just sit there and never resume. So it seems perhaps the hardware timer isn't working on your hardware. You'll need to port the PIT code in
elks/arch/i86/kernel/timer-8254.c (or other) for your system.

ghaerr, 2025-04-18:
Now that I think of it, the hardware timer issue may be that the PIC mask is not set to allow the interrupt. The kernel keeps certain masks the way they were set by the bios, and i'm thinking this may not have been set up on your system like it is on a PC.

ghaerr, 2025-04-18:
If this is the problem, you can have the kernel set the mask for you, I think the file is irq-8259.c or something like that.

fhendrikx, 2025-04-18:
> Now that I think of it, the hardware timer issue may be that the PIC mask is not set to allow the interrupt. The kernel keeps certain masks the way they were set by the bios, and i'm thinking this may not have been set up on your system like it is on a PC.

That is definitely something to check..  thank you. 

I had also look at the EMU86 in your repositories list, and it has additional code for other platforms (mostly 8018x), but nothing obviously different for the timer stuff. 

Just added some debug to the timer_proc() in EMU86, and I can watch it count between 0 and 3000 for elks on both IBMPC and the new platform. I think this part may be working... but I like your mask idea. Will have a look at that now.

fhendrikx, 2025-04-18:
The only place I can find any "masks" are in arch/i86/kernel/irq-8259.c... which uses them for enable_irq() and disable_irq()... that said, I'm not using that file.. I have a custom irq-solo86.c. The only thing interesting in there is the irq_vector which returns irq + 20h.

fhendrikx, 2025-04-18:
Is it useful to send you a URL to the changes?

ghaerr, 2025-04-18:
Sure, I'll look at them tomorrow. 

You can copy masking code to your file, The point is to check the code to see whether our kernel actually unmasks IRQ 0 so that the CPU gets the interrupt. 

Also, by default, our interrupt handler calls the original INT 0 vector every five hardware timer interrupts. So you'll want to make sure that that vector is initialized to point to a RETI. That is in irqtab.S.

fhendrikx, 2025-04-18:
The code in irqtab.S doesn't call the original INT 0 unless ARCH_IBMPC is defined. 

My changes are here:

[https://github.com/fhendrikx/elks/tree/ARCH_SOLO86](ELKS changes for Solo)

Thanks in advance.

ghaerr, 2025-04-18:
> I have a custom irq-solo86.c. 

> The point is to check the code to see whether our kernel actually unmasks IRQ 0 so that the CPU gets the interrupt.

Your "custom" IRQ handling file function enable_irq() is empty! I'm kind of running blind here, how does your system enable/disable external interrupts? Normally the enable_irq() and disable_irq() adjust the 8259 mask register for this purpose. 

It would seem the problem you're having is that the hardware timer interrupt is not being received by the kernel.

> The code in irqtab.S doesn't seem to call the original INT 0 unless ARCH_IBMPC is defined. Might not be an issue after all?

Not an issue, sorry for the confusion. I forgot that was for IBM PC only and wasn't looking at the code.

fhendrikx, 2025-04-18:
> Your "custom" IRQ handling file function enable_irq() is empty! I'm kind of running blind here, how does your system enable/disable external interrupts? Normally the enable_irq() and disable_irq() adjust the 8259 mask register for this purpose.

Yep, that is fine. External interrupts are controlled by the device initialisation, not a mask. For the timer, I simply need to enable the port (TIMER_ENBL_PORT) and it's good to go. This is being done in the timer-8254 code.
 
> It would seem the problem you're having is that the hardware timer interrupt is not being received by the kernel.

Yeah, I agree. I must be missing something somewhere else... possibly the mapping from IRQ0 to INT 0x20?

Thanks for looking.

fhendrikx, 2025-04-18:
Probably worth saying that this board is like the PC platform, but a lot of the "supporting" chips are replaced by a single CPLD that does all the hard work. Makes some things a lot simpler.

ghaerr, 2025-04-18:
> possibly the mapping from IRQ0 to INT 0x20?

The IBM PC mapping from IRQ 0 is to INT 8. The interrupt vector table offset is specified in `irq_vector`, like you have it, so not sure what is going on. To quickly debug this, I sometimes will put a printk in the timer interrupt routine itself: timer.c::timer_tick(). That should show you whether you're ever getting an interrupt. It is also possible you're getting just one interrupt and something needs to be done to get the next one.

> External interrupts are controlled by the device initialisation, not a mask. 

Ok, got it.

>  I simply need to enable the port (TIMER_ENBL_PORT) and it's good to go. This is being done in the timer-8254 code.

Maybe double check that?

fhendrikx, 2025-04-18:
As suspected, I'm not seeing the output of the printk in the timer_tick() function. Now I need to figure out if the issue lies with EMU86, or with the ELKS changes.

ghaerr, 2025-04-18:
> Now I need to figure out if the issue lies with EMU86

I seem to remember something about counting instructions before simulating an INT 8 (IRQ 0) on EMU86... or it could have been another emulator. The point is that the HLT instruction in the idle main loop could be preventing the generation of the interrupt if that's the case. To test that, just comment out idle_halt() in the idle loop in main.c. This doesn't answer why things aren't working on real hardware though.

fhendrikx, 2025-04-20:
Right, I tracked down the issue with EMU86.. it's expecting an EOI (or something of that ilk). So I've added one, and it's now working fine. I've also tracked down the issue with the "corrupted" init process (off-by-one error when loading the ROMFS into RAM). Finally, I've also commented out the HLT in the idle loop, and all is looking much happier.

[image]

That said, I press enter, and it seems to hang... then I press CTRL-C, and the prompt comes back.. Then I type something, and it hangs... press CTRL-C and the prompt returns. Not quite sure what that is about, any thoughts?

ghaerr, 2025-04-20:
>  I press enter, and it seems to hang... then I press CTRL-C, and the prompt comes back

Now that I think of it, /bin/sh does send some ANSI strings and other stuff at every shell prompt, and it's possible your terminal emulator isn't handling it. Can you configure the system to run /bin/sash instead of /bin/sh? I'm not sure how you're building the ROMS image, but usually CONFIG_APP_TAGS=:defash ... rather than :ash ... will do the trick. That installs sash as /bin/sh, which doesn't do anything funky with the terminal.

> I've also commented out the HLT in the idle loop

Why was this required for real hardware, do you know?

fhendrikx, 2025-04-20:
Thanks for your help in this... it's been very useful. 👍 

I have CONFIG_APP_TAGS=":defash|:128k|:192k|", but when booting it says "/bin/init: Exec failed: /bin/sh (errno 2)"? 

> Why was this required for real hardware, do you know?

I'm not sure... we tested the logic again and can confirm that in the halt state, an interrupt, as it should, takes it out of that state. We tested interrupts with the timer and keyboard, and it works correctly. More investigation is required. Maybe I'm just wrong on that, and it was something else I changed. Tomorrow will tell.

toncho11, 2025-04-21:
@fhendrikx What is this hardware platform you are porting to?

fhendrikx, 2025-04-21:
> Errno 2 is ENOENT (see include/linuxmt/errno.h) which means it didn't find /bin/sh. Perhaps the :defash isn't working, you'll have to check the output of the mkromfs program at the end of the build as previously discussed to see for sure.

It wasn't working, but its actually "defsash". That command works, in that there is a shell, but still not the right one. I will need to dig into that script.

> Finally, you can punt and ls -l elks/target to see the filesystem tree that is about to be created using image/Make.image::romfs where mkromfs is used to generate the ROMFS. This could be executed manually by copying files to elks/target then executing `cd image; make romfs`.
> 
> I hope that helps.

It surely does. Thanks again.

It's coming together nicely now. Just need to sort the character translation internally, and we should be good.

fhendrikx, 2025-04-21:
> [@fhendrikx](https://github.com/fhendrikx) What is this hardware platform you are porting to?

We're porting to Solo/86, a newly developed system built around a 286. It is intended to be simple and easy to put together and exists primarily for hobbyists. We didn't want to design yet another PC compatible, so we built something clean and simple without all of the legacy. Some things are done in the tried-and-true ways, and others are wholly new. 

This effort with ELKS is to provide us with an OS and prove that everything in the hardware works correctly, before opening up the github repositories (with the designs, etc) publicly.

fhendrikx, 2025-04-22:
Just a quick update... everything is now working well with the 128K ROM. 

Thanks to @ghaerr for all your help. It is much appreciated.

I will move onto the storage hardware next.

ghaerr, 2025-04-22:
> It's coming together nicely now. 

What were the final changes you had to do to make everything work? Did more code have to be written, or was it mostly getting the configuration file(s) organized to work for your system? I'm interested in what you had to do to get things working.

> This effort with ELKS is to provide us with an OS and prove that everything in the hardware works correctly,

Nice!! Thank you for choosing ELKS to run on your new hardware! :)

fhendrikx, 2025-04-22:
> What were the final changes you had to do to make everything work?

I think I fixed the issue I was seeing with the timer some days ago, but it didn't make it in immediately. Probably because the code was right, just the definition kept in a .h file somewhere was wrong.

> Did more code have to be written, or was it mostly getting the configuration file(s) organized to work for your system? I'm interested in what you had to do to get things working.

We needed to do some more work on our UART. This handles both a physical screen/keyboard interface, and a telnet interface. Some changes to the way telnet does the initial connection to our hardware, and then ensuring we handled 0A and 0D correctly from both interfaces. 

There are some remaining questions around interrupt handling that we are discussing internally, but I don't expect it will make much difference at this point. 

Finally, I don't think I've bottomed out on the elkscmd stuff yet. Thankfully after all the changes above we were quite fine running the default ash shell. I'll look into this more as we build and debug the storage stuff next. 

> Nice!! Thank you for choosing ELKS to run on your new hardware! :)

Well, I should be thanking you for your work on ELKS. You made our choice really easy!  :)

toncho11, 2025-04-25:
@fhendrikx Let us know when you release Solo86 :) It is already interesting!

fhendrikx, 2025-04-27:
> [@fhendrikx](https://github.com/fhendrikx) Let us know when you release Solo86 :) It is already interesting!

Of course, I'm looking forward to it  :)

fhendrikx, 2025-05-09:
@ghaerr 

I'm keen to build a CF driver for Solo. As you know, the CF stuff is just standard ATA really. Any recommendations in terms of integration with existing code? I see the directhd driver is ATA too... but it's marked as "not working".

Any thoughts welcome.

ghaerr, 2025-05-09:
Hi @fhendrikx,

Yes I recommend you start from scratch using ssd.c as a base framework for the driver, it was written recently and has none of the huge cruft that the directhd.c former driver has within it.

The ssd.c (/dev/ssd) base framework has most of what you need to start, and you'll only have to write two routines - ssddev_read and ssddev_write. The framework also allows for async (IODELAY) completion, which I don't think you'll want or need - just read and write immediately in the calling kernel thread. 

The ssd.c example keeps the actual read/write device in separate files (ssd-xms.c for XMS I/O, ssd_asm.s for an 8018X SSD implementation, and ssd-test.c for testing). You'll want to copy ssd.c to a wholly new file for your driver, and probably include the read/write routines directly in a single file. You'll also see there's an ioctl implementation that allows setting the size of the CF, but that's only useful for ram disks and testing, so it will likely be able to be wholly removed.

When I recently wrote the XMS ramdisk, I was able to use the ssd.c framework as a start and got the driver finished in a couple hours. Yours will likely take longer, but feel free to copy some of the ATA init/read/write stuff out of directhd.c if you like it, otherwise just start from scratch - should be pretty straightforward.

You can choose to keep /dev/ssd as the registered block device, or perhaps better just use the major/minor numbers for /dev/hda. Things get a lot more complicated for partitions, do you think you'll need them?

Thank you!

fhendrikx, 2025-05-09:
@ghaerr thank you again, that was super useful. 

> The ssd.c (/dev/ssd) base framework has most of what you need to start, and you'll only have to write two routines - ssddev_read and ssddev_write. The framework also allows for async (IODELAY) [...]

Agreed.

> The ssd.c example keeps the actual read/write device in separate files (ssd-xms.c for XMS I/O, ssd_asm.s for an 8018X SSD implementation, and ssd-test.c for testing).

That all looks very logical. Will look to copy that.

> You'll want to copy ssd.c to a wholly new file for your driver, and probably include the read/write routines directly in a single file.

Ah, so that deviates a little from the others, any reason why?

> You'll also see there's an ioctl implementation that allows setting the size of the CF, but that's only useful for ram disks and testing, so it will likely be able to be wholly removed.

Nice.

> You can choose to keep /dev/ssd as the registered block device, or perhaps better just use the major/minor numbers for /dev/hda. Things get a lot more complicated for partitions, do you think you'll need them?

In the spirit of keeping things simple, we will probably not bother with partitions for the moment.

Any reason to use /dev/hda over /dev/ssd?

Thanks!

toncho11, 2025-05-10:
So everyone with IDE drive will be able to use this new direct HD driver?
I mean everyone with CF card and normal real IDE HDD drive on 386 or 486 for example (or higher), right?

ghaerr, 2025-05-10:
> so that deviates a little from the others, any reason why?

I thought perhaps you might have special handling to perform, but if an additional ssd-ata.c works for you, that's great.

> Any reason to use /dev/hda over /dev/ssd?

Primarily only if the device would be used like a hard drive, /dev/hda is more understandable to users. This is the kind of customization that, if wanted, may be easier just to have all in a single separate file.

> So everyone with IDE drive will be able to use this new direct HD driver?

If the actual device is rotating medium, rather than CF, I would recommend using the async version of the ssd.c code, so as to not have the kernel waiting for data.

@fhendrikx, for simplicity, perhaps go ahead and start with just using a sub driver in ssd-ata.c. We can then see whether it makes sense to then move this into its own driver file and support rotating hard drives, etc.

ghaerr, 2025-05-10:
> So everyone with IDE drive will be able to use this new direct HD driver?
> everyone with CF card and normal real IDE HDD drive 

Yes, this could be a nice way to get direct HD support, by starting with a simple framework rather than the old complicated directhd.c driver. However as always, things get more complicated if the CHS of the drive is not easy determined or reported incorrect, etc as well as drives with partitions need to have a complicated data structure included in the driver .c file so that the rest of the system can access numbered partitions automatically.

Nonetheless, I think I would prefer approaching this from scratch using the ssd framework, and then adding partition support, rather than trying to get the ancient directhd.c driver running, since there's been a lot added to the kernel since that driver was written 30 years ago.

fhendrikx, 2025-05-11:
Thanks @ghaerr.

I'll start on the ssd-ata driver, with the aim of it being generic, but simple to start.

I'll come back here to ask any questions :)

fhendrikx, 2025-05-12:
So, some initial questions that are likely to affect initial direction:

- ssddev_write and ssddev_read functions are geared towards a "single device" model (i.e. they only have a start sector and buffer per call). Presumably we'd need to be able to support master/slave configuration at the minimum?
- some of the comments talk about CHS... do you care about LBA/CHS? 

Thanks

toncho11, 2025-05-12:
OK so the initial driver will only support these devices:

[image]

and no real HDDs.

fhendrikx, 2025-05-15:
> just start with a new file (cf-ata.c, or perhaps hd-ata.c, hd-cf.c?) from the existing framework and use a device number,

Super... I will allocate a new MAJOR number... or using an existing one? 

From my understanding, I'd need to use the same major as the device on disk. "hd*" etc all use MAJOR 3. Does that make sense?

(sorry, the driver part is easy, getting into an ecosystem is always harder)

ghaerr, 2025-05-16:
You’ll want to register the block device with the the same major number 3, and then use the same minor number(s) for the non-partitioned /dev/hd* devices, such as /dev/hda, /dev/hdb, etc. This basically allows this driver to replace the BIOS or direct floppy driver, but function with CF hardware.

fhendrikx, 2025-06-22:
Okay, so I finally got some time to spend on the ATA CF driver.

This is now compiling, and running:

[image]

The driver correct detects the first CF card, and gets the right sector count for it... but I cannot read from the device.

[image]

I'll need to add some debug to work out what the issue here actually is... any tips/advice for debugging at this point?

Thanks in advance

fhendrikx, 2025-06-26:
I tracked down the bug, and it seems to be reading stuff.. or at least not reporting any errors.

Although right now I don't have a R/W filesystem anywhere to dump files to... I tried being hopeful and mounting a DOS filesystem, but that fails. So, any suggestions for creating some R/W space where I can dump "dd" files?

BTW I've also committed all of the changes to my copy of the repo.

Cheers.

[image]

ghaerr, 2025-06-26:
> it may have created a FAT32 file system, which presumably ELKS won't like?

ELKS understands and works well with FAT32. I think instead there's an issue with either reading the CF data itself, or perhaps reading the correct track/head/sector of the CF drive. In the above screenshot, the "FAT: ..." display shows a bad filesystem, which usually means incorrect sectors are being read, if the filesystem was in fact created succesfully.

In the ELKS FAT filesystem mount code, `#f=0` indicates the number of the FAT tables - obviously zero is incorrect. `floc=` indicates the starting sector for the FAT table, and `fsz=` is the FAT size, which of course should not be negative. 

You can see what a good ELKS FAT mount looks like by mounting a known good FAT floppy on an ELKS desktop system for comparison.

You might use `hd /dev/hda` to dump the CF card in hex, that should help.

> I don't have a R/W filesystem anywhere to dump files to

You can create a ramdisk on /dev/rd0 after boot, but that's probably premature at this point; instead I recommend using ELKS `hd` to inspect the boot sector (LBA 0) to check the DOS BPB, and then the FAT table, etc. This is more easily done by comparing dumping a good FAT12/16/32 filesystem on Linux to see what it looks like, then trying the same on ELKS.

fhendrikx, 2025-06-26:
Everything looks to work, but when I umount the fs, it seems to hang after ata_cf_release is called (the "close" debug outputs in the image below).

[image]

Given the ATA writes are all PIO mode, is there some way to force synchronous writing while testing this driver?

ghaerr, 2025-06-26:
If you're running synchronous I/O (that is, not CONFIG_ASYNCIO) then all reads and writes should already be synchronous. That is, the subdriver performs the write and does not return until it is completed. I assume you're using a modified version of ssd.c, with IODELAY not set.

I'm not sure what the issue is with your screenshot above, are you wanting the ata_write's to occur prior to the first close? Remember that the filesystem itself is flushed seperately at a high level with unmount. The following code in your driver would get called to flush the buffers, which sets up more I/O:
```
static void ssd_release(struct inode *inode, struct file *filp)
{       
    kdev_t dev = inode->i_rdev;

    debug_blk("SSD: release\n");
    if (--access_count == 0) {
        fsync_dev(dev);
        invalidate_inodes(dev);
        invalidate_buffers(dev);
    }
}
```
Thus, the driver may not get the write calls until the system flushes its buffers. That can be minimized with setting fewer buffers, but it can't be eliminated, since ELKS, like Linux, always buffers block I/O.

fhendrikx, 2025-06-26:
> If you're running synchronous I/O (that is, not CONFIG_ASYNCIO) then all reads and writes should already be synchronous. That is, the subdriver performs the write and does not return until it is completed. I assume you're using a modified version of ssd.c, with IODELAY not set.

Funnily enough, I'm not sure on this front, hence the question. I will check.

> I'm not sure what the issue is with your screenshot above, are you wanting the ata_write's to occur prior to the first close? Remember that the filesystem itself is flushed seperately at a high level with unmount. 

Yeah, it looks good, sadly the issue is that it hangs at that last prompt. Not sure what it's doing.

> The following code in your driver would get called to flush the buffers, which sets up more I/O:
> ```
> static void ssd_release(struct inode *inode, struct file *filp)
> {
>     kdev_t dev = [...]

Yep, the "close" debug shown the in the screenshot is from the ATA-CF driver's version of that function.

> Thus, the driver may not get the write calls until the system flushes its buffers. That can be minimized with setting fewer buffers, but it can't be eliminated, since ELKS, like Linux, always buffers block I/O.

I don't think that's an issue.

Time for more debug, thanks for the suggestions :)

ghaerr, 2025-06-28:
Glad it's working!

Go ahead and prepare a PR with your changes: newly created driver files should be very straightforward for commit, as they will only be compiled and/or built for your system/hardware, and Makefiles or other integration changes can be guarded with either a new CONFIG_ARCH_ define if required, or using a new CONFIG_BLK_DEV_ define specific to your new driver, which is probably preferred.

fhendrikx, 2025-06-29:
Thanks, a PR has been submitted.

The Solo/86 platform is using the ARCH_SOLO86 flag, while the new ATA-CF uses the new CONFIG_BLK_DEV_ATA_CF flag.

fhendrikx, 2025-06-30:
BTW I'm super glad to see someone using an old Amstrad 1640..  I had one of those back in the day, great old machines.

toncho11, 2025-07-01:
Xt IDE emulates an mfm controller. Mfm controller to IDE. 

Here is a summary from chatgpt: 

ChatGPT

You said:

Does xt IDE loads an extension BIOS in ram? 

ChatGPT said:

The XT IDE (a popular open-source IDE controller for vintage PCs like the IBM PC/XT) uses an Option ROM (often referred to as an extension BIOS or ROM BIOS) to integrate itself with the system's BIOS. Here's how it works regarding RAM and ROM:

Here's what happens:

ROM location: The XT IDE has a ROM chip (typically an EEPROM) that contains its Option ROM BIOS.

BIOS initialization: During system boot, the main BIOS scans the address space between C0000h and EFFFFh (the "Option ROM space") for ROM extensions.

ROM shadowing (optional): On some newer systems (286/386+), the contents of this ROM may be shadowed into RAM for faster execution, but on a true IBM PC/XT or compatible 8088/8086, it is executed directly from ROM.

Execution: The XT IDE ROM contains an INT 13h BIOS extension that hooks into the system, allowing the BIOS and DOS to recognize and boot from IDE drives.

toncho11, 2025-07-01:
## 📦 Basic I/O Mechanism: Talking to a CF Card on XT-IDE

When a CompactFlash (CF) card is used in **True IDE Mode**, it behaves just like a standard ATA/IDE hard drive. Communication is done using a set of **I/O port registers**, usually starting at **0x1F0** (or another XT-IDE base address, like `0x300` or `0x320`).

### 📋 IDE Register Map

| Offset (from base) | Register Name       | Direction | Description                          |
|--------------------|---------------------|-----------|--------------------------------------|
| `+0x00`            | Data                | R/W       | Read/write 16-bit data               |
| `+0x01`            | Error / Features    | R / W     | Error code (read) or features (write)|
| `+0x02`            | Sector Count        | W         | Number of sectors to read/write      |
| `+0x03`            | LBA Low             | W         | LBA bits 0–7                         |
| `+0x04`            | LBA Mid             | W         | LBA bits 8–15                        |
| `+0x05`            | LBA High            | W         | LBA bits 16–23                       |
| `+0x06`            | Drive / Head / LBA  | W         | Drive select + LBA bits 24–27       |
| `+0x07`            | Status / Command    | R / W     | Status (read), Command (write)       |

> Replace the base (e.g., `0x1F0`) with your actual XT-IDE I/O port address.

---

### 📘 Reading a Sector (Using LBA)

1. **Wait for the drive to be ready**
   - Poll the status register (`+0x07`) until `BSY=0` and `DRDY=1`.

2. **Set up registers**
   - `+0x02`: Sector count (usually 1)
   - `+0x03`–`+0x05`: LBA low/mid/high bytes
   - `+0x06`: Set LBA mode and drive/head bits
   - `+0x07`: Write command `0x20` (READ SECTOR)

3. **Wait for DRQ (Data Request)**
   - Poll status until `DRQ=1`

4. **Read 256 words (512 bytes)**
   - Repeatedly read from `+0x00` (Data register)

---

### 📘 Writing a Sector (Using LBA)

1. **Wait for DRDY**
2. **Set up LBA and sector count**
3. **Send `0x30` (WRITE SECTOR) command**
4. **Wait for DRQ**
5. **Write 256 words (512 bytes) to `+0x00`**

---

### 🛑 Status Register Bits (Offset `+0x07`)

| Bit | Name   | Meaning                             |
|-----|--------|-------------------------------------|
| 7   | BSY    | Busy (1 = drive is busy)            |
| 6   | DRDY   | Drive ready                         |
| 3   | DRQ    | Data request (ready to read/write)  |
| 0   | ERR    | Error occurred                      |

---

### 🧪 Example Command Sequence (Pseudocode)

```asm
; Wait for DRDY
in al, 0x1F7
test al, 0x40  ; DRDY
jz wait_ready

; Setup LBA address
mov al, 1
out 0x1F2, al  ; Sector count
mov al, LBA0
out 0x1F3, al
mov al, LBA1
out 0x1F4, al
mov al, LBA2
out 0x1F5, al
mov al, 0xE0   ; 0xE0 = LBA mode, master drive
out 0x1F6, al

; Issue READ SECTOR command
mov al, 0x20
out 0x1F7, al

; Wait for DRQ
wait_drq:
in al, 0x1F7
test al, 0x08
jz wait_drq

; Read 512 bytes (256 words)
mov cx, 256
read_loop:
in ax, 0x1F0
stosw
loop read_loop

fhendrikx, 2025-07-01:
@toncho11 Yep, that is exactly how the ATA-CF driver works. 

I believe @ghaerr was saying that at present, the ATA-CF driver shares it's major number with some other devices in the kernel, and that might present other issues. Once that is sorted, you can try the ATA-CF driver out on your hardware.

@ghaerr also noted that it might not make much difference to your overall speed. I tend to concur, as the overhead of calling the XT-IDE BIOS is probably low when compared to the overall cost of reading/writing 256 shorts using in and out operations.

Cheers!

toncho11, 2025-07-01:
True IDE Mode: CF card behaves like an IDE hard drive (hardware-level setting).

LBA Mode: Sector addressing method that simplifies access (protocol-level setting).

@fhendrikx Do you use lbs mode?

toncho11, 2025-07-02:
Actually the whole time I am trying to understand if it is possible to have a direct xt IDE + CF card driver. 

Chatgpt says: 

```
XT-IDE hardware does not follow standard IDE register behavior on an 8-bit bus.

Some ports are emulated or behave non-standard to allow 8-bit access to 16-bit IDE devices.

To write a direct XT-IDE driver, ELKS would need:

To speak to the custom 8-bit interface logic of the XT-IDE card.

Or reverse-engineer the behavior of XTIDE BIOS and implement it in the kernel.
```

toncho11, 2025-07-02:
# Reading a CF Card via XT-IDE in C (8-bit ISA, LBA28)

This example demonstrates how to read a **CompactFlash (CF) card** attached to an **XT-IDE card** (8-bit ISA) using direct I/O port access in **C**. This is suitable for systems like the **IBM 5160 (XT)** or similar machines, running custom real-mode software.

> ⚠️ **NOTE:** This is *not* for modern systems or full 16-bit IDE controllers. It is specifically for 8-bit XT-IDE cards using ports like `0x300`.

---

## 📟 XT-IDE I/O Port Map (Default at `0x300`)

| Port Offset | Register       | Description                          |
|-------------|----------------|--------------------------------------|
| `0x300`     | Data           | 8-bit data register (LSB only)       |
| `0x301`     | Error / Features | Error status or feature flags       |
| `0x302`     | Sector Count   | Number of sectors to transfer        |
| `0x303`     | LBA Low        | LBA bits 0–7                         |
| `0x304`     | LBA Mid        | LBA bits 8–15                        |
| `0x305`     | LBA High       | LBA bits 16–23                       |
| `0x306`     | Drive/Head     | LBA bits 24–27 and device select     |
| `0x307`     | Command / Status | Send commands or check status      |

---

## ✅ Sample C Code

```c
#include <dos.h>
#include <stdint.h>
#include <conio.h>

#define IDE_BASE 0x300
#define IDE_DATA        (IDE_BASE + 0)
#define IDE_ERROR       (IDE_BASE + 1)
#define IDE_SECTOR_CNT  (IDE_BASE + 2)
#define IDE_LBA_LOW     (IDE_BASE + 3)
#define IDE_LBA_MID     (IDE_BASE + 4)
#define IDE_LBA_HIGH    (IDE_BASE + 5)
#define IDE_DRIVE_HEAD  (IDE_BASE + 6)
#define IDE_COMMAND     (IDE_BASE + 7)
#define IDE_STATUS      (IDE_BASE + 7)

// IDE command to read a sector
#define CMD_READ_SECTORS  0x20

// Wait until drive is not busy and DRQ is set
int ide_wait_ready() {
    uint8_t status;
    while (1) {
        status = inp(IDE_STATUS);
        if (!(status & 0x80)) break;  // BSY clear
    }
    return (status & 0x08) ? 0 : -1;  // DRQ set?
}

// Read a 512-byte sector via LBA28 into buffer
int ide_read_sector(uint32_t lba, uint8_t *buffer) {
    int i;

    if (ide_wait_ready() != 0)
        return -1;

    // Set up LBA address and sector count
    outp(IDE_SECTOR_CNT, 1);
    outp(IDE_LBA_LOW,  (uint8_t)(lba & 0xFF));
    outp(IDE_LBA_MID,  (uint8_t)((lba >> 8) & 0xFF));
    outp(IDE_LBA_HIGH, (uint8_t)((lba >> 16) & 0xFF));
    outp(IDE_DRIVE_HEAD, 0xE0 | ((lba >> 24) & 0x0F));  // Master, LBA mode

    outp(IDE_COMMAND, CMD_READ_SECTORS);

    if (ide_wait_ready() != 0)
        return -2;

    // Read 512 bytes (1 sector) via 8-bit port
    for (i = 0; i < 512; i++) {
        buffer[i] = inp(IDE_DATA);
    }

    return 0;
}

// --- Example usage ---
void main() {
    uint8_t sector[512];
    int i;

    clrscr();
    if (ide_read_sector(0, sector) == 0) {
        printf("Sector 0 read:\n");
        for (i = 0; i < 512; i++) {
            if (i % 16 == 0) printf("\n%04X: ", i);
            printf("%02X ", sector[i]);
        }
    } else {
        printf("Failed to read sector.\n");
    }

    getch();
}

Notes & Limitations
✅ LBA-only mode
This example uses 28-bit Logical Block Addressing (LBA), which supports drives up to 128 GiB. CHS is not supported in this example.

🔄 No DMA or interrupts
All I/O is done using Programmed I/O (PIO) through inp() and outp() in 8-bit mode — there is no DMA or IRQ handling.

🧠 8-bit-only support
This is designed for 8088/8086-class CPUs with 8-bit XT-IDE cards. It will not work correctly on standard 16-bit IDE interfaces without modification.

💾 CF card compatibility
Not all CF cards work with XT-IDE in 8-bit mode. Known good cards include:

SanDisk Ultra (≤ 4GB)

Transcend Industrial

Older 512MB–2GB models
Avoid newer or UDMA-only CF cards.

🚫 No BIOS usage
This code bypasses any BIOS and accesses hardware directly. It is designed for use in environments where the XTIDE Universal BIOS is present, but not used (e.g., custom OS like ELKS).

🐢 Slow access
Due to byte-wise transfers over the 8-bit bus and polling, performance is very limited (~30–100 KB/s depending on CPU speed).

toncho11, 2025-07-03:
This should be enough I think:

- every 16-bit transfer becomes two 8-bit I/O operations

- data, registers, and status — all must be handled byte-wise in the correct ATA sequence

fhendrikx, 2025-07-03:
> This should be enough I think:
> 
> * every 16-bit transfer becomes two 8-bit I/O operations

Yes, that makes sense.
  
> * data, registers, and status — all must be handled byte-wise in the correct ATA sequence

There seem to be, interestingly, two schools of thought on the order of operations. However, the one I have implemented seemed to work with everything I could throw at it. Which is a good sign, since different cards, etc have different tolerances and levels of ATA implementation.

fhendrikx, 2025-07-04:
> Also, there was a big bug with INITPROC being used but init.h not included, so the Solo/86 repo will want to get updated after testing this PR.

Hmm... I didn't have any issues with compilation... but I'll be standing by to test when your PR comes through.

> Instead of sharing the major device number 3 between ATA-CF and BIOSHD which disallowed their simultaneous inclusion and use within the kernel, I came up with a new idea I'm calling "block device overlays"

Nice idea.

> For SOLO/86, it was desired to replace /dev/hda and /dev/hdb (major 3) with the ATA-CF driver, so that users could just mount or boot from /dev/hda, without knowing it's actually /dev/cf0 (or cf1). Thus the ATA-CF driver overlays the BIOSHD driver now for SOLO/86 and everything should work (although it's untested on SOLO/86).

I'm not too worried about the final device names for Solo/86.. happy to use /dev/hda or /dev/cf0. That said, if the overlays are working, I'll happily use those too.

 
> In the future, we may want to replace the BIOSHD hard drive driver with the ATA driver as a "direct HD" driver, thus not having to use the BIOS. This will be easily done by overlaying ATA-CF onto BIOSHD for general purpose ELKS, rather than having to use a separate /dev/cf0 etc. What's nice about this arrangement is that only the BIOS HD driver is overlaid, the BIOS FD driver remains working (which isn't ATA compatible). This same mechanism can also be potentially used for the direct FD driver, replacing /dev/df0 with /dev/fd0, and possibly changeable at runtime, rather than recompiling the kernel. This lets various drivers substitute for the well known /dev/fdX and /dev/hdX driver names.

It's a nice tidy idea.

> > every 16-bit transfer becomes two 8-bit I/O operations
> 
> I haven't done this yet, but will change the 16-bit I/O to two 8-bit operations when the CPU is 8086 or 80186.

Both of these CPUs have 16-bit IO. It might be better to make it some kind of feature flag? (assuming it works)

Cheers, Ferry

fhendrikx, 2025-07-04:
@ghaerr RE 8-bit transfer mode... I've just been trawling through the ATA-2 spec document (circa 1996), and we'll also need to explicitly set the 8-bit transfer mode using the "SET FEATURES" command.

Overall, 8-bit mode is not well documented. I can't see of any way to "detect" 8-bit support ahead of time, so some experimentation may be required. According to what I've read, some devices silently ignore 8-bit transfer mode.

As you have the hood open, I'll leave you to work out the exact set up, but you'll need to add something like the following:

ata.h:

`#define ATA_CMD_FEAT 0xEF`

ata.c:

```
void ata_set8bitmode()
{
    // set 8-bit transfer mode

    outb(0x01, ATA_PORT_FEAT);
    outb(ATA_CMD_FEAT, ATA_PORT_CMD);

    // wait for drive to be not-busy

    for (i = 0; i < ATA_RETRY; i++)
    {
        status = ata_wait(ATA_STATUS_BSY);

        // are we done?

        if (status == 0)
            break;
    }

    if (status)
        return (status);

    // check for error

    status = inb(ATA_PORT_STATUS);

    if (status & ATA_STATUS_ERR)
        return (-EIO);

    return (1);
}
```

Cheers, Ferry

ghaerr, 2025-07-04:
> I didn't have any issues with compilation

On my system, I saw a "void INITPROC ata_init(void)" not previously defined warning, which could mean that INITPROC didn't get the FAR declaration required. But init.h was included, so INITPROC was probably defined and the warning was purely a recommendation. So no big deal now that I think of it. Usually errors like this would cause an immediate system crash from the mismatched near/far function call.

> Both of these CPUs have 16-bit IO. 

Yes - I was thinking about XT system bus being 8-bit only, and wondering whether there is any bus logic to convert 16-bit I/O to two 8-bit cycles. I think not. But your idea of setting the ATA chip into 8-bit mode is a good one. I'll take a first pass at it, then you might play with it to see if you can get it working on Solo/86 hardware and/or given @toncho11's test response.

> It might be better to make it some kind of feature flag?

We can easily test for CPU type, and then either use 16-bit like you wrote the driver, or 8-bit I/O only for the XT case. I should be able to test the 8-bit case if QEMU emulates ATA_CMD_FEAT properly... we'll soon find out!

Thanks for the code example.

toncho11, 2025-07-04:
<div class="flex basis-auto flex-col -mb-(--composer-overlap-px) [--composer-overlap-px:55px] grow overflow-hidden"><div class="relative h-full"><div class="flex h-full flex-col overflow-y-auto [scrollbar-gutter:stable_both-edges] @[84rem]/thread:pt-(--header-height)"><div class="@thread-xl/thread:pt-header-height flex flex-col text-sm keyboard-open:pb-[calc(var(--composer-height,100px)+var(--screen-keyboard-height,0))] pb-25"><article class="text-token-text-primary w-full" dir="auto" data-testid="conversation-turn-46" data-scroll-anchor="false"><div class="text-base my-auto mx-auto py-5 [--thread-content-margin:--spacing(4)] @[37rem]:[--thread-content-margin:--spacing(6)] @[72rem]:[--thread-content-margin:--spacing(16)] px-(--thread-content-margin)"><div class="[--thread-content-max-width:32rem] @[34rem]:[--thread-content-max-width:40rem] @[64rem]:[--thread-content-max-width:48rem] mx-auto flex max-w-(--thread-content-max-width) flex-1 text-base gap-4 md:gap-5 lg:gap-6 group/turn-messages focus-visible:outline-hidden" tabindex="-1"><div class="group/conversation-turn relative flex w-full min-w-0 flex-col agent-turn"><div class="relative flex-col gap-1 md:gap-3"><div class="flex max-w-full flex-col grow"><div data-message-author-role="assistant" data-message-id="4f621d83-2fe8-45e4-9095-4071976009b9" dir="auto" class="min-h-8 text-message relative flex w-full flex-col items-end gap-2 text-start break-words whitespace-normal [.text-message+&amp;]:mt-5" data-message-model-slug="gpt-4o"><div class="flex w-full flex-col gap-1 empty:hidden first:pt-[3px]"><div class="markdown prose dark:prose-invert w-full break-words dark"><p data-start="962" data-end="1094">XT-IDE BIOS or a custom driver <strong data-start="993" data-end="1022">exploits ATA's 8-bit mode</strong> to make ATA/CF drives work with the <strong data-start="1059" data-end="1081">8-bit-only ISA bus</strong> of the 8088.</p>
<h3 data-start="1096" data-end="1106">Steps:</h3>
<ol data-start="1108" data-end="1634">
<li data-start="1108" data-end="1286">
<p data-start="1111" data-end="1133"><strong data-start="1111" data-end="1132">Enable 8-bit mode</strong>:</p>
<ul data-start="1137" data-end="1286">
<li data-start="1137" data-end="1212">
<p data-start="1139" data-end="1212">Send <code data-start="1144" data-end="1165">SET FEATURES (0xEF)</code> command with feature code <code data-start="1192" data-end="1198">0x01</code> to the drive.</p>
</li>
<li data-start="1216" data-end="1286">
<p data-start="1218" data-end="1286">This tells the CF card or IDE drive to accept single-byte transfers.</p>
</li>
</ul>
</li>
<li data-start="1288" data-end="1496">
<p data-start="1291" data-end="1326"><strong data-start="1291" data-end="1325">Use only low byte of data port</strong>:</p>
<ul data-start="1330" data-end="1496">
<li data-start="1330" data-end="1425">
<p data-start="1332" data-end="1425">XT-IDE BIOS or driver performs I/O using <code data-start="1373" data-end="1377">IN</code>/<code data-start="1378" data-end="1383">OUT</code> on port <code data-start="1392" data-end="1400">0x01F0</code>, ignoring the high byte.</p>
</li>
<li data-start="1429" data-end="1496">
<p data-start="1431" data-end="1496">Drive/CF card adapts to this by internally managing 8-bit access.</p>
</li>
</ul>
</li>
<li data-start="1498" data-end="1634">
<p data-start="1501" data-end="1543"><strong data-start="1501" data-end="1531">Perform data I/O 512 times</strong> per sector:</p>
<ul data-start="1547" data-end="1634">
<li data-start="1547" data-end="1634">
<p data-start="1549" data-end="1634">Instead of 256 16-bit transfers per 512-byte sector, XT-IDE does 512 8-bit transfers.</p>
</li>
</ul>
</li>
</ol>
<hr data-start="1636" data-end="1639">
<h2 data-start="1641" data-end="1662">🧠 Why It’s Needed</h2>
<div class="_tableContainer_80l1q_1"><div tabindex="-1" class="_tableWrapper_80l1q_14 group flex w-fit flex-col-reverse">
8088 Limitation | Effect
-- | --
8-bit data bus | Cannot read/write 16-bit I/O in one cycle
IN/OUT limited to 8-bit | Only IN AL, port and OUT port, AL are fast
16-bit IN AX needs 2 cycles | Slow, not atomic — not safe for ATA

ghaerr, 2025-07-04:
It looks like QEMU does not support ATA 8-bit mode. So I'll probably punt and write as much as I can that's hopefully correct and then have someone else test on real hardware. @fhendrikx, do you know whether the Solo/86 hardware supports an 8-bit mode?

fhendrikx, 2025-07-05:
> It looks like QEMU does not support ATA 8-bit mode. So I'll probably punt and write as much as I can that's hopefully correct and then have someone else test on real hardware. [@fhendrikx](https://github.com/fhendrikx), do you know whether the Solo/86 hardware supports an 8-bit mode?

It's all generic ATA CF stuff.... and it depends more on the CF card than anything else. I'm more than happy to try it out.

toncho11, 2025-07-05:
@fhendrikx So setting the 8 bit mode and performing double reads and writes should in theory work on any CF card that supports 8 bit mode, not just XT-IDE?

[14 more replies]