ghaerr/elks #2313: fmemcpy doesn't restore DS

FrenkelS, 2025-04-20:
Elksdoom has its own implementation of a far memcpy: [`_fmemcpy`](https://github.com/FrenkelS/elksdoom/blob/main/i_elks.c#L76). The game crashes when ELKS's implementation is used, [`fmemcpy`](https://github.com/ghaerr/elks/blob/master/libc/watcom/asm/fmemcpy.asm).

But it does work if `ds` is restored at the end of `fmemcpy`. For example with the following code:
```
push ss
pop ds
```

ghaerr, 2025-04-20:
> But it does work if ds is restored at the end of fmemcpy

Thank you @FrenkelS, this answers a question I have had for some time about OpenWatom ASM routines: whether the value of DS needs to be saved or not. I had assumed that the value was scratch, as it gets loaded by other Watcom-generated code; but apparently Watcom expects either the same value to be in DS across function calls, or that it expects DS == SS.

I will fix ELKS `fmemcpy`. I think the best solution will be to fully save DS by `push ds` and restore afterwards though, just in case DS has been set != SS in previous Watcom generated code?

Thank you!

FrenkelS, 2025-04-20:
> I think the best solution will be to fully save DS by `push ds` and restore afterwards though, just in case DS has been set != SS in previous Watcom generated code?

Oh right, of course. I've been using the medium memory a lot lately, where DS = SS. But that isn't the case in the large memory model. So your solution is better than mine.

closed 2025-04-24 by ghaerr