ghaerr/elks #2317: Research ZX0 instead of Exomizer decompression

asiekierka, 2025-04-21:
> If you want to play with LZ4_8088, the first step would be compressing executables with it, and see how small they get. Using Exomizer, we're seeing about a ~28-33% decrease in size.

Out of curiosity, I checked a few compressors: Exomizer (as used by ELKS), LZ4 (as suggested by Vutshi), and ZX0 (my personal go-to on retro platforms):

-rw-r--r-- 1 asie asie 31482 Apr 21 19:37 ash.exo
-rwxr-xr-x 1 asie asie 37565 Apr 21 19:21 ash.lz4
-rw-r--r-- 1 asie asie 31367 Apr 21 19:38 ash.zx0
-rwxr-xr-x 1 asie asie  4755 Apr 21 19:49 login.exo
-rwxr-xr-x 1 asie asie  5620 Apr 21 19:21 login.lz4
-rw-r--r-- 1 asie asie  4655 Apr 21 19:50 login.zx0

ZX0 turns out to be an interesting alternative choice:

- it presents compression ratios similar to or narrowly beating Exomizer...
- requiring a different (per-file) "gap" between compressed and decompressed data. For example, `login` requires 35 bytes reserved, while `ash` requires 3 bytes reserved,
- and with potentially faster decompression speeds, owing to an optimized native 8088/8086 decompressor existing [in under 100 bytes](https://github.com/emmanuel-marty/salvador/blob/main/asm/8088/unzx0_8088.S)!

As such, I think it deserves further evaluation.

Vutshi, 2025-04-21:
@asiekierka, have you tried LH4-HC? It is supposed to compress stronger while being as fast to decompress as normal mode of LH4.

asiekierka, 2025-04-21:
> [@asiekierka](https://github.com/asiekierka), have you tried LH4-HC? It is supposed to compress stronger while being as fast to decompress as normal mode of LH4.

I used `lz4 --best` to compress for the test above, which shold be equivalent to `LZ4-HC`. The standard `lz4` is even less favorable:

```
-rwxr-xr-x 1 asie asie 48720 Apr 21 20:02 ash
-rwxr-xr-x 1 asie asie 37565 Apr 21 19:21 ash.lz4
-rwxr-xr-x 1 asie asie 40198 Apr 21 20:02 ash.lz4_1
```

ghaerr, 2025-04-22:
> Out of curiosity, I checked a few compressors

Thanks @asiekierka for running the tests! The 35% reduction in size of ash by Exomizer is even more than the 33% I predicted. LZ4 isn't doing nearly as well at 23%.

The couple of points you bring up for ZX0 (compression % near Exomizer and ability to decompress in place) are necessary to replace Exomizer, but still not sufficient. 

We still have to have a known size ("gap") that cannot be exceeded by any compressed file during decompression. The reason for this is that the a.out header till holds the decompressed section sizes for .text, .ftext and .data; these are allocated before decompression in sys_exec() along with the guaranteed maximum gap, which for Exomizer is 16 bytes. 

Also, it's not enough to run the "standard" compressor - a custom compressor has to be built that individually compresses each a.out section separately, and then rewrites the a.out header. This means the compressor needs to be able to be converted to library format, and then built into our elks-compress tool in elks/tools/.

As @asiekierka asked about the other day, it already takes twice the forking process' data size to exec a new program, and if the max decompress size is not known, then a full 64k would have to be additionally (hopefully) allocated somewhere. Of course, this could be guessed smaller, but failure to guess correctly results in the inability to exec the program.

Can ZX0 guarantee a known gap that is never exceeded? If not, I would rule out it's use for ELKS.

The next big issue is resources required for decompression. Exomizer uses a fixed decompressor table never exceeding 156 bytes. What size does ZX0 require? BTW, 12-bit LZW (at least the one I wrote for Nano-X) requires a decompressor dictionary of 32k bytes. This has to fit in the kernel data segment so it's a big deal.

The current Exomizer decompressor is written in C in fs/exodecr.c.  Exomizer also has an 8086 ASM version of its decompressor in elks/tools/elks-compress/exomizer/rawdecrs/8086/P47/deexo.asm.txt. It may very well be worth the time to use that decompressor (note: P47 only), or write an ASM version from C. This would surely speed things up, but I'm not really sure by how much. I wasn't able to test the 8086 ASM decompressor during research because I couldn't run the code directly on my macOS laptop.

Given that ZX0 and Exomizer's compression %'s look very similar, I would be quite surprised to learn that ZX0's decompressor is much or any faster than Exomizer's. And even more importantly, I don't think that most 8088 users will ever want a system that uses compressed binaries - they're just too slow to load. In basic testing, the savings in reading floppy disk sectors about but didn't quite make up for the increased time to decompress. For HD users, definitely not worth the wait.

There's a bunch of test code checked in under elks/tools/elks-compress/exomizer/rawdecrs that might be useful should someone really want to compare Exomizer with other decompressors, but frankly I see the whole thing as somewhat as a lost cause. I spent way too many hours comparing decompressors, testing various compression modes, expanding the a.out header format, and then getting it all working, only to find that on older systems... it's just too slow. The only reason we're using it now for distribution media is that ELKS just has too many useful executables to fit on almost any image size. We might be better off to try to figure a way to distribute multi-disk images that work in a single floppy drive, perhaps.

> an optimized native 8088/8086 decompressor existing [in under 100 bytes](https://github.com/emmanuel-marty/salvador/blob/main/asm/8088/unzx0_8088.S)!

That's nice, it does look quite small. Does it not have any external data requirements? I can't quite tell from looking at it.

> it may also provide faster decompression speeds

Is there C code that can be used for testing decompression speeds? The results could be compared with the testing I did in rawdecr/s, if someone really wants to do it.

> I think it deserves further evaluation.

My opinion is that it's not worth my effort to try getting another executable compression system that'll operate within people's expectations on 8088. But if someone wants to get involved and make it all work, I'll be glad to help. Serious timing comparisons should be done first so its not a waste of effort.

asiekierka, 2025-04-22:
> Can ZX0 guarantee a known gap that is never exceeded?

Not with the standard compressor. Maybe it can be modified to guarantee such a gap; but I was thinking of extending a header format to include this information, myself.

> Exomizer uses a fixed decompressor table never exceeding 156 bytes. What size does ZX0 require?

0 bytes. There is no table; there is just the decompression algorithm I linked.

> Is there C code that can be used for testing decompression speeds? The results could be compared with the testing I did in rawdecr/s, if someone really wants to do it.

I don't think testing on a modern machine is a necessarily good proxy of decompression performance on a retro machine. At least on the Z80 chip, ZX0 was [substantially faster](https://www.cpcwiki.eu/forum/programming/new-cruncher-zx0/) to decompress than Exomizer; but I think any decompression testing has to be done on an 8088, ideally comparing both formats' C and ASM implementations. That is, indeed, a lot of work, however.

ghaerr, 2025-04-22:
>  I was thinking of extending a header format to include this information, myself.

You mean extending a.out with yet more fields, to show the gap length required, as well as the original sizes, each, for text, ftext and data? That's a lot of extension and also invalidates lots of tools that would have to be updated to handle the new format; not to mention the file would have to be both compressed and decompressed to determine the gap, just to fill it in the header.

> At least on the Z80 chip

Has ZX0 been used as an executable compressor, done in-place? Or has its past use mostly been for general decompression, such as to a file where it doesn't matter what the gap is?

> ZX0 was [substantially faster](https://www.cpcwiki.eu/forum/programming/new-cruncher-zx0/) to decompress than Exomizer

> ZX0      : 10718
AP-Lib   : 10892
Exomizer : 11026
LZSA2    : 11415
ZX7      : 11651

What are those numbers? Msecs, or Secs? If Msecs, not much difference, really.

> Average ratio like exomizer, very slow crunching (i mean, VERY) but faster decrunch than exomizer

Very slow crunching is not great when you're developing and making test images... I already thought exocrunch was extremely slow.

> That is, indeed, a lot of work, however.

Yeah - too much for me, since I've already implemented a pretty fast scheme once! And users don't care how much time we put into it, it just needs to be *fast* or its no good. Sigh. 

There's a number of other speed-related 8088 issues that might be more interesting to tackle: take console ANSI sequence processing, including all the other kernel overhead. Run `sl` to see what I mean. It can run 20x slower on 8088's than other desktops.

asiekierka, 2025-04-22:
> Has ZX0 been used as an executable compressor, done in-place? Or has its past use mostly been for general decompression, such as to a file where it doesn't matter what the gap is?

It has been intended for use on the ZX Spectrum, where due to tight memory constraints you often want to decompress graphical data (for, say, splash screens) in-place. So, it didn't matter how large exactly the gap is, just that it was small - and one would usually work with fixed data for this purpose. That's why I assume nobody made the gap configurable.

> What are those numbers? Msecs, or Secs? If Msecs, not much difference, really.

Bytes. You're looking for this quote (and the graph below it, which charts decompression speed relative to compression ratio):

"ZX0 seems to be as fast as ZX7 which is 4x slower than LDIR (exomizer is 15x slower than LDIR and fastest decrunch like LZ4 are 1.5x slower than LDIR)"

> Very slow crunching is not great when you're developing and making test images... I already thought exocrunch was extremely slow.

That's for the original compressor by einar-saukas; emmanuel-marty's salvador is significantly faster (and with a slightly better compression ratio, too), though I haven't checked exactly how fast it is to compress. This is not an entirely fair comparison, but it looks promising:

```
$ time wf-zx0-salvador -stats ash ash.zx0
[...]

real	0m0.666s
user	0m0.640s
sys	0m0.013s

$ time ../../elks/tools/bin/elks-compress ash
ash: compressed from 48720 to 31697 (35% reduction)

real	0m0.978s
user	0m0.904s
sys	0m0.063s
```

asiekierka, 2025-04-22:
> You mean extending a.out with yet more fields, to show the gap length required, as well as the original sizes, each, for text, ftext and data?

Looking into this before bed - there already is an `esh_reserved` field, and to simplify, it could simply store the maximum gap required across all compressed sections (since there's nothing wrong with reserving a larger gap, merely with reserving a smaller one). 

(In fact, if one were to reduce the gap size to paragraphs, and reject files with an unusually large gap, one could limit it to an 8-bit value...)

> not to mention the file would have to be both compressed and decompressed to determine the gap, just to fill it in the header.

That's incorrect. The gap size is reported by the compressor as part of the compression process; one does not need to decompress the file for this.

ghaerr, 2025-04-23:
> there already is an esh_reserved field

I don't think it a wise idea to use any of our extremely scarce reserved fields for such a thing as decompressor gap. Partly because I think there's better ways but also this would tie an executable header to specific decompressor. There is no version or compressor ID, and frankly, subjects the rest of the tools with added complexity for something we don't even know we want or will get used.

> The gap size is reported by the compressor as part of the compression process; one does not need to decompress the file for this.

> reject files with an unusually large gap, one could limit it to an 8-bit value...)

Very good news here: if the gap size is known, and you think that the gap could be in 8 bits, then I would suggest, after doing more research on all the executables in /bin, that the max gap be set as a constant, stick it in a .h file, reject compressing any files with a larger gap, and having sys_exec always allocate the executable decompressed size + constant gap every time. This simplifies things enormously.

> The proven ones being smaller decompressor, faster compressor, and similar compression ratio;

I view those not so much as advantages, but as being most equal, and not of much importance, especially for all the work involved. The current compressor is 703 bytes C code, new one 100 - not a big deal, especially if we just rewrite our decompressor - it'll be smaller and likely lots faster.

> the unproven ones being faster decompressor

IMO that's the only thing that matters - and we don't really have any tested data points on it.

> The only disadvantage is that it requires breaking the existing Exomizer executables or supporting both formats 

I'm not sure that's a disadvantage, as I'm not sure there's many users that have custom executables that wouldn't run on an updated kernel decompressor - users can easily replace the standard executables with new versions if really needed. I would not think we'd want to support both formats. 

My conclusion remains that considering a new decompressor is lots of time for mostly no results, especially from users; and that if we really care about decompression speed someone should rewrite the existing compressor in ASM so that we at least compare apples and apples with ZX0's ASM decompressor, before more time is spent on a complicated project.