ghaerr/elks #2340: PCM player for elks 0.9

drachen6jp, 2025-05-23:
https://github.com/drachen6jp/ems_pcm

Previously, I wrote software to play PCM with beep.
However, it was unable to run under the current elks xms system environment, so I have improved it.

It is written on the assumption that FS and GS are already set to flat 4G in unreal mode. On a 32-bit CPU, unreal mode is used to expand PCM to memory, and IO access to the 8254 is similarly performed during playback.

I decided to use EMS for 16-bit machines.
It worked on a PCEM award286 machine, but I don't have the actual machine.
For PC-98, I have three types of EMS boards, so it is compatible with those. It is also compatible with the NekoProject2 emulator.

I will rewrite it again once the OS itself supports EMS.
Well then,

drachen6jp, 2025-05-23:
Thank you for your reply.

>Is this for 386 CPU only, or would you like it to run on 286 also?

Both can play PCM.386 system uses highmemory with unrealmode,and 16bit machines use low memory bank access with EMS.

I considered using xms and loadall for block transfer, but first of all there was a speed problem.

This program does what it can during an interrupt. For a 286 12MHz class, it assumes a cycle of about 4000Hz. It seemed that block transfer combined with loadall would not be able to keep up with that.

So I decided to try using bank memory this time. It was a challenge. Also, using EMS would allow 8086 and V30 to use a lot of memory, so I was hoping for that. When I tried this on a real machine, I didn't get the results I expected (V30 can only play at a low quality of about 2000Hz (just at the edge of the human audible range)).

I understand that it would be very, very difficult for elks to support EMS. It was also difficult to control the hardware EMS to write this program. I was being selfish and hoping that the OS would support this. I'm happy to hear that xms will be released to user programs.

Thank you!

rafael2k, 2025-05-23:
Hey, I had some high workload these times. I get this when trying to compile with Watcom C compiler:

```
owcc -c -O2 -mcmodel=l -bnone -march=i86 -std=c99 -fno-stack-check -fnostdlib -Wc,-fpi87 -Wc,-zev -Wc,-zls -Wc,-x -Wc,-wcd=303 -Wall -Wextra -I/home/rafael2k/programs/devel/elks/libc/include -I/home/rafael2k/programs/devel/elks/elks/include -I/home/rafael2k/programs/devel/elks/libc/include/watcom -D__ELKS__ -DNDEBUG -o emspcm.obj emspcm.c
emspcm.c(59): Error! E1055: Unable to open 'i86.h'
emspcm.c(1116): Warning! W131: No prototype found for function 'FP_SEG'
emspcm.c(1118): Warning! W131: No prototype found for function 'FP_OFF'
emspcm.c(1601): Error! E1031: Name 'main_used' not found in struct/union mem_usage
emspcm.c(1601): Error! E1031: Name 'main_used' not found in struct/union mem_usage
emspcm.c(1601): Error! E1031: Name 'main_free' not found in struct/union mem_usage
emspcm.c(1601): Error! E1031: Name 'main_free' not found in struct/union mem_usage
emspcm.c(1603): Error! E1031: Name 'xms_used' not found in struct/union mem_usage
emspcm.c(1603): Error! E1031: Name 'xms_used' not found in struct/union mem_usage
emspcm.c(1603): Error! E1031: Name 'xms_free' not found in struct/union mem_usage
emspcm.c(1603): Error! E1031: Name 'xms_free' not found in struct/union mem_usage
emspcm.c(1605): Error! E1031: Name 'xms_used' not found in struct/union mem_usage
```
I committed the Makefile here:
https://github.com/rafael2k/ems_pcm

ghaerr, 2025-05-24:
@rafael2k:

> emspcm.c(1601): Error! E1031: Name 'main_used' not found in struct/union mem_usage

The field names in the struct mem_usage in linuxmt/mem.h change recently - these errors require that you pull the latest sources in order to compile.

@drachen6jp: what C compiler are you using to compile your project?

drachen6jp, 2025-05-24:
Hi.@rafael2k.
thanks your writing Makefile.
>ps: In the past I also tried to run your player in real hardware 

elks 0.9 have GDT for unreal mode.so past code  was gone.
I think past code can't run real 286(IBM-PC) or my code had many bugs why you cannot play pcm with before0.8.

>@drachen6jp: what C compiler are you using to compile your project?

openwatcom 2.0 on gentoo linux x86_64.

drachen6jp, 2025-05-24:
by the way.
Can IBM-PC DMA Memory-memory transfer  move Highmemory to Lowmemory ?
PC-98 DMA cannot do it.

If IBM DMA can move them, this is very useful.

tyama501, 2025-06-22:
Hello @ghaerr ,

I think I have pulled the latest elks and recompiled the libc,
but I also have Error! E1055: Unable to open 'i86.h' if I do ewcc emspcm.c

[image]

Does this header still available?

ghaerr, 2025-06-22:
Hello @tyama501,

The i86.h header is in the OWC bld/hdr/dos/ directory and is not included in ELKS libc because this is a non-standard C header and are primarily DOS functions. You can likely get the program compiled by copying this header to the ems_pcm directory.

Thank you!

tyama501, 2025-06-23:
Hello @ghaerr ,

I see.
Yes, now I could compile it.

Thank you.

tyama501, 2025-06-29:
It seems that ChatGPT used some python library.
This noise does not exist in the original sound so it is limitation of this hardware.

closed 2025-11-19 by ghaerr