ghaerr/elks #2344: What are all the system calls. in assembly

MetalRules0, 2025-06-28:
Hey, i was wondering where the documentation about what the system calls where for coding in assembly and how do i use them. thanks.

MetalRules0, 2025-06-30:
i see the documentation, but it isnt very clear about what specific registers do in each system call. i was wondering if there is a tech help manual or ctyme style documentation for programing in assembly

rafael2k, 2025-09-18:
Here you can find some userland assembly examples using as86 assembler, which is pretty neat to use, and can run natively on ELKS:
https://github.com/ghaerr/8086-toolchain/tree/dev86/examples

rafael2k, 2025-09-30:
And we don't need to rule out nasm, widely used around, and can create .obj objects which could be read by wlink I think, or as86 objects, to be linked with ld86 (need to check if latest changes made something incompatible.

Example of a nice port to have: https://github.com/rafael2k/badapple

closed 2025-11-19 by ghaerr