ghaerr/elks #2372: Problem with df0

toncho11, 2025-07-26:
I tried dd with of=/dev/df0 and it did not work. It says something about not enough space. I tried the same command with fd0 and it worked.

ghaerr, 2025-07-26:
Can you give me the exact command you're running using ELKS `dd` that fails with of=/dev/df0?

Yes, they should be interchangeable, but there may be issues if you've booted from fd0 then try using df0. You should only use one or the other.

toncho11, 2025-07-26:
I see. I booted from fd0 indeed.
`dd if=image of=/dev/df0 bs=512`

ghaerr, 2025-07-26:
>  I booted from fd0 indeed.
> dd if=image of=/dev/df0 bs=512

I think we're ok booting from /dev/fd0, but you would need to set root=/dev/df0 in /bootopts to force the kernel to only use the DF driver, never the BIOS driver. 

If you have root=/dev/df0 in /bootopts, I would say this is a bug, otherwise, I would suggest that it is too much effort to track down a fix since the BIOS and DF drive could naturally overwrite certain things with regards to the floppy controller or presumed BIOS settings.

ghaerr, 2025-07-27:
> but you would need to set root=/dev/df0 in /bootopts to force the kernel to only use the DF driver

@toncho11: another thing to think about, even if booting from /dev/df0, then using `dd` to write another image to /dev/df0 - this  tries to rewrite raw image data to an actively-mounted (root) filesystem, which is has a high probability of corrupting data without knowing it. That is, the system is running and may at any time sync unfleshed system buffers to the (original) root filesystem, potentially overwriting what was just written. Likewise, the system may read from the root filesystem and read garbage using data just written from `dd`.

If it is required that a system boot from floppy and use the same floppy drive to (re)write a new image to it, not only does the same floppy driver have to be used by using `root=` in /bootopts, but the root filesystem would have to be mounted read-only at boot time to make sure no system buffers overwrite the `dd` data. This can be done using `root=fd0 ro` or `root=df0 ro`, for instance. `sync` should be run and the system then be rebooted immediately after the `dd` completes for safety.

A better approach would be booting from HD which doesn't require the floppy drive being mounted.

toncho11, 2025-07-27:
No. Sorry I booted from hdd actually, not floppy when I did the dd. It was late here at night and I somehow got confused.

closed 2025-07-27 by toncho11