ghaerr/elks #2382: Telnet won't connect

jomagalo, 2025-08-22:
**Description**

I've read and reread the manual's instructions for port forwarding and tried a thousand ways, but I can't get it to connect, either from host to ELKS or ELKS to host.

`qemu -net nic,model=ne2k_isa -net user,hostfwd=tcp:127.0.0.1:2323-10.0.2.15:23 -hda hd32mbr-minix.img`
`qemu -net nic,model=ne2k_isa -net user,hostfwd=tcp:0.0.0.0:2323-10.0.2.15:23 -hda hd32mbr-minix.img`
`qemu -net nic,model=ne2k_isa -net user,hostfwd=tcp:192.168.1.x:2323-10.0.2.15:23 -hda hd32mbr-minix.img`

I'd like to compile Telnet to see if I can fix the problem, but it seems the `ttn.c` file is missing.

Thanks for your help.

ghaerr, 2025-08-22:
Hello @jomagalo,

Have you tried the options in qemu.sh? These look similar to what you're trying, but have been tested on macOS and are known to work. There may be issues with QEMU on Windows host, not sure, since I have no other information about "can't connect" error messages or more exactly what is happening.

> I'd like to compile Telnet to see if I can fix the problem, but it seems the ttn.c file is missing.

Sounds like you're looking at the `build_telnetc.sh` script in elkscmd/inet/telnet? That is outdated and should be removed: telnet won't compile on the host, only ELKS, since it now uses `ipaddr_t`, an ELKS libc address type. To rebuild ELKS telnet, just type `make` in elkscmd/, that will rebuild elkscmd/inet/telnet/telnet for further testing on ELKS.

Thank you!

jomagalo, 2025-08-22:
The options have been extracted from qemu.sh. Telnet doesn't connect in any way, and there are no error messages in the terminal. It doesn't seem to work on Windows x86. I've tested it on Raspberry Pi with ubuntu 20.04 and it works perfectly, except for FTP, which I can't get to connect.

Thanks for the information on the build.

ghaerr, 2025-08-22:
> I've tested it on Raspberry Pi with ubuntu 20.04 and it works perfectly,

> It doesn't seem to work on Windows x86.

Good to know. Considering that ELKS's interaction with QEMU is only thru the very low level link layer, that is, NIC ethernet card, then I would guess that the Windows version of QEMU itself isn't emulating the job of interpreting the passed ethernet packet to the virtual hardware device outside its own process to a Windows network daemon or onto the local network.

> Telnet doesn't connect in any way, and there are no error messages in the terminal. 

This usually means that the ELKS TCP/IP stack produced a TCP connection request that ends up going out the NIC, and all is waiting for a SYN return packet, which never arrives. Linux Wireshark allows for seeing this activity on the desktop, not sure whether there's something like that for snooping Windows desktop network activity outside QEMU.

When telnet is used on ELKS and connects to localhost, the implementation doesn't actually write to a NIC driver, but is instead looped back at the IP layer (above the link layer in the protocol stack). So QEMU wouldn't even see any NIC activity.

> except for FTP, which I can't get to connect.

You might try setting or unsetting passive mode; IIRC there are some issues with outside connections coming into an arbitrary port address. This is somewhat documented within the ELKS ftp.c source code alongside QEMU-referencing comments.

jomagalo, 2025-08-22:
Thank you very much for the clarifications; they have been very helpful.

Best regards

SuperMaxusa, 2025-08-22:
> Qemu Manager v0.11.1

Maybe it uses an outdated QEMU version? Are you using SLIRP or TAP backend?

Works for me on QEMU 10.0 (SLIRP) and Windows 10, one thing: I removed the IP addresses in `hostfwd`[^1].

 - Telnet: ELKS (`10.0.2.15:23`) -> host (`localhost:2323`):

[image]

 - HTTP server (should work with any other TCP server, `hostfwd` is not required): host (`localhost:8000`) -> ELKS (`10.0.2.2:8000`[^2]):

[image]
<br>

[^1]:  `-netdev user,id=mynet,hostfwd=tcp::2323-:23 -device ne2k_isa,irq=12,netdev=mynet`
[^2]: see https://wiki.qemu.org/File:Slirp_concept.png